espressif/esp-idf #7401: HTTP OTA update feature

krupis, 2021-08-10:
Hello. I would like to ask for a http OTA feature. Now the esp-idf only have https OTA implementation which is nice but it would be also nice to have a simple http OTA. For example in my project I want to implement both HTTPS and HTTP. HTTPS would be priority 1 and if it fails then it will attempt to do HTTP on another server. 

I have managed to implement my own http OTA function by duplicating esp_https_ota_begin function and naming it "esp_http_ota_begin_custom" , the only difference is that it has certificate check code commented out. I do not think that modifying the esp-idf library code is a good idea and I would much rather prefer to have both https and htpp ota functions out of the box when I install esp-idf.

My suggestion is to simply create another function which would allow to do OTA update without the need of certificate and

kayzen9, 2021-08-10:
https://github.com/espressif/esp-idf/blob/83956ebbae5b335fbe7537fee737c0784d1cf471/components/esp_https_ota/Kconfig#L3

krupis, 2021-08-10:
I am aware of this feature. 

Can you please tell me if I enable this, will I then be able to use HTTPS ?

As I mentioned before, I want both HTTPS and HTTP and not either this or that.

AxelLin, 2021-08-10:
> I am aware of this feature.
> 
> Can you please tell me if I enable this, will I then be able to use HTTPS ?
Yes.

krupis, 2021-08-10:
How does it then work? How do I must call the function to petform Http OTA
without providing certificate?

Keep in mind that I need 2 seperate functions:
One would perform HTTPS OTA with the certificate

The other would do just a simple HTTP OTA

krupis, 2021-08-11:
I have enabled OTA_ALLOW_HTTP and now it is able to perform HTTP OTA completely ignoring the certificate. You have mentioned that I can use HTTPS as well with this config enabled. Can you please explain how?

negativekelvin, 2021-08-12:
1. Https url, provide certificate
2. Http url, clear certificate

mahavirj, 2021-08-18:
Similar discussion (and one additional approach) covered at https://esp32.com/viewtopic.php?f=13&t=22672&p=81702. Closing the issue.

closed 2021-08-18 by mahavirj