espressif/esp-idf #7403: sd_card example

bmartinga, 2021-08-10:
Hello,
I've been running the sd_card example on the Wrover kit. We have new hardware that uses different gpio pins for the sd card interface. I've changed the pins as shown in the code snippet below.
#if CONFIG_IDF_TARGET_ESP32 || CONFIG_IDF_TARGET_ESP32S2
/*
#define PIN_NUM_MISO 2
#define PIN_NUM_MOSI 15
#define PIN_NUM_CLK  14
#define PIN_NUM_CS   13
*/
#define PIN_NUM_MISO 27
#define PIN_NUM_MOSI 14
#define PIN_NUM_CLK  12
#define PIN_NUM_CS   13
I have also uncommented the USE_SPI_MODE.

It works fine in either spi or smmc mode on the wrover kit. I've got pullups on each spi line but I always get the following
when using my target hardware.
I'm using idf 4.3. I've been working on this for 3 days with no luck. Any help would be appreciated.
Thanks,

E (402) sdmmc_sd: sdmmc_init_sd_if_cond: send_if_cond (1) returned 0x108
E (402) vfs_fat_sdmmc: sdmmc_card_init failed (0x108).
I (402) gpio: GPIO[13]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 0| Pulldo
wn: 0| Intr:0
E (412) example: Failed to initialize the card (ESP_ERR_INVALID_RESPONSE). Make
sure SD card lines have pull-up resistors in place.

nopnop2002, 2021-08-17:
GPIO12 is used as a bootstrapping pin to select output voltage of an internal regulator which powers the flash chip (VDD_SDIO). 

This pin has an internal pulldown.

This may be the cause.

adokitkat, 2021-10-03:
GPIO12 is used as MTDI bootstrapping pin read at startup to select the output voltage of an internal regulator. Read more in [documentation](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/peripherals/sd_pullup_requirements.html#mtdi-strapping-pin) please.

closed 2021-10-03 by adokitkat