espressif/esp-idf #7406: GATTC_Discover GATT_BUSY error in  GATTC client  detected continuously

sowmimalu, 2021-08-11:
Hai ,
Module or chip used: ESP32-WROVER
IDF version: v4.3
Compiler version: xtensa-esp32-elf-gcc (crosstool-NG esp-2020r3) 8.4.0

I have used BLE GATTC CLIENT example in my application to connect ble  device and sending a message via service and disconnects the connected device.

scanning for 10 secs to connect a device , if found stops the scans and sending data in esp_ble_gattc_write_char , in my case at a time connecting and sending data to one connection only . but sometimes if sending was failed then disconnection happens and later always getting the below  error for all the next new connections every time.
how to resolve from this ? 

I (6399901) GATTC_COASTER: Received MAC: dc:60:98:7:50:16
I (6399921) GATTC_COASTER: scan start success --- 0
I (6400241) GATTC_COASTER: Connecting to the remote device.
I (6400251) GATTC_COASTER: stop scan successfully
E (6400271) TW_READER: No S3/SP Found
E (6400521) BT_GATT: GATTC_Discover GATT_BUSY conn_id = 2
E (6400521) BT_APPL: discovery on server failed
E (6400521) GATTC_COASTER: discover service failed, status 132
I (6400531) GATTC_COASTER: open success
E (6400531) BT_GATT: GATTC_ConfigureMTU GATT_BUSY conn_id = 2

Thanks,
Sowmiya

sowmimalu, 2021-08-16:
Awaiting for reply ..

chegewara, 2021-08-16:
You are performing actions incorrect. Here for example you are trying to discover services before connection is accomplished:
```
E (6273931) BT_APPL: discovery on server failed
E (6273931) GATTC_COASTER: discover service failed, status 132
I (6273941) GATTC_COASTER: open success
```

If you dont wait to finish 1 action before call another you can get GATT_BUSY

sowmimalu, 2021-08-16:
connecting one connection at a time , after disconnection trying to another , still getting GATT BUSY continuosly

sowmimalu, 2021-08-18:
hai  ,
   Is anything wrong with the flow ?

sowmimalu, 2021-09-02:
Hai @xiewenxiang i have attached my demo code flow ...please let me know if have anything missed in flow.

[ble_demo.txt](https://github.com/espressif/esp-idf/files/7096345/ble_demo.txt)

Thanks
sowmiya

xiewenxiang, 2021-09-12:
@sowmimalu 

`ble_demo.txt` line:321

It seems that there should be modify as bellow:

```c
esp_ble_gattc_search_service(gattc_if, param->cfg_mtu.conn_id, &remote_filter_service_uuid);
```
=> 
```c
esp_ble_gattc_search_service(gattc_if,  param->dis_srvc_cmpl.conn_id, &remote_filter_service_uuid);
```

sowmimalu, 2021-09-13:
Hai @xiewenxiang ... after changing this connection also getting the same , the scenario i found was when one connection if gattc write were not happened after the successful connection of client and server ....then i disconnected the connection when the allocated  timer ends .again i am trying the next connection that time Continuously getting

E (396400) BT_GATT: GATTC_Discover GATT_BUSY conn_id = 2
E (396400) BT_APPL: discovery on server failed
W (396410) BT_APPL: gattc_conn_cb: if=1 st=0 id=1 rsn=0x16
W (396410) BT_APPL: gattc_conn_cb: if=2 st=0 id=2 rsn=0x16
E (396420) TW_READER: No S3/SP Found
E (396420) GATTC_COASTER: discover service failed, status 132---0
I (396430) GATTC_COASTER: ESP_GATTC_DISCONNECT_EVT, reason = 22

how to clear the gattc queue message if it fails ?

sowmimalu, 2021-09-16:
when GATTC BUSY occurs servcie discover failed by restarting the device for the next  conection  working prorperly . but when the issue occurs after the disconnection of peer device still for the next new connection occurs the same . how to clear or recover from the error i have used gattc clear cache / refresh before the disconnection , it need to recover after the disconnection . 
awaiting for reply
thanks

xiewenxiang, 2021-09-17:
I tried to reproduce this problem with the code you provided, I deleted part of the code that affected compile the demo, but it cannot reproduce this issue. 

Could you please provide an example that can be successfully compiled? 

The attachments are modified based on the code you provided, and some code are deleted because they affect compile.

xiewenxiang, 2021-09-17:
[ble_demo.zip](https://github.com/espressif/esp-idf/files/7182334/ble_demo.zip)

sowmimalu, 2021-09-17:
could you please share your mail id ...will share the demo for your reference

TianaESP, 2021-09-17:
Hi @sowmimalu
Can you please email sales@espressif.com, with the demo code and let our sales transfer to @xiewenxiang.  Thanks.

sowmimalu, 2021-09-17:
Thanks for the quick reply,

i have mailed the demo code.

I (504105) GATTC_DEMO: Connecting to the remote device.
I (504115) GATTC_DEMO: stop scan successfully
W (505615) BT_HCI: DiscCmpl evt: hdl=0, rsn=0x22
E (507175) BT_GATT: GATTC_Discover GATT_BUSY conn_id = 1
E (507175) BT_APPL: discovery on server failed
I (507175) GATTC_DEMO: ESP_GATTC_CONNECT_EVT conn_id 0, if 1
I (507175) GATTC_DEMO: REMOTE BDA:
I (507185) GATTC_DEMO: e3 22 90 83 f9 c3
E (507185) GATTC_DEMO: discover service failed, status 132
I (507195) GATTC_DEMO: open success
E (507195) BT_GATT: GATTC_ConfigureMTU GATT_BUSY conn_id = 1
E (507205) BT_APPL: No pending command
I (507945) GATTC_DEMO: one shot called

thanks
sowmiya

xiewenxiang, 2021-09-17:
@sowmimalu 
I have received the demo.
[image]

xiewenxiang, 2021-09-17:
@sowmimalu 

But it still fails to reproduce the issue you mentioned, what else should be paid attention to?

sowmimalu, 2021-09-17:
the scenario to reproduce - when  connection established with the device after service discovery and gattc opened the message sending has to fail then after the timeout disconnection.
the next new connection getting the same errror.

sowmimalu, 2021-09-17:
[GATTC_ERROR_LOG.txt](https://github.com/espressif/esp-idf/files/7183730/GATTC_ERROR_LOG.txt)

@xiewenxiang please check this log after the write suucess count one message got failure after OPEN succes then it was disconnected . but the new connection getting GATTC BUSY to find service.

sowmimalu, 2021-09-18:
Hi @xiewenxiang Any update on this ? i m looking into this error but not able to recover for the next connection in Example gattc client . have to add this gattc into my application file . Awaiting for reply please provide me any suggestions on this if you can able to reproduce the same in demo code.

Thanks

zhp0406, 2023-09-11:
@sowmimalu 
Have you resolved the issue?

Indastri, 2023-11-22:
Closing this issue since there is no reaction from the user side, if there is still investigation needed feel free to reopen or create a new issue.

closed 2023-11-22 by Indastri