espressif/esp-idf #7409: SIG_SUPPLICANT_DEL_TASK race condition in esp_common.c

cmsedore, 2021-08-12:
In esp_common.c, the function esp_supplicant_post_evt has the following code:

       SUPPLICANT_API_LOCK();
	if (xQueueSend(s_supplicant_evt_queue, &evt, 10 / portTICK_PERIOD_MS ) != pdPASS) {
		SUPPLICANT_API_UNLOCK();
		os_free(evt);
		return -1;
	}

	 SUPPLICANT_API_UNLOCK();

Unfortunately, when the SIG_SUPPLICANT_DEL_TASK is posted as the event, that results in this code block being executed in esp_btm_rrm_task:

	if (s_supplicant_api_lock) {
		vSemaphoreDelete(s_supplicant_api_lock);
		s_supplicant_api_lock = NULL;
	}

You may notice that this causes the s_supplicat_api_lock to be released/nulled and then when the function call unwinds, it often causes an assertion failure and panic.  I don't know enough of how this code is structured to do a proper fix, but I changed the first code block to this to address it in my work:

	SUPPLICANT_API_LOCK();
	if (xQueueSend(s_supplicant_evt_queue, &evt, 10 / portTICK_PERIOD_MS ) != pdPASS) {
		SUPPLICANT_API_UNLOCK();
		os_free(evt);
		return -1;
	}
	if (evt_id!=SIG_SUPPLICANT_DEL_TASK)
	    SUPPLICANT_API_UNLOCK();

This is probably not the ideal fix, but at least I don't get a panic 9/10 times when running esp_wifi_deinit().

To be clear, you should trigger this by doing:

esp_wifi_stop();
esp_wifi_deinit();

It may require some specific other circumstances to trigger the race condition (my build is running as a single core, for example).  For me it is triggered >90% of the time.

kapilkedawat, 2021-08-12:
Hi @cmsedore , thanks for pointing out, we will make appropriate changes.

closed 2021-12-28 by espressif-bot

AxelLin, 2021-12-28:
@kapilkedawat 

In https://github.com/espressif/esp-idf/blob/master/components/wpa_supplicant/esp_supplicant/src/esp_common.c#L272-L277
If s_supplicant_api_lock = xSemaphoreCreateRecursiveMutex(); fails:

s_supplicant_api_lock is NULL, then it calls
```
  esp_supplicant_common_deinit
    esp_supplicant_post_evt
      SUPPLICANT_API_LOCK -> calls xSemaphoreTakeRecursive with **NULL**.
```

kapilkedawat, 2021-12-28:
esp_supplicant_post_evt  checks s_supplicant_api_lock  before calling SUPPLICANT_API_LOCK() https://github.com/espressif/esp-idf/blob/a5227db2a75102ca1a17860188c3c352a529a01b/components/wpa_supplicant/esp_supplicant/src/esp_common.c#L594

AxelLin, 2021-12-28:
oops. my bad. Just realize I check wrong branch in my local code.
BTW, needs to free evt before return -1 in below line.
https://github.com/espressif/esp-idf/blob/a5227db2a75102ca1a17860188c3c352a529a01b/components/wpa_supplicant/esp_supplicant/src/esp_common.c#L597

kapilkedawat, 2022-01-03:
> And when the code return from above line, the btm_rrm_task is never deleted.

This actually mean task either never got created or already got deleted.

AxelLin, 2022-01-03:
> > And when the code return from above line, the btm_rrm_task is never deleted.
> 
> This actually mean task either never got created or already got deleted.

no.  btm_rrm_task is already created.
https://github.com/espressif/esp-idf/blob/a5227db2a75102ca1a17860188c3c352a529a01b/components/wpa_supplicant/esp_supplicant/src/esp_common.c#L274

cmsedore, 2022-02-25:
This bug seems to have reappeared in 4.4 Release.

```
int esp_supplicant_post_evt(uint32_t evt_id, uint32_t data)
{
	supplicant_event_t *evt = os_zalloc(sizeof(supplicant_event_t));
	if (evt == NULL) {
		return -1;
	}
	evt->id = evt_id;
	evt->data = data;

	SUPPLICANT_API_LOCK();
	if (xQueueSend(s_supplicant_evt_queue, &evt, 10 / portTICK_PERIOD_MS ) != pdPASS) {
		SUPPLICANT_API_UNLOCK();
		os_free(evt);
		return -1;
	}
	SUPPLICANT_API_UNLOCK();
	return 0;
}
```