espressif/esp-idf #7411: Support rwlock's in pthread

MabezDev, 2021-08-12:
As part of an ongoing effort to bring Rust support to Espressif chips, on requirement we found is the need for `rwlock` to fully support the Rust standard library. For our initial implementation was accepted on the basis that rwlock support would come to esp-idf soon :tm: (https://github.com/rust-lang/libc/pull/2310#issuecomment-891403847). 

We are currently getting away with the lack of support with a nasty hack [using a mutex](https://github.com/esp-rs/esp-idf-sys/blob/04bc5f02c4350428174bf3bd274a32765565e4c9/src/pthread_rwlock.rs). Obviously we want to get rid of this asap.

0xjakob, 2021-08-25:
@MabezDev Any requirements regarding the implementation? In particular, is writer starvation prevention necessary?

MabezDev, 2021-08-25:
From the Rust standard library on the `RWLock`'s:

> The priority policy of the lock is dependent on the underlying operating system’s implementation, and this type does not guarantee that any particular policy will be used. In particular, a writer which is waiting to acquire the lock in write might or might not block concurrent calls to read.

Therefore I don't think we need writer starvation prevention initially, and we are open to change it later if we wish too. I think for now the simplest implementation will do.

closed 2021-11-08 by espressif-bot