espressif/esp-idf #7414: ESP32-C3 SPI3_HOST not recognized

bbru72, 2021-08-13:
Hi,

I have an issue with SPI master.
ret = spi_bus_initialize(SPI2_HOST, &buscfg, DMA_CHAN); this is OK
but
ret = spi_bus_initialize(SPI3_HOST, &buscfg, DMA_CHAN); this is not OK
During execution, it fails with the error message:
spi_bus_initialize(627): invalid host_id
But SPI3_HOST should be recognized as illustrated hereunder.

in spi_types.h
SPI_EV_BUF_TXtypedef enum {
//SPI1 can be used as GPSPI only on ESP32
SPI1_HOST=0, ///< SPI1
SPI2_HOST=1, ///< SPI2
SPI3_HOST=2, ///< SPI3
} spi_host_device_t;

bbru72, 2021-08-13:
**Environment:**
board ESP32-C3-DevKitC-02
ESP-IDF V4.3
chip revision 3

igrr, 2021-08-13:
Could you please check [this documentation page](https://docs.espressif.com/projects/esp-idf/en/latest/esp32c3/api-reference/peripherals/spi_master.html#overview-of-esp32-c3-s-spi-peripherals), does it answer your question?

bbru72, 2021-08-13:
Thanks Igrr for your feedback
I did some search and found that in spi_common.c line 102
_**static inline bool is_valid_host(spi_host_device_t host)
{
#if (SOC_SPI_PERIPH_NUM == 2)
    return host >= SPI1_HOST && host <= SPI2_HOST;
#elif (SOC_SPI_PERIPH_NUM == 3)
    return host >= SPI1_HOST && host <= SPI3_HOST;
#endif
}**_
looking at socs_caps.h line 175
**_#define SOC_SPI_PERIPH_NUM          2_**
So why it is 2 instaed or 3, ESP32-C3 has 3 SPI interface and SPI3_HOST is for general purpose

igrr, 2021-08-13:
That's correct that ESP32-C3 has 3 SPI peripherals. As explained in the documentation link I posted, SPI0 and SPI1 are used to access the main flash chip where the program is stored. SPI2 is the general purpose SPI that can be used with SPI master driver.

I think SOC_SPI_PERIPH_NUM 2 refers to the fact that there are 2 peripherals which could potentially be used by software: SPI1 and SPI2. However as also mentioned in that documentation page, currently SPI Master driver doesn't support SPI1 bus on ESP32-C3. At some point it might get supported — then SPI1 (attached to the flash chip) and SPI2 (general purpose) will be available for use.

bbru72, 2021-08-13:
Thanks for your feedback to summarize: 
SPI2_HOST = SPI1 = 1
SPI3_HOST = SPI2 = 2
So calling  **is_valid_host(SPI3_HOST)** will failed due to the fact that SOC_SPI_PERIPH_NUM = 2
Did you try on your side to call: **spi_bus_initialize**(SPI3_HOST, &buscfg, DMA_CHAN); ?

igrr, 2021-08-13:
No I think SPI1_HOST is SPI1 (not supported by spi_master driver on C3 yet), SPI2_HOST is SPI2 (general purpose SPI, supported by the driver on C3), SPI3_HOST doesn't exist in C3 chip.

bbru72, 2021-08-15:
It's correct with SPI1, it works

Alvin1Zhang, 2021-08-16:
@bbru72 Thanks for reporting and sharing updates, feel free to reopen.

closed 2021-08-16 by Alvin1Zhang