espressif/esp-idf #7415: Very slow HTTP server response

atlascoder, 2021-08-13:
## Environment

- Development Kit: ESP32-CAM - A.I. Thinker
- Kit version (for WroverKit/PicoKit/DevKitC): unknown
- Module or chip used:  [ESP32-S]
- IDF version (run ``git describe --tags`` to find it):
    Docker image espressif/idf:release-v4.1
- Build System: idf.py
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it):
    xtensa-esp32-elf-gcc (crosstool-NG esp-2020r3) 8.4.0
- Operating System: Linux (Docker)
- Using an IDE?: VSCode
- Power Supply: external 5V

## Problem Description

Very, very slow responses from HTTP server, when ESP is running in STA mode. A sending of a JPEG shot having 125KB may take more than 10 seconds !!!

From another side, when I run ESP in SoftAP mode - it works just fine, less than a second!

I tried to create manual HTTP server implementation using Socket API - result is the same..

I changed WiFi channel on my router, suspecting that maybe WiFi channel is too busy - no significant changes. I also have many other devices (Not esp - smartphones, tv, laptops, etc) - all work fine in this WiFi network.

I also tried different boards - nothing helps. That wan not ESP32-CAM, but DevKit-C, a custom PICO-D4 board - the result is the same.

I turned of AMPDU, changed WiFi channel witdh, played with other params and just can't get a good result.

I've tried to set `esp_wifi_set_ps(WIFI_PS_NONE)` as it was an option from here #7328 - doesn't help.

### Expected Behavior

My application is very critical in power consumption. Its workcycle means many wakeups from sleeping for takink a photo and sending to a server. I understand that here is a difference: the ESP intiates a request, since ESP plays as a client, but I came to ESP32-CAM testing scenario just when trying to localize the problem in my project, where I send a shot to a server using lwip socket API.

I need to send data as fast as possible.

### Actual Behavior

Very slow data sending via socket.

### Steps to reproduce

1. clone the repo `https://github.com/atlascoder/esp32-cam.git`, checkout `slow_http_issue` branch
2. set WiFi credentials using menuconfig -> Component config -> Example configuration -> WiFi configuration
3. Build and flash the app
4. Run idf.py monitor, reboot the ESP and get IP address of the ESP once it will get connected to WiFi
5. Open a browser on a host that is in the same WiFi network and open picture http://**ESP_IP_ADDRESS**/jpg
6. Check reply time in the monitor logs

### Code to reproduce this issue

I took the code from here https://github.com/raphaelbs/esp32-cam-ai-thinker from `examples` and made just a few modifications for WiFi handling in IDF-4.1

## Debug Logs

Replyting with JPG data 192KB in 19045ms:

```
D (6622) example:http_jpg: httpd started
D (10592) httpd: httpd_server: processing listen socket 54
D (10592) httpd: httpd_accept_conn: newfd = 57
D (10592) httpd_sess: httpd_sess_new: fd = 57
D (10602) httpd: httpd_accept_conn: complete
D (10602) httpd: httpd_server: doing select maxfd+1 = 58
D (10612) httpd: httpd_server: processing socket 57
D (10612) httpd_sess: httpd_sess_process: httpd_req_new
D (10622) httpd_txrx: httpd_recv_with_opt: requested length = 128
D (10622) httpd_txrx: httpd_recv_with_opt: received length = 128
D (10632) httpd_parse: read_block: received HTTP request block size = 128
D (10632) httpd_parse: cb_url: message begin
D (10642) httpd_parse: cb_url: processing url = /jpg
D (10642) httpd_parse: verify_url: received URI = /jpg
D (10652) httpd_parse: cb_header_field: headers begin
D (10652) httpd_txrx: httpd_unrecv: length = 109
D (10662) httpd_parse: pause_parsing: paused
D (10662) httpd_parse: cb_header_field: processing field = Host
D (10672) httpd_txrx: httpd_recv_with_opt: requested length = 128
D (10682) httpd_txrx: httpd_recv_with_opt: pending length = 109
D (10682) httpd_parse: read_block: received HTTP request block size = 109
D (10692) httpd_parse: continue_parsing: skip pre-parsed data of size = 5
D (10692) httpd_parse: continue_parsing: un-paused
D (10702) httpd_parse: cb_header_value: processing value = 192.168.0.124
D (10712) httpd_parse: cb_header_field: processing field = Connection
D (10712) httpd_parse: cb_header_value: processing value = keep-alive
D (10722) httpd_parse: cb_header_field: processing field = Cache-Control
D (10732) httpd_parse: cb_header_value: processing value = max-age=0
D (10732) httpd_parse: cb_header_field: processing field = Upgrade-Insecure-Requests
D (10742) httpd_parse: cb_header_value: processing value = 1
D (10752) httpd_parse: cb_header_field: processing field = User-Age
D (10752) httpd_parse: parse_block: parsed block size = 109
D (10762) httpd_txrx: httpd_recv_with_opt: requested length = 128
D (10762) httpd_txrx: httpd_recv_with_opt: received length = 128
D (10772) httpd_parse: read_block: received HTTP request block size = 128
D (10782) httpd_parse: cb_header_field: processing field = nt
D (10782) httpd_parse: cb_header_value: processing value = Mozilla/5.0 (X11; Linux x86_64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/92.0.4515.107 Safari/537.36
D (10802) httpd_parse: cb_header_field: processing field = Accept
D (10802) httpd_parse: cb_header_value: processing value = text/html
D (10812) httpd_parse: parse_block: parsed block size = 237
D (10822) httpd_txrx: httpd_recv_with_opt: requested length = 128
D (10822) httpd_txrx: httpd_recv_with_opt: received length = 128
D (10832) httpd_parse: read_block: received HTTP request block size = 128
D (10842) httpd_parse: cb_header_value: processing value = ,application/xhtml+xml,application/xml;q=0.9,image/avif,image/webp,image/apng,*/*;q=0.8,application/signed-exchange;v=b3;q=0.9
D (10852) httpd_parse: parse_block: parsed block size = 365
D (10862) httpd_txrx: httpd_recv_with_opt: requested length = 128
D (10862) httpd_txrx: httpd_recv_with_opt: received length = 76
D (10872) httpd_parse: read_block: received HTTP request block size = 76
D (10882) httpd_parse: cb_header_field: processing field = Accept-Encoding
D (10882) httpd_parse: cb_header_value: processing value = gzip, deflate
D (10892) httpd_parse: cb_header_field: processing field = Accept-Language
D (10902) httpd_parse: cb_header_value: processing value = en-US,en;q=0.9,uk;q=0.8
D (10902) httpd_parse: cb_headers_complete: bytes read     = 462
D (10912) httpd_parse: cb_headers_complete: content length = 0
D (10922) httpd_parse: pause_parsing: paused
D (10922) httpd_parse: cb_no_body: message complete
D (10932) httpd_parse: httpd_parse_req: parsing complete
D (10932) httpd_uri: httpd_uri: request for /jpg with type 1
D (10942) httpd_uri: httpd_find_uri_handler: [0] = /jpg
D (10942) httpd_txrx: httpd_resp_set_hdr: new header = Content-Disposition: inline; filename=capture.jpg
D (10952) httpd_txrx: httpd_send_all: sent = 67
D (10962) httpd_txrx: httpd_send_all: sent = 19
D (10962) httpd_txrx: httpd_send_all: sent = 2
D (10972) httpd_txrx: httpd_send_all: sent = 28
D (10972) httpd_txrx: httpd_send_all: sent = 2
D (10972) httpd_txrx: httpd_send_all: sent = 2
D (16592) httpd_txrx: httpd_send_all: sent = 37291
D (21612) httpd_txrx: httpd_send_all: sent = 84736
D (26642) httpd_txrx: httpd_send_all: sent = 41648
D (29992) httpd_txrx: httpd_send_all: sent = 33145
I (29992) example:http_jpg: JPG: 192KB 19045ms
D (29992) httpd_sess: httpd_sess_process: httpd_req_delete
D (29992) httpd_sess: httpd_sess_process: success
```

atlascoder, 2021-08-16:
# Update
I've disabled tcpip_adapter backward compatibility in the ESP-NETIF Component in Menuconfig. Some code fixes were made to use NETIF only. Reflashed the app, and requested a JPEG.. The logs:
```
...
D (17112) httpd: httpd_server: doing select maxfd+1 = 58
D (17112) httpd: httpd_server: processing socket 57
D (17122) httpd_sess: httpd_sess_process: httpd_req_new
D (17122) httpd_txrx: httpd_recv_with_opt: requested length = 128
D (17132) httpd_txrx: httpd_recv_with_opt: received length = 128
D (17132) httpd_parse: read_block: received HTTP request block size = 128
D (17142) httpd_parse: cb_url: message begin
D (17152) httpd_parse: cb_url: processing url = /jpg
D (17152) httpd_parse: verify_url: received URI = /jpg
D (17162) httpd_parse: cb_header_field: headers begin
D (17162) httpd_txrx: httpd_unrecv: length = 109
D (17172) httpd_parse: pause_parsing: paused
D (17172) httpd_parse: cb_header_field: processing field = Host
D (17182) httpd_txrx: httpd_recv_with_opt: requested length = 128
D (17182) httpd_txrx: httpd_recv_with_opt: pending length = 109
D (17192) httpd_parse: read_block: received HTTP request block size = 109
D (17202) httpd_parse: continue_parsing: skip pre-parsed data of size = 5
D (17202) httpd_parse: continue_parsing: un-paused
D (17212) httpd_parse: cb_header_value: processing value = 192.168.0.124
D (17212) httpd_parse: cb_header_field: processing field = Connection
D (17222) httpd_parse: cb_header_value: processing value = keep-alive
D (17232) httpd_parse: cb_header_field: processing field = Upgrade-Insecure-Requests
D (17232) httpd_parse: cb_header_value: processing value = 1
D (17242) httpd_parse: cb_header_field: processing field = User-Agent
D (17252) httpd_parse: cb_header_value: processing value = Mozilla/5.0 (X11; Linu
D (17252) httpd_parse: parse_block: parsed block size = 109
D (17262) httpd_txrx: httpd_recv_with_opt: requested length = 128
D (17272) httpd_txrx: httpd_recv_with_opt: received length = 128
D (17272) httpd_parse: read_block: received HTTP request block size = 128
D (17282) httpd_parse: cb_header_value: processing value = x x86_64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/92.0.4515.107 Safari/537.36
D (17292) httpd_parse: cb_header_field: processing field = Accept
D (17302) httpd_parse: cb_header_value: processing value = text/html,application/xhtml+xml,app
D (17312) httpd_parse: parse_block: parsed block size = 237
D (17312) httpd_txrx: httpd_recv_with_opt: requested length = 128
D (17322) httpd_txrx: httpd_recv_with_opt: received length = 128
D (17322) httpd_parse: read_block: received HTTP request block size = 128
D (17332) httpd_parse: cb_header_value: processing value = lication/xml;q=0.9,image/avif,image/webp,image/apng,*/*;q=0.8,application/signed-exchange;v=b3;q=0.9
D (17352) httpd_parse: cb_header_field: processing field = Accept-Encoding
D (17352) httpd_parse: cb_header_value: processing value = gzip, def
D (17362) httpd_parse: parse_block: parsed block size = 365
D (17362) httpd_txrx: httpd_recv_with_opt: requested length = 128
D (17372) httpd_txrx: httpd_recv_with_opt: received length = 50
D (17382) httpd_parse: read_block: received HTTP request block size = 50
D (17382) httpd_parse: cb_header_value: processing value = late
D (17392) httpd_parse: cb_header_field: processing field = Accept-Language
D (17402) httpd_parse: cb_header_value: processing value = en-US,en;q=0.9,uk;q=0.8
D (17402) httpd_parse: cb_headers_complete: bytes read     = 437
D (17412) httpd_parse: cb_headers_complete: content length = 0
D (17422) httpd_parse: pause_parsing: paused
D (17422) httpd_parse: cb_no_body: message complete
D (17432) httpd_parse: httpd_parse_req: parsing complete
D (17432) httpd_uri: httpd_uri: request for /jpg with type 1
D (17442) httpd_uri: httpd_find_uri_handler: [0] = /jpg
D (17442) httpd_txrx: httpd_resp_set_hdr: new header = Content-Disposition: inline; filename=capture.jpg
D (17452) httpd_txrx: httpd_send_all: sent = 67
D (17462) httpd_txrx: httpd_send_all: sent = 19
D (17462) httpd_txrx: httpd_send_all: sent = 2
D (17472) httpd_txrx: httpd_send_all: sent = 28
D (17472) httpd_txrx: httpd_send_all: sent = 2
D (17472) httpd_txrx: httpd_send_all: sent = 2
D (22812) httpd_txrx: httpd_send_all: sent = 8571
D (28312) httpd_txrx: httpd_send_all: sent = 14384
D (33312) httpd_txrx: httpd_send_all: sent = 11504
D (38312) httpd_txrx: httpd_send_all: sent = 7200
D (43312) httpd_txrx: httpd_send_all: sent = 2880
D (48812) httpd_txrx: httpd_send_all: sent = 15824
D (53812) httpd_txrx: httpd_send_all: sent = 1440
D (58812) httpd_txrx: httpd_send_all: sent = 1440
D (63812) httpd_txrx: httpd_send_all: sent = 4304
D (68812) httpd_txrx: httpd_send_all: sent = 4320
D (73812) httpd_txrx: httpd_send_all: sent = 1440
D (79312) httpd_txrx: httpd_send_all: sent = 20128
D (84312) httpd_txrx: httpd_send_all: sent = 2880
D (89812) httpd_txrx: httpd_send_all: sent = 2864
D (94812) httpd_txrx: httpd_send_all: sent = 43104
D (99812) httpd_txrx: httpd_send_all: sent = 50272
D (104812) httpd_txrx: httpd_send_all: sent = 2864
D (110312) httpd_txrx: httpd_send_all: sent = 5760
D (115612) httpd_txrx: httpd_send_all: sent = 17264
D (115612) httpd_txrx: httpd_send_all: sent = 2028
I (115612) example:http_jpg: JPG: 215KB 98167ms
D (115612) httpd_sess: httpd_sess_process: httpd_req_delete
D (115622) httpd_sess: httpd_sess_process: success
...
```

Very slow sending: 215 KB in 98 sec!

Also, from `D (28312) httpd_txrx: httpd_send_all: sent = 14384` it looks that logging function is called with stable intervals 5 or 6.5 seconds, and sometimes, only single Ethernet packet is sent in 6 seconds...

atlascoder, 2021-08-18:
# Update

I made an experiment with running the ESP32-CAM board in isolated WiFi network.

A WiFi network was created by a SoftAP running on an Linux laptop. WiFi with SSID 'apapa' was set on channel 6, which was previousely discovered as not very habitated channel using LinSSID Sofware.
[image]

The ESP board was connected with IP 192.168.8.98 to the SoftAP, also an iPhone was connected to the AP with IP 192.168.8.81. `http://192.168.8.89/jpg` was requested from iPhone and all the TCP session was captured by Wireshark.

[image]
[image]
[image]
[image]
[image]
[image]
[image]

I took this data an made a timeline chart: blue - is X axe is time im microseconds, red bars are transmissions from ESP to iPhone, blue bars - from iPhone to ESP. in this session, ESP had 1Mbps connction, an iPhone 11Mbps, so transmissions from iPhone to ESP looks as disappiring lines. (Open the chard below in separate window and scale).

[image]

The transmission took ~8 seconds for 192K of data!!

## My conclusions

- the problemis likely in the PHY or Data link layer
- it looks like ESP doesn't use WiFi channel since transmissions have large pauses, though the channel is relatively free

ESP-YJM, 2021-08-19:
@atlascoder It looks like a poor environment causes TCP retransmissions, resulting in longer transmission time. Could you please use ethernet instead of WIFI to test. I think it shouldn't be a problem with http server.

atlascoder, 2021-08-19:
@ESP-YJM, I agree with you, that the problem is not likely in the `http_server` component.. Please look at my previous comment. There is only one TCP restransmission in the cpatured WiFi, but it looks like the ESP uses WiFi channel poorely, a lot of time is not used for transmission while it's clear that the channel can be used more intensively.

WiFi signal quality is very good - please, look at the details of transmission to ESP

[image]

and from ESP

[image]

Maybe it's reasonable to rename this question to WiFi related?

AxelLin, 2021-08-19:
@atlascoder

Could you upload raw pcap file instead of the png file.
(You probably need to rename .pcap to .pcap.txt for upload)

atlascoder, 2021-08-19:
@AxelLin, please, [get_jpg.pcapng.gz](https://github.com/espressif/esp-idf/files/7012486/get_jpg.pcapng.gz)

atlascoder, 2021-08-19:
@AxelLin , please, ask every additional info if needed

ESP-YJM, 2021-08-19:
@atlascoder Could you please share your router AP ssid and password that i can decode the packet.

atlascoder, 2021-08-19:
@ESP-YJM 
SSID: apapa
Pass: HeyWiFiWiFi

ESP-YJM, 2021-08-19:
@atlascoder I check the packet and find that the TCP retransmissions almost retransmit once and not occupy to much time. The reason that send 192 KB cost  8 seconds due to too slow wifi data rate, that is only 1 Mbps most time and 11 Mbps some time.  And your AP(apapa) support max data rate is 11Mbps. WIFI retransmissions will make data rate reduce to 1Mbps. So it will cost much time to send data. Suggest to change a AP which can support 54Mbps data rate over.

AxelLin, 2021-08-19:
> WIFI retransmissions will make data rate reduce to 1Mbps. So it will cost much time to send data. 

Why WIFI retransmissions will make data rate reduce to 1Mbps?
I'm curious if this is a **standard** behavior?

atlascoder, 2021-08-19:
@ESP-YJM , yes, you describe the picture as I see it, but in my opinion, the problem is not in 1Mbps connection - it is in that transmissions are very spreaded in time. Please, look at this picture (open in separate window and zoom):

[image]

- I made it from the same pcap file and it shows how transmissions are distributed in time. Red bars - from ESP, blue ones - from iPhone.

The question I have regarding this chart is: why there is so big pauses between transmissions?

Please, note, that this experiment a made after days of surveying the reason when connection to my relatively modern Tp-Link Archer C2 hardware router that is working with many devices in my network without any problem of this kind.

I will collect more data today with using my hardware router network.

atlascoder, 2021-08-19:
> I'm curious if this is a **standard** behavior?
- yes, with this SoftAP router it always become 1 Mbps, though, it's really strange why, because signal is really of good quality.. I will check with another router later.

ESP-YJM, 2021-08-19:
@AxelLin Yes, it is a standard behavior. For example, we support 1, 11, 54 Mbps wifi rate. And we use 54 rate to transmit at first. If wifi WIFI retransmissions always happen, it will reduce to 11 Mbps and retransmissions still exist, it will reduce 1 Mbps, but if retransmissions disappear，it will increase to 11 and 54.

atlascoder, 2021-08-19:
> @AxelLin Yes, it is a standard behavior. For example, we support 1, 11, 54 Mbps wifi rate. And we use 54 rate to transmit at first. If wifi WIFI retransmissions always happen, it will reduce to 11 [...]

Packet # 427 in the pcap is 1 packet in 4-way handshake, and it has 1 Mbps already. Packet # 422 is association request also goes with 1 Mbps. Maybe here is something wrong?

Any thoughts about long pauses between transmissions?

ESP-YJM, 2021-08-19:
> Packet # 427 in the pcap is 1 packet in 4-way handshake, and it has 1 Mbps already. Packet # 422 is association request also goes with 1 Mbps. Maybe here is something wrong?
>
> Any thoughts about long [...]

For 4-way handshake, it think the data rate 1Mbps is ok. This is control frame. But for data frame, it seems always 1 Mbps, i not know why and something wrong. 

Could you please ponit the detailed packet that pauses long time?

atlascoder, 2021-08-19:
@ESP-YJM it's not about a separate packet, I mean sequence of packets during TCP session behind a HTTP request. I filtered this sequence by applying 'tcp' filter in Wireshark. The chart, I attached above, is made for this filtered packets - I am referencing this chart when saying about pauses..

ESP-YJM, 2021-08-20:
@atlascoder I think the pause between TCP packet affects by the peer ack(TCP layer) and WIFI transmission channel preemption(WIFI layer). For TCP layer, if we want to send next packet and our send buffer is full, we need wait the peer ack to inform the userspace that can send data. So the speed of the ack reply from the peer will affect the interval to send packets. For WIFI layer, if we want to send packet to air, we need grab channel to send data. So I think it has something to do with environmental factors. The device requesting jpg can be wired connection, and the environmental factors of wireless connection are excluded. In addition, recommended to choose a high-speed AP.

atlascoder, 2021-08-20:
I seems that I've found the problem source.. I tried again to change WiFi channel and on some channels, transmissions were just fine - 200-500 ms for a shot with ~100KB. For me it's good enough in contrast how it was..

But it is still unclear, why while other devices work fast in the same channel, ESP can be so slow?

This can be a stable source of customers complains in production and it is hard to explain to users, why other device work well and this one - doesn't.

Unfortunately, I don't have have a tool to measure particular WiFi chanel utilization. Also, I have many other routers around, of different manufacturers, so they may have bugs in PHY/MAC implementation, or antenna of my ESP may also be not good for the channel that was slow.

ESP-YJM, 2021-08-20:
@atlascoder I don't know the reason that why other devices work fast in the same channel.  Maybe signal, antenna, PHY so on .... For this issue, you can re-open an wifi issue to seek for help. What you asked may be beyond my knowledge.

atlascoder, 2021-08-20:
@ESP-YJM, I agree, it is not about http/lwip, the issue can be closed. Thank you!
Later, I will open another issue related to WiFi with reference on this one.

atlascoder, 2021-08-20:
The problem is not related to HTTP/LwIP.

closed 2021-08-20 by atlascoder