espressif/esp-idf #7416: Problem when using ESP internal clock when using ethernet PHY evaluation board lan8920 in ESP-AT

charon22, 2021-08-13:
## Environment

- Development Kit:      [ESP32-DevKitC]
- Module or chip used:  [ESP32-WROOM-32]
- IDF version (run ``git describe --tags`` to find it):
    // 4.2.2
- Operating System:     [Windows]
- (Windows only) environment type: [PowerShell].
- Using an IDE?: [No]
- Power Supply:         [USB]

## Problem Description

Problem when using ESP internal clock when using ethernet PHY evaluation board lan8920 in ESP-AT.
We tested three methods with the problem that occurred when using the ESP-AT command by connecting the ethernet phy to the ESP internal clock. 
The conclusion is that when using UART1 in ESP32, it cannot be used due to a strange signal.
I have inquired to ESP-AT for the following information, but they said that it is better to contact ESP-IDF.

**Method 1) use UART2 for ESP-AT cmd:**
>   When using UART2(GPIO 16,17) of ESP as ESP-AT command, and when using GPIO0 as REF_CLK of RMII, there was a problem with the clock. CLOCK is output normally, but it fails if there is a problem with SYNC. So, I solved it by extending the RMII clock line. This is difficult to solve in H/W when making a PCB board, so REF_CLK of RMII is modified to GPIO17.
[image]

**Question**:
>If we use UART2 as UART port for ESP-AT command, we want to know why the line of RMII REF_CLK has to be doubled. How can I solve this problem?

**Method 2) use UART1 for ESP-AT cmd:**
>  When using UART1(GPIO 9,10) of ESP as ESP-AT command, and when using GPIO17 as REF_CLK of RMII, there was no problem with the clock. But, A problem occurred when using the ESP-AT command. When using as UART1, abnormal signals of 1us sometimes come in, making UART communication impossible. See below for more details on this. 
[image]
[image]
>Below is the captured UART Rx, Tx signal through logic analysis when using SYSFLASH of ESP-AT command from device to ESP board.
[2021-07-30_17-18-26-2.zip](https://github.com/espressif/esp-at/files/6926861/2021-07-30_17-18-26-2.zip)
> A normal UART signal has a period of 8.6us, but an abnormal signal has a period of 1.6 us, making UART unusable.

**Question**:
> If we want to know how to cancel the signal such as noise generated when we use UART1 as UART port for ESP-AT command. How can I solve this problem? I would like to know if this problem only occurs on LAN8720.
=> If method 1) does not solve the problem, method 2) has to be used.

**Method3 ) use UART0 for ESP-AT cmd:**
> I modified uart0 to use ESP-AT command, and modified uart1 to uart console display. 
> At this time, it was possible to send and receive AT commands normally with UART0. 
> But when I check UART1, I get an error. Only 1.6us strange signal is generated from UART1 of ESP board, and UART Console message could not be checked. 
> Below is the captured UART Tx signal through logic analysis when the ESP board is reset.
[2021-08-04_14-35-46_uart1.zip](https://github.com/espressif/esp-at/files/6934578/2021-08-04_14-35-46_uart1.zip)

**Question** 
> If method 3 is used, I want to know if there is any way to see normal esp console messages with UAR1.

Thanks

kostaond, 2021-08-16:
Hi @charon22, it seems you have some issues with your hardware design, please see below:
Ad 1) You have signal integrity problem in this configuration. GPIO0 and GPIO17 has the same source of clock -  APLL_CLK. However, clock at GPIO17 is inverted by 180°. Hence its name EMAC_CLK_180. When you extend wire, you add a delay to the line so the clock appears to be inverted and the input of the LAN8920. Therefore, if you want to use REF_CLK at GPIO0, you need to invert the clock using inverter, please see [here](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/hw-reference/esp32/get-started-ethernet-kit.html#rmii-clock-sourced-internally-from-esp32-s-apll) for more details. 
Ad 2) Your UART does not work because you use GPIO's which are already taken by integrated SPI flash, please see [here](https://www.espressif.com/sites/default/files/documentation/esp32-wroom-32_datasheet_en.pdf), page 9.
Ad 3) The same issue as above.

Please have a look and let me know if you needed further clarification.

charon22, 2021-08-17:
Hi @kostaond, 
Thank you for the reply. In the end, it seems that 2 and 3 are not available methods.
I want to know what is the requirement of  max propagation delay for the inverter used for internal clock output from GPIO0.
Thanks

negativekelvin, 2021-08-17:
You don't have to use default gpio for uart because you can remap to any pin

kostaond, 2021-08-18:
@charon22 I cannot provide you any exact answer for your question since it will be highly dependent on your actual PCB design. You need to do your own research and analysis. Below are some links you can start with:
https://electronics.stackexchange.com/questions/390523/inverting-refclock-of-ethernet-phy
https://resources.pcb.cadence.com/blog/2019-mii-and-rmii-routing-guidelines-for-ethernet
http://timing-diagrams.com/

Please close this issue if there is no ESP-IDF software related issue.

charon22, 2021-08-18:
Hi  @negativekelvin @kostaond ,
Thanks for your reply.
I thought that only those registered as UARTs(UART0,UART1,UART2) in the PIN map could be set as UARTs.
I tested by resetting the UART to a different port as per your advice and it worked without any problems. There was no need to use an Inverter either. 
Thank you for your advises.

Alvin1Zhang, 2021-08-19:
Thanks for reporting and sharing updates, feel free to reopen.

closed 2021-08-19 by Alvin1Zhang