espressif/esp-idf #7417: xTaskNotify hangs

KaeLL, 2021-08-13:
- IDF version: v4.4-dev-2594-ga20df743f

Title. Here's the task's core dump.
```cpp
#0  0x4000bff0 in _xtos_set_intlevel ()
#1  0x40095f85 in vPortExitCritical (mux=<optimized out>) at /esp-idf/components/freertos/port/xtensa/port.c:447
#2  0x4009562f in xTaskGenericNotify (xTaskToNotify=0x3ffddff4, ulValue=1, eAction=eSetValueWithOverwrite, pulPreviousNotificationValue=<optimized out>) at /esp-idf/components/freertos/tasks.c:5430
```
Btw, the function is called frequently, if that matters.
What are the possible causes?

Dazza0, 2021-09-07:
@KaeLL Could be a number of things, but its hard to tell without more information. Could you provide the following:

- Chip version
- What type of crash was occurring? Was it an `abort()`, Guru Meditation error etc. The crash log will be helpful.
- If task A is sending the notification, what is its priority (and affinity if applicable). Likewise, if a Task B is receiving the notification, what was its priority and affinity?

The core dump stops right at `_xtos_set_intlevel()` which is right when interrupts are being enabled again. So it's possible that the task that calls `xTaskGenericNotify()` is being preempted (e.g., by an ISR) and the actual hang is happening in the ISR itself.

AxelLin, 2021-11-15:
@KaeLL 
Any feedback regarding https://github.com/espressif/esp-idf/issues/7417#issuecomment-913984698

KaeLL, 2021-11-16:
@AxelLin no. I ended up abandoning the idea that led to this problem. Might as well just close the issue.

Alvin1Zhang, 2021-11-17:
@KaeLL Thanks for reporting, and thanks to @AxelLin  for the notice, feel free to reopen.

closed 2021-11-17 by Alvin1Zhang