espressif/esp-idf #7418: Support client key passwords for HTTPS connections

joncmaloney, 2021-08-14:
**Is your feature request related to a problem? Please describe.**

A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]

**Describe the solution you'd like**

A clear and concise description of what you want to happen.

**Describe alternatives you've considered**

A clear and concise description of any alternative solutions or features you've considered.

Please give as many details as you can. Include suggestions for useful APIs or interfaces if relevant.

**Additional context**

Add any other context or screenshots about the feature request here.

joncmaloney, 2021-08-14:
I don't know how to add a key into an esp32 bin image securely. We have a requirement that all secrets are encrypted at rest and in transit. If the key is not password protected it is in plain text and anyone with access to the bin image is able to view the key. Therefore the key is not encrypted at rest. If a build server was to secure the image by encrypting it needs to be decrypted to be transported via UART bootloader and hence not encrypted in transit. This is unless we use host generated key (not recommended). Further if host generated keys are use the keys need to be transmitted to ESP32 in plain text on first program. This is also doesn't meet our security requirements. 

We are currently using password protected keys then manually removing the password from these key using the mbedtls libraries.

negativekelvin, 2021-08-14:
How do you protect the passphrase?

joncmaloney, 2021-08-15:
It's not going to be cryptography secured. It can only be obfuscated.

closed 2021-08-29 by espressif-bot