espressif/esp-idf #7421: unable to reflash bootloader: secure boot check fail (no flash encryption enabled)

albrownwood, 2021-08-15:
Hello,  
I am unable to reflash the bootloader, here are the steps followed:  
-signing key has been burnt into the esp32 and I was able to load the initial signed app images correctly.  
-using esp-idf commit 166c30e7b2ed1dcaae56179329540a862915208a

I wanted to generate a new bootloader as follow:  
```
idf.py bootloader
...
v/esp-idf/components/partition_table/check_sizes.py --offset 0xc000 bootloader 0x1000 /Users/xxx/dev/esp-homekit-sdk-mfi/examples/irrigation/build/bootloader/bootloader.bin
Bootloader binary size 0x8b00 bytes. 0x2500 bytes (27%) free.

Bootloader build complete.
```
then flashing using the following:  
```
esptool.py --chip esp32 --port "$ESPPORT" --before=default_reset --after=no_reset write_flash --flash_mode dio --flash_freq 40m --flash_size 4MB 0x0 /Users/xxx/dev/esp-homekit-sdk-mfi/examples/irrigation/build/bootloader/bootloader-reflash-digest.bin
...
Flash will be erased from 0x00000000 to 0x00009fff...
Compressed 39680 bytes to 23068...
Wrote 39680 bytes (23068 compressed) at 0x00000000 in 2.5 seconds (effective 128.6 kbit/s)...
Hash of data verified.

Leaving...
Staying in bootloader.
```

when I go in monitor mode I see this message:  
```
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0038,len:9840
load:0x40078000,len:22128
load:0x40080400,len:3532
0x40080400: _init at ??:?

secure boot check fail
ets_main.c 371
```
The bootloader size is:
```
-rw-r--r--    1 xxx  staff   39680 Aug  3 20:39 bootloader-reflash-digest.bin
-rw-r--r--    1 xxx  staff   35584 Aug  3 20:39 bootloader.bin
-rwxr-xr-x    1 xxx  staff  942692 Aug  3 20:39 bootloader.elf
```

and the partition table offset has been set to 0xC000 which leaves 45K which should be sufficient space for the bootloader.  

partition table:  
```
# Name,         Type,   SubType,    Offset,    Size,    Flags
# Note: Firmware partition offset needs to be 64K aligned, initial 36K (9 sectors) are reserved for bootloader and partition table
sec_cert,       0x3F,      ,        0xd000,    0x3000,  ,  # Never mark this as an encrypted partition
nvs,            data,   nvs,        0x10000,   0x6000,
otadata,        data,   ota,        ,          0x2000,  encrypted
phy_init,       data,   phy,        ,          0x1000,
ota_0,          app,    ota_0,      0x20000,   1600K,
ota_1,          app,    ota_1,      ,          1600K,
factory_nvs,    data,   nvs,        0x340000,  0x6000
nvs_keys,       data,   nvs_keys,   0x346000,  0x30000,  encrypted
coredump,       data,   coredump,   ,          0x60000
```

the efuses:  
```
Detecting chip type... ESP32
espefuse.py v3.2-dev
EFUSE_NAME (Block)                       Description  = [Meaningful Value] [Readable/Writeable] (Hex Value)
----------------------------------------------------------------------------------------
Calibration fuses:
BLK3_PART_RESERVE (BLOCK0):              BLOCK3 partially served for ADC calibration data   = False R/W (0b0)
ADC_VREF (BLOCK0):                       Voltage reference calibration                      = 1093 R/W (0b10001)

Config fuses:
XPD_SDIO_FORCE (BLOCK0):                 Ignore MTDI pin (GPIO12) for VDD_SDIO on reset     = False R/W (0b0)
XPD_SDIO_REG (BLOCK0):                   If XPD_SDIO_FORCE, enable VDD_SDIO reg on reset    = False R/W (0b0)
XPD_SDIO_TIEH (BLOCK0):                  If XPD_SDIO_FORCE & XPD_SDIO_REG                   = 1.8V R/W (0b0)
CLK8M_FREQ (BLOCK0):                     8MHz clock freq override                           = 55 R/W (0x37)
SPI_PAD_CONFIG_CLK (BLOCK0):             Override SD_CLK pad (GPIO6/SPICLK)                 = 0 R/W (0b00000)
SPI_PAD_CONFIG_Q (BLOCK0):               Override SD_DATA_0 pad (GPIO7/SPIQ)                = 0 R/W (0b00000)
SPI_PAD_CONFIG_D (BLOCK0):               Override SD_DATA_1 pad (GPIO8/SPID)                = 0 R/W (0b00000)
SPI_PAD_CONFIG_HD (BLOCK0):              Override SD_DATA_2 pad (GPIO9/SPIHD)               = 0 R/W (0b00000)
SPI_PAD_CONFIG_CS0 (BLOCK0):             Override SD_CMD pad (GPIO11/SPICS0)                = 0 R/W (0b00000)
DISABLE_SDIO_HOST (BLOCK0):              Disable SDIO host                                  = False R/W (0b0)

Efuse fuses:
WR_DIS (BLOCK0):                         Efuse write disable mask                           = 256 R/W (0x0100)
RD_DIS (BLOCK0):                         Efuse read disable mask                            = 2 R/W (0x2)
CODING_SCHEME (BLOCK0):                  Efuse variable block length scheme
   = NONE (BLK1-3 len=256 bits) R/W (0b00)
KEY_STATUS (BLOCK0):                     Usage of efuse block 3 (reserved)                  = False R/W (0b0)

Identity fuses:
MAC (BLOCK0):                            Factory MAC Address
   = 9c:9c:1f:c9:65:a0 (CRC 0x2f OK) R/W
MAC_CRC (BLOCK0):                        CRC8 for factory MAC address                       = 47 R/W (0x2f)
CHIP_VER_REV1 (BLOCK0):                  Silicon Revision 1                                 = True R/W (0b1)
CHIP_VER_REV2 (BLOCK0):                  Silicon Revision 2                                 = False R/W (0b0)
CHIP_VERSION (BLOCK0):                   Reserved for future chip versions                  = 2 R/W (0b10)
CHIP_PACKAGE (BLOCK0):                   Chip package identifier                            = 0 R/W (0b000)
MAC_VERSION (BLOCK3):                    Version of the MAC field                           = 0 R/W (0x00)

Security fuses:
FLASH_CRYPT_CNT (BLOCK0):                Flash encryption mode counter                      = 0 R/W (0b0000000)
UART_DOWNLOAD_DIS (BLOCK0):              Disable UART download mode (ESP32 rev3 only)       = False R/W (0b0)
FLASH_CRYPT_CONFIG (BLOCK0):             Flash encryption config (key tweak bits)           = 0 R/W (0x0)
CONSOLE_DEBUG_DISABLE (BLOCK0):          Disable ROM BASIC interpreter fallback             = True R/W (0b1)
ABS_DONE_0 (BLOCK0):                     Secure boot V1 is enabled for bootloader image     = True R/W (0b1)
ABS_DONE_1 (BLOCK0):                     Secure boot V2 is enabled for bootloader image     = False R/W (0b0)
JTAG_DISABLE (BLOCK0):                   Disable JTAG                                       = True R/W (0b1)
DISABLE_DL_ENCRYPT (BLOCK0):             Disable flash encryption in UART bootloader        = False R/W (0b0)
DISABLE_DL_DECRYPT (BLOCK0):             Disable flash decryption in UART bootloader        = False R/W (0b0)
DISABLE_DL_CACHE (BLOCK0):               Disable flash cache in UART bootloader             = False R/W (0b0)
BLOCK1 (BLOCK1):                         Flash encryption key
   = 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 R/W
BLOCK2 (BLOCK2):                         Secure boot key
   = ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? -/-
BLOCK3 (BLOCK3):                         Variable Block 3
   = 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 R/W

Flash voltage (VDD_SDIO) determined by GPIO12 on reset (High for 1.8V, Low/NC for 3.3V).
```

Please let me know if you see what can be wrong here.

KonstantinKondrashov, 2021-08-16:
Hi @albrownwood!
Could you check the new bootloader image with the `espsecure.py  verify_signature` cmd?

albrownwood, 2021-08-16:
Hi @KonstantinKondrashov, thanks for taking a look, I get the following out put:

```
espsecure.py verify_signature --version 1 --keyfile ~/Documents/xxx/secure_boot_signing_key.pem build/bootloader/bootloader-reflash-digest.bin
espsecure.py v3.2-dev

A fatal error occurred: Signature block has version 16775494. This version  of espsecure only supports version 0.
```

KonstantinKondrashov, 2021-08-17:
Hi @albrownwood!
Sorry, I was wrong, you can use verify_signature  to verify a signature of app and data part but not bootloader.
`espsecure.py verify_signature --version 1 --keyfile secure_boot_signing_key_ecdsa.pem build/hello-world.bin `

"secure boot check fail" message comes from the ROM and indicates that the digest calculated for the bootloader is invalid, so it is refusing to continue.

I had a board with enabled secure boot V1. I am able to update the bootloader on this board using the same 166c30e version as you, it works. Are you sure that you did not miss the first step?
```
Bootloader built and secure digest generated.
Secure boot enabled, so bootloader not flashed automatically.
Burn secure boot key to efuse using:
       espefuse.py burn_key secure_boot_v1 /home/kostia/esp/hello_world/build/bootloader/secure-bootloader-key-256.bin
```
If you miss it then the first bootloader boot generates the random key and burns it to eFuse (in this case you do not have this key to re-sign the bootloader). This is a cause of why you get this error.

To update the bootloader you use the right cmd. `esptool.py ... write_flash ... 0x0 bootloader-reflash-digest.bin`

BTW, you do not need to define the offsets in the partition table (leave them empty), the partition tool will manage it using the partition table offset.

albrownwood, 2021-08-17:
Hi @KonstantinKondrashov,
I was pretty sure I did the burn_key operation on that particular chip but I am probably mistaken..  
I retried from a blank ESP32 and this time I am able to reflash the bootloader as expected, so the error was most likely on my side.
Thanks for helping with resolution and also for the  recommendation on offsets.

closed 2021-08-17 by albrownwood