espressif/esp-idf #7424: We get the same WiFi Mac on all our ESP32 chips

hassandraga, 2021-08-16:
## Environment

- Development Kit: -
- Kit version -
- Module or chip used:  [ESP32-WROOM-32U]
- IDF version v4.4-dev-1254-g639e7ad49
- Build System: [idf.py]
- Compiler version: xtensa-esp32-elf-gcc (crosstool-NG esp-2020r3) 8.4.0
- Operating System: [Windows/Linux]
- (Windows only) environment type: [Plain Command Prompt].
- Using an IDE?: [No]
- Power Supply: [24V DC]

## Problem Description

We get the same WiFi Mac address on all boards.

### Expected Behavior

Different WiFi Mac on each board.

### Actual Behavior

Same WiFi Mac (3FFBC37A) on each board.

### Steps to reproduce

1. Use the code below.

### Code to reproduce this issue

```cpp
uint8_t chipid[10] = {0};

esp_read_mac(chipid, ESP_MAC_WIFI_STA);
// Or
esp_efuse_mac_get_default(chipid);
// Or
esp_wifi_get_mac(ESP_IF_WIFI_STA, chipid);

printf("The Mac address is: %X", chipid);
```

## Debug Logs

```
The Mac address is: 3FFBC37A
```

**We use the same binary build on all our ESP32 boards, and all print the same MAC address, which it should be different.**

felmue, 2021-08-16:
Hello @hassandraga 

I think you aren't printing the full MAC as a MAC address has 6 bytes. Try something like this:
```
  uint8_t mac[6];

  ESP_ERROR_CHECK(esp_efuse_mac_get_default(mac));
  printf("%02X:%02X:%02X:%02X:%02X:%02X\n", mac[0], mac[1], mac[2], mac[3], mac[4], mac[5]);
```
Thanks
Felix

hassandraga, 2021-08-17:
Hi @felmue, Yeah I miss this one, Thank you.

closed 2021-08-17 by hassandraga

Alvin1Zhang, 2021-08-18:
@felmue Thanks for the notes.