espressif/esp-idf #7425: CSI capture on ESP-IDF v4.3 causes TG1WDT_SYS_RESET

Gi-z, 2021-08-16:
## Environment

- Development Kit:      none
- Module or chip used:  ESP32-WROOM-32D
- IDF version: v4.3
- Build System:         Make
- Compiler version: xtensa-esp32-elf-gcc (crosstool-NG esp-2020r3) 8.4.0
- Operating System:     macOS
- Using an IDE?: Yes, VSCode
- Power Supply:        USB

## Problem Description

Attempting to initiate CSI capture via `esp_wifi_set_csi(1)` will cause a TG1WDT_SYS_RESET to occur shortly after boot.
Have tested with both ESP32-WROOM-32D and WIFI Kit 32, both exhibiting the same behaviour. This problem does not occur when running ESP-IDF v4.0. 

Let me know if there's anything else I can provide to assist with identifying the issue. I did see some newer "wifi radar" functions used in the [esp-csi](https://github.com/espressif/esp-csi/tree/master/examples/console_test) examples, however I could not find anything about them in the ESP-IDF tree.

### Expected Behavior

CSI capture is initiated and packets are logged.

### Actual Behavior

Device continually reboots and crashes after a seemingly random interval.

### Code to reproduce this issue

https://gist.github.com/Gi-z/ead662e9e4c23adcb33fed2b51514deb

## Debug Logs

```
CSI captured with source MAC: AB:C3:88:71:80:0E
ets Jun  8 2016 00:22:57

rst:0x8 (TG1WDT_SYS_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:6952
load:0x40078000,len:14260
load:0x40080400,len:3676
entry 0x40080674
I (54) boot: ESP-IDF v4.3 2nd stage bootloader
I (54) boot: compile time 16:36:41
I (54) boot: chip revision: 1
I (57) boot_comm: chip revision: 1, min. bootloader chip revision: 0
I (64) boot.esp32: SPI Speed      : 40MHz
I (69) boot.esp32: SPI Mode       : DIO
I (73) boot.esp32: SPI Flash Size : 4MB
W (78) boot.esp32: PRO CPU has been reset by WDT.
W (83) boot.esp32: WDT reset info: PRO CPU PC=0x400803c0
0x400803c0: _DoubleExceptionVector at /Users/glennforbes/esp/esp-idf/esp-idf-v4.3/components/freertos/port/xtensa/xtensa_vectors.S:455

W (89) boot.esp32: WDT reset info: APP CPU PC=0x400803c0
0x400803c0: _DoubleExceptionVector at /Users/glennforbes/esp/esp-idf/esp-idf-v4.3/components/freertos/port/xtensa/xtensa_vectors.S:455
```

## Other items if possible

[csi_43_crash_demo.zip](https://github.com/espressif/esp-idf/files/6993759/csi_43_crash_demo.zip)

- [X] sdkconfig file (attach the sdkconfig file from your project folder)
- [X] elf file in the ``build`` folder (**note this may contain all the code details and symbols of your project.**)
- [ ] coredump (This provides stacks of tasks.)

Alvin1Zhang, 2021-08-17:
@Gi-z Thanks for reporting, we already had one fix, would you please help try https://github.com/espressif/esp-idf/commit/7e8afda8b23117cc6c5e96061ca0c67e181fc433 resolved your issue? Thanks.

Gi-z, 2021-08-17:
Thanks for the prompt reply. That commit works perfectly!

Alvin1Zhang, 2021-08-18:
@Gi-z Thanks for sharing the updates, and glad to hear it works perfectly. Feel free to reopen.

closed 2021-08-18 by Alvin1Zhang