espressif/esp-idf #7426: Destructive uart_get_buffered_data_len()

lcrocker, 2021-08-16:
## Environment

- Development Kit: [none]
- Module or chip used:  [ESP32-WROOM-32D]
- IDF version (run ``git describe --tags`` to find it): v4.4-dev-2184-g166c30e7b
- Build System:         [idf.py]
- Compiler version: xtensa-esp32-elf-gcc (crosstool-NG esp-2020r3) 8.4.0
- Operating System:     [Linux] // Ubuntu 20.04 LTS
- Using an IDE?: [No]
- Power Supply:  [External]

## Problem Description

My code initializes UART with:

        uart_config_t uconfig = {
                .baud_rate = 19200,
                .data_bits = UART_DATA_8_BITS,
                .stop_bits = UART_STOP_BITS_1,
                .flow_ctrl = UART_HW_FLOWCTRL_DISABLE
        };
        uart_param_config(this->id, &uconfig);
        uart_set_pin(this->id, 17, 16, -1, -1);
        uart_driver_install(this->id, 256, 256, 10, &this->uq, 0));
        uart_set_mode(this->id, UART_MODE_UART);

Later, after sending it a character from the attached device, I call

        uart_get_buffered_data_len(this->id, &avail);

and "avail" gets set to 1.

Now, before I call uart_read() or any other uart function, a second call
to uart_get_buffered_data_len() returns 0. 

### Expected Behavior

uart_get_buffered_data_len() should report the number of characters
available in the read buffer. Only actually reading them should reduce the
number (receiving more should increase it, up to the size of the buffer).

### Actual Behavior

Seems to be destructive: data disappears from buffer before being read.

dzarda, 2021-12-11:
Hi @lcrocker, I'm battling some reception UART issues. Your problem seems interesting, especially in that I'm having a hard time comprehending what could be causing the observed behavior. The function impl is as follows:

```c
esp_err_t uart_get_buffered_data_len(uart_port_t uart_num, size_t* size)
{
    UART_CHECK((uart_num < UART_NUM_MAX), "uart_num error", ESP_FAIL);
    UART_CHECK((p_uart_obj[uart_num]), "uart driver error", ESP_FAIL);
    *size = p_uart_obj[uart_num]->rx_buffered_len;
    return ESP_OK;
}
```

dzarda, 2021-12-11:
Could it be there's another task reading that data?