espressif/esp-idf #7428: Help need in How to  use secure boot firmware

MarkEvens, 2021-08-17:
## Environment

- Development Kit:      [ESP32-DevKitC]
- Kit version (DevKitC): [v1|]
- Module or chip used:  [ESP32-WROOM-32]
- IDF version (run ``git describe --tags`` to find it):
    // v4.2
- Build System:         [Make]
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it):
    //xtensa-esp32-elf-gcc (crosstool-NG esp-2020r3) 8.4.0
- Operating System:     [Linux]
- Using an IDE?: [Yes (VScode)]
- Power Supply:         [USB]

## Problem Description
What I did to make secure.
I take hello world example for use and type command "make menuconfig" set as below screenshot
[image]

After this setting I saved it and closed menuconfig.
After that generate .pem file using this command "espsecure.py generate_signing_key"
and enter command "bootloader"
after it flash that bootloader to device using suggested command that show in above execution.
and last I run "make flash"
and run "make monitor"
 
//Detailed problem description goes here.

### Actual Behavior
I am geting this on monitor
```
--- idf_monitor on /dev/ttyUSB0 115200 ---
--- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:4
load:0x3fff0034,len:9436
load:0x40078000,len:20456
load:0xffffffff,len:-1
ets Jun  8 2016 00:22:57

rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:4
load:0x3fff0034,len:9436
load:0x40078000,len:20456
load:0xffffffff,len:-1
ets Jun  8 2016 00:22:57

rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:4
load:0x3fff0034,len:9436
load:0x40078000,len:20456
load:0xffffffff,len:-1
ets Jun  8 2016 00:22:57

rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:4
[... 1181 lines cut ...]
# CONFIG_TCP_OVERSIZE_DISABLE is not set
CONFIG_UDP_RECVMBOX_SIZE=6
CONFIG_TCPIP_TASK_STACK_SIZE=3072
CONFIG_TCPIP_TASK_AFFINITY_NO_AFFINITY=y
# CONFIG_TCPIP_TASK_AFFINITY_CPU0 is not set
# CONFIG_TCPIP_TASK_AFFINITY_CPU1 is not set
CONFIG_TCPIP_TASK_AFFINITY=0x7FFFFFFF
# CONFIG_PPP_SUPPORT is not set
CONFIG_ESP32_PTHREAD_TASK_PRIO_DEFAULT=5
CONFIG_ESP32_PTHREAD_TASK_STACK_SIZE_DEFAULT=3072
CONFIG_ESP32_PTHREAD_STACK_MIN=768
CONFIG_ESP32_DEFAULT_PTHREAD_CORE_NO_AFFINITY=y
# CONFIG_ESP32_DEFAULT_PTHREAD_CORE_0 is not set
# CONFIG_ESP32_DEFAULT_PTHREAD_CORE_1 is not set
CONFIG_ESP32_PTHREAD_TASK_CORE_DEFAULT=-1
CONFIG_ESP32_PTHREAD_TASK_NAME_DEFAULT="pthread"
CONFIG_SPI_FLASH_WRITING_DANGEROUS_REGIONS_ABORTS=y
# CONFIG_SPI_FLASH_WRITING_DANGEROUS_REGIONS_FAILS is not set
# CONFIG_SPI_FLASH_WRITING_DANGEROUS_REGIONS_ALLOWED is not set
CONFIG_SUPPRESS_SELECT_DEBUG_OUTPUT=y
CONFIG_SUPPORT_TERMIOS=y
CONFIG_SEMIHOSTFS_MAX_MOUNT_POINTS=1
CONFIG_SEMIHOSTFS_HOST_PATH_MAX_LEN=128
# End of deprecated options
```

formus14, 2021-08-17:
what chip revision are you using .. because that will define what secure boot revision you should use first!

```
esp_chip_info_t chip_info;
esp_chip_info(*chip_info);
print("Chip rev %d\n",chip_info->revision );
```
watch out once secure boot is booted, efuse gets burnt and you won't be able to reflash your chip over JTAG neither USB!

MarkEvens, 2021-08-17:
@formus14  my chip revision is 
```
Chip is ESP32-D0WDQ6 (revision 1)
```

formus14, 2021-08-17:
Good, then first Choose Revision 1 in your SDK config ;) 
`
CONFIG_ESP32_REV_MIN_1=y
CONFIG_ESP32_REV_MIN=1`

That means you have to use Secure boot rev1 as you have configured already.

I will normally use a bit different setting for generating a key and signing though:

- disable this option `CONFIG_SECURE_BOOT_BUILD_SIGNED_BINARIES=`
- espsecure generate_signing_key --version 1 signing_key_v1.pem
- espsecure.py extract_public_key --keyfile signing_key_v1.pem verification_key_v1.pem
- add a relative path to the generated key here , for example `CONFIG_SECURE_BOOT_SIGNING_KEY="../verification_key_v1.pem"`
- Build the project
- extract the built bins (forexample application.bin and bootloader.bin) and sign each using the script below
```
espsecure.py sign_data --keyfile signing_key_v1--output signed_app.bin app.bin
espsecure.py sign_data --keyfile signing_key_v1--output signed_bootloader.bin bootloader.bin
```
- Flash the signed applications using the script below corresponds to your partition table (notsure your flash size is really 1mb but check it again)
`esptool.py --chip esp32 -p COM20 --before=default_reset --after=no_reset write_flash --flash_mode dio --flash_freq 40m --flash_size 1MB 0x1000 signed_bootloader.bin 0x8000 partition_table.bin 0xd000 ota_data_initial.bin 0x10000 signed_app.bin
`
- esp will bootup , burn the efuse and you got your secured boot chip (again not reflashable)

negativekelvin, 2021-08-17:
https://docs.espressif.com/projects/esp-idf/en/release-v4.2/esp32/security/secure-boot-bootloader-size.html#secure-boot-bootloader-size

MarkEvens, 2021-08-18:
@negativekelvin  My partiotion offset is already at 0xa000 and my bootloader.bin file sise is "34,544 bytes"

negativekelvin, 2021-08-18:
Not according to your sdkconfig `CONFIG_PARTITION_TABLE_OFFSET=0x8000`

MarkEvens, 2021-08-18:
This is new sdkconfig but still not working 
```
#
# Automatically generated file. DO NOT EDIT.
# Espressif IoT Development Framework (ESP-IDF) Project Configuration
#
CONFIG_IDF_TARGET="esp32"
CONFIG_IDF_TARGET_ESP32=y
CONFIG_IDF_FIRMWARE_CHIP_ID=0x0000

#
# SDK tool configuration
#
CONFIG_SDK_TOOLPREFIX="xtensa-esp32-elf-"
CONFIG_SDK_PYTHON="python"
CONFIG_SDK_MAKE_WARN_UNDEFINED_VARIABLES=y
# CONFIG_SDK_TOOLCHAIN_SUPPORTS_TIME_WIDE_64_BITS is not set
# end of SDK tool configuration

#
# Build type
#
CONFIG_APP_BUILD_TYPE_APP_2NDBOOT=y
# CONFIG_APP_BUILD_TYPE_ELF_RAM is not set
CONFIG_APP_BUILD_GENERATE_BINARIES=y
CONFIG_APP_BUILD_BOOTLOADER=y
CONFIG_APP_BUILD_USE_FLASH_SECTIONS=y
# end of Build type

#
# Application manager
#
CONFIG_APP_COMPILE_TIME_DATE=y
# CONFIG_APP_EXCLUDE_PROJECT_VER_VAR is not set
# CONFIG_APP_EXCLUDE_PROJECT_NAME_VAR is not set
# CONFIG_APP_PROJECT_VER_FROM_CONFIG is not set
CONFIG_APP_RETRIEVE_LEN_ELF_SHA=16
# end of Application manager

#
# Bootloader config
#
CONFIG_BOOTLOADER_COMPILER_OPTIMIZATION_SIZE=y
# CONFIG_BOOTLOADER_COMPILER_OPTIMIZATION_DEBUG is not set
# CONFIG_BOOTLOADER_COMPILER_OPTIMIZATION_PERF is not set
# CONFIG_BOOTLOADER_COMPILER_OPTIMIZATION_NONE is not set
# CONFIG_BOOTLOADER_LOG_LEVEL_NONE is not set
# CONFIG_BOOTLOADER_LOG_LEVEL_ERROR is not set
# CONFIG_BOOTLOADER_LOG_LEVEL_WARN is not set
CONFIG_BOOTLOADER_LOG_LEVEL_INFO=y
# CONFIG_BOOTLOADER_LOG_LEVEL_DEBUG is not set
# CONFIG_BOOTLOADER_LOG_LEVEL_VERBOSE is not set
CONFIG_BOOTLOADER_LOG_LEVEL=3
# CONFIG_BOOTLOADER_VDDSDIO_BOOST_1_8V is not set
CONFIG_BOOTLOADER_VDDSDIO_BOOST_1_9V=y
# CONFIG_BOOTLOADER_FACTORY_RESET is not set
# CONFIG_BOOTLOADER_APP_TEST is not set
CONFIG_BOOTLOADER_WDT_ENABLE=y
# CONFIG_BOOTLOADER_WDT_DISABLE_IN_USER_CODE is not set
CONFIG_BOOTLOADER_WDT_TIME_MS=9000
# CONFIG_BOOTLOADER_APP_ROLLBACK_ENABLE is not set
CONFIG_BOOTLOADER_RESERVE_RTC_SIZE=0
# CONFIG_BOOTLOADER_CUSTOM_RESERVE_RTC is not set
# end of Bootloader config

#
# Security features
#
CONFIG_SECURE_SIGNED_ON_BOOT=y
CONFIG_SECURE_SIGNED_ON_UPDATE=y
[... 1104 lines cut ...]
CONFIG_ESP32_PTHREAD_STACK_MIN=768
CONFIG_ESP32_DEFAULT_PTHREAD_CORE_NO_AFFINITY=y
# CONFIG_ESP32_DEFAULT_PTHREAD_CORE_0 is not set
# CONFIG_ESP32_DEFAULT_PTHREAD_CORE_1 is not set
CONFIG_ESP32_PTHREAD_TASK_CORE_DEFAULT=-1
CONFIG_ESP32_PTHREAD_TASK_NAME_DEFAULT="pthread"
CONFIG_SPI_FLASH_WRITING_DANGEROUS_REGIONS_ABORTS=y
# CONFIG_SPI_FLASH_WRITING_DANGEROUS_REGIONS_FAILS is not set
# CONFIG_SPI_FLASH_WRITING_DANGEROUS_REGIONS_ALLOWED is not set
CONFIG_SUPPRESS_SELECT_DEBUG_OUTPUT=y
CONFIG_SUPPORT_TERMIOS=y
CONFIG_SEMIHOSTFS_MAX_MOUNT_POINTS=1
CONFIG_SEMIHOSTFS_HOST_PATH_MAX_LEN=128
# End of deprecated options
```
and this is partition.csv
```
# Name,   Type, SubType, Offset,  Size, Flags
# Note: if you change the phy_init or app partition offset, make sure to change the offset in Kconfig.projbuild

nvs,      data, nvs,     0xa000,  0x3000
otadata,  data, ota,     0xd000,  0x2000
phy_init, data, phy,     0xf000,  0x1000
ota_0,  app, ota_0,      0x10000, 256K
```

MarkEvens, 2021-08-18:
@negativekelvin  Issue is of bootloader size fixed by changing offset at partion.csv to 0xb000 and in sdkconfig to 0xa000 
Thank you

mahavirj, 2021-08-18:
@MarkEvens 

Alternatively you can remove offset field from partition table as shown at:

https://github.com/espressif/esp-idf/blob/a20df743f1c51e6d65b021ed2ffd3081a2feec64/examples/security/flash_encryption/partitions_example.csv#L2

This will allow offset to be auto adjusted based on partition table offset in sdkconfig, more information captured at https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/partition-tables.html#offset-size

Please feel free to close the issue if your problem is solved

closed 2021-08-19 by MarkEvens