espressif/esp-idf #7430: HCI UART is not working at all - lack of documentation and information

danergo, 2021-08-17:
## Environment

- Development Kit: NodeMCU ESP-32S
- Module or chip used:  ESP32-WROOM-32
- IDF version: v4.4-dev-2533-g83956ebbae
- Build System: idf.py
- Compiler version: xtensa-esp32-elf-gcc (crosstool-NG esp-2021r1) 8.4.0
- Operating System: Linux
- Using an IDE?: No
- Power Supply: USB

## Problem Description

I want to use ESP32 as a Bluetooth Host Controller via HCI (over UART) to provide bluetooth functionality to my Linux PC.
I need to use BlueZ tools, like `btattach`, `bluetootctl`, and other older ones like `hcitool` and `hciconfig`.

I've cloned the IDF, and compiled and flashed https://github.com/espressif/esp-idf/tree/master/examples/bluetooth/hci/controller_hci_uart example.

I'm using an FTDI232 USB-UART module to provide another UART to my computer (besides NodeMCU's default one on /dev/ttyUSB0).

I've wired NodeMCU and FTDI232 as follows:

ESP32 UART 1 <-----> FTDI232 (/dev/ttyUSB1)
    TX 5 <-----> RX
    RX 18 <-----> TX
    RTS 19 <-----> CTS
    CTS 23 <-----> RTS

I attached the device:
    
    btattach -B /dev/ttyUSB1 -S 921600 --protocol h4
    Attaching Primary controller to /dev/ttyUSB1
    Switched line discipline from 0 to 15
    Device index 0 attached

Then I tried to bring up the hci0 interface:

```
bluetoothctl 
Agent registered
[bluetooth]# devices 
No default controller available
[bluetooth]# power on
No default controller available
[bluetooth]# 
```

### Expected Behavior

I need to have a controller which can be turned on, and can be used over HCI for Bluetooth operations.
(Preferrably both BLE and Classic BT).

### Actual Behavior

See above.

### Steps to reproduce

See above.

Also, readme on this example is pretty short providing almost no information on what exactly is this example for:

> This is a btdm controller use UART as HCI IO. This require the UART device support RTS/CTS mandatory. It can do the configuration of UART number and UART baudrate by menuconfig.

Nice, and how shall I do the wiring? How shall I debug in case something is not working? 

Please guys provide me more information on this topic.

danergo, 2021-08-17:
In spite of I set the UART to be used as 1 in `menuconfig` it seems it's not being considered by nice `idf.py build`...

danergo, 2021-08-17:
I have partially succeed, but I needed to manually add `#define CONFIG_BT_HCI_UART_NO 1` to top of `main.c` to make it work.

And actually `hcitool lescan` is not working, also `l2ping BT_ADDR` is not working. They are timing out.

danergo, 2021-08-18:
So guys, I know you are busy, but here are my results:

After manually adding that define, it started working. This is a proof that in case of this example, menuconfig is not working as it should. (It's not creating the right sdkconfig file).

This shall be fixed on your side, otherwise the example is not easy to use, especially for someone who is not that experienced (yet) in your software.

closed 2021-08-18 by danergo

xulongzhuang, 2021-08-19:
Hi, @danergo , Thank you for reporting this issue. We have optimized this issue but have not merged yet. It will be merged into the master branch soon. If `hcitool lescan` cannot be used but other commands such as `hcitool inq` can be used, it may be due to bluez installation and configuration problems. The `hcitool lescan` I tested can be used normally on my machine using this demo.

danergo, 2021-08-21:
Hi, @xulongzhuang: all `hcitool` commands are working now as intended.