espressif/esp-idf #7432: Bluetooth controller: HCI_Read_Clock failed read local Bluetooth Clock.

andryblack, 2021-08-17:
## Environment

- Development Kit:      [ESP32-Wrover-Kit]
- Kit version: [v1]
- Module or chip used:  [ESP32-WROOM-32]
- IDF version (run ``git describe --tags`` to find it):
    // v4.3-dev-5661-ga20df743f
- Build System:         [idf.py]
- Compiler version:
    // xtensa-esp32-elf-gcc (crosstool-NG esp-2021r1) 8.4.0
- Operating System:     [macOS]
- Using an IDE?: [No]
- Power Supply:         [USB]

## Problem Description

//Detailed problem description goes here.
After sending HCI_Read_Clock command with parameters Which_Clock = 0 and Connection_Handle = 0. Controller emit Command Complete event with status = 0x02  

### Expected Behavior
As described at 7.5.6 BLUETOOTH SPECIFICATION Version 4.2 [Vol 2, Part E]:
If the Which_Clock value is 0, then the Connection_Handle shall be ignored, the local Bluetooth Clock value shall be returned and the accuracy parameter shall be set to 0.

### Actual Behavior

Invalid status reported

### Steps to reproduce

1. build
2. flash
3. monitor

### Code to reproduce this issue

https://gist.github.com/andryblack/74ab2040ad73daf46ad17855ddb1801b

## Debug Logs

```
I (27) boot: ESP-IDF v4.3-dev-5661-ga20df743f 2nd stage bootloader
I (27) boot: compile time 00:11:41
I (27) boot: chip revision: 1
I (31) boot_comm: chip revision: 1, min. bootloader chip revision: 0
I (38) boot.esp32: SPI Speed      : 40MHz
I (43) boot.esp32: SPI Mode       : DIO
I (47) boot.esp32: SPI Flash Size : 2MB
I (52) boot: Enabling RNG early entropy source...
I (57) boot: Partition Table:
I (61) boot: ## Label            Usage          Type ST Offset   Length
I (68) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (76) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (83) boot:  2 factory          factory app      00 00 00010000 00100000
I (91) boot: End of partition table
I (95) boot_comm: chip revision: 1, min. application chip revision: 0
I (102) esp_image: segment 0: paddr=00010020 vaddr=3f400020 size=0a564h ( 42340) map
I (126) esp_image: segment 1: paddr=0001a58c vaddr=3ffbdb60 size=03450h ( 13392) load
I (131) esp_image: segment 2: paddr=0001d9e4 vaddr=40080000 size=02634h (  9780) load
I (136) esp_image: segment 3: paddr=00020020 vaddr=400d0020 size=349f0h (215536) map
I (218) esp_image: segment 4: paddr=00054a18 vaddr=40082634 size=13bb0h ( 80816) load
I (251) esp_image: segment 5: paddr=000685d0 vaddr=50000000 size=00010h (    16) load
I (262) boot: Loaded app from partition at offset 0x10000
I (263) boot: Disabling RNG early entropy source...
I (274) cpu_start: Pro cpu up.
I (274) cpu_start: Starting app cpu, entry point is 0x400810e4
0x400810e4: call_start_cpu1 at /esp32/esp-idf/components/esp_system/port/cpu_start.c:156

I (0) cpu_start: App cpu up.
I (288) cpu_start: Pro cpu start user code
I (288) cpu_start: cpu freq: 160000000
I (288) cpu_start: Application information:
I (293) cpu_start: Project name:     ble_adv
I (298) cpu_start: App version:      v1.4.1-9-g51ddac10e-dirty
I (304) cpu_start: Compile time:     Aug 18 2021 00:11:33
I (310) cpu_start: ELF file SHA256:  2d006dfd0189f39e...
I (316) cpu_start: ESP-IDF:          v4.3-dev-5661-ga20df743f
I (323) heap_init: Initializing. RAM available for dynamic allocation:
I (330) heap_init: At 3FFAFF10 len 000000F0 (0 KiB): DRAM
I (336) heap_init: At 3FFB6388 len 00001C78 (7 KiB): DRAM
I (342) heap_init: At 3FFB9A20 len 00004108 (16 KiB): DRAM
I (348) heap_init: At 3FFC26A8 len 0001D958 (118 KiB): DRAM
I (355) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (361) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (367) heap_init: At 400961E4 len 00009E1C (39 KiB): IRAM
I (375) spi_flash: detected chip: generic
I (378) spi_flash: flash io: dio
W (382) spi_flash: Detected size(4096k) larger than the size in the binary image header(2048k). Using the size in the binary image header.
I (397) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (490) BTDM_INIT: BT controller compile version [bfbbe1e]
I (500) system_api: Base MAC address is not set
I (500) system_api: read default base MAC address from EFUSE
I (500) phy_init: phy_version 4670,719f9f6,Feb 18 2021,17:07:07
I (1090) TEST: task start
I (2090) TEST: hci_cmd_send_reset
I (2090) TEST: flag_send_avail: 1, cmd_sent: 1

I (2090) TEST: host rcv pkt: 
I (2090) TEST: 04 0e 04 05 03 0c 00 
I (2090) TEST: controller rcv pkt ready

I (3090) TEST: hci_cmd_send_read_clock
I (3090) TEST: host rcv pkt: 
I (3090) TEST: 04 0e 0c 05 07 14 02 00 00 00 00 00 00 00 00 
I (3090) TEST: controller rcv pkt ready

I (3090) TEST: flag_send_avail: 1, cmd_sent: 2
```

## Other items if possible

[sdkconfig.txt](https://github.com/espressif/esp-idf/files/7003104/sdkconfig.txt)

BetterJincheng, 2021-09-16:
Hi @andryblack ,

The attachment is the fix lib. You can just replace the `esp-idf/components/bt/controller/lib_esp32/esp32/libbtdm_app.a` with the attachment (extract first). Turn to us anytime.

Thanks!

[libbtdm_app.zip](https://github.com/espressif/esp-idf/files/7176377/libbtdm_app.zip)

Alvin1Zhang, 2021-11-23:
Thanks for reporting, fix on master is available at https://github.com/espressif/esp-idf/commit/534cd5bb3100d64016970f9bd916565d42270612, we are also back porting this fix to release/4.3, will update once fix on release/4.3 is available, thanks.

Alvin1Zhang, 2021-11-25:
Thanks for reporting, fix on release/4.3 is available https://github.com/espressif/esp-idf/commit/0c8fb540b2d34adcde81cb8f6c262c788daae0d6, feel free to reopen if the issue still happens, thanks.

closed 2021-11-25 by Alvin1Zhang