espressif/esp-idf #7433: ESP BLE GATT Server adding multiple characteristics

krupis, 2021-08-18:
Hello. Is it possible to update the BLE GATT server example code to show how to add multiple characteristics on a single service? Current example code creates 2 services but only 1 characteristics each. I have found many other people having issues figuring out what is the esp-idf intended way of adding multiple characteristics. Some people suggested various workarounds but I havent found a single clean method yet.

The characteristic is being added with this code:

```
        ESP_LOGI(GATTS_TAG, "CREATE_SERVICE_EVT, status %d,  service_handle %d\n", param->create.status, param->create.service_handle);
        gl_profile_tab[PROFILE_A_APP_ID].service_handle = param->create.service_handle;
        gl_profile_tab[PROFILE_A_APP_ID].char_uuid.len = ESP_UUID_LEN_16;
        gl_profile_tab[PROFILE_A_APP_ID].char_uuid.uuid.uuid16 = GATTS_CHAR1_UUID;
        printf("esp_ble_gatts_start_service \n");
        esp_ble_gatts_start_service(gl_profile_tab[PROFILE_A_APP_ID].service_handle);
        a_property = ESP_GATT_CHAR_PROP_BIT_READ | ESP_GATT_CHAR_PROP_BIT_NOTIFY | ESP_GATT_CHAR_PROP_BIT_WRITE;
        esp_err_t add_char_ret = esp_ble_gatts_add_char(gl_profile_tab[PROFILE_A_APP_ID].service_handle, &gl_profile_tab[PROFILE_A_APP_ID].char_uuid,
                                                        ESP_GATT_PERM_READ | ESP_GATT_PERM_WRITE,
                                                        a_property,
                                                        &gatts_demo_char1_val, NULL);
        if (add_char_ret){
            ESP_LOGE(GATTS_TAG, "add char failed, error code =%x",add_char_ret);
        }
 ```

Unfortunately, dublicating this block of code and changing a few parameters (property,UUID..) did not work. 

What I have found out so far :

1. I am currently analysing the suggested example(gatts table create demo) from the following discussion:
https://www.esp32.com/viewtopic.php?t=2877

I am trying to understand what are the differenes between this and GATT_Server example code. In the GATT_Server example, the characteristics are being created during ESP_GATTS_CREATE_EVT event. This event does not even exist in gatts table create demo
example code. Could someone help me understand?

2. When creating a service using esp_ble_gatts_create_service. Someone suggested to increase num_handle parameter number. Is that true? reference: https://github.com/espressif/esp-idf/issues/280

xiewenxiang, 2021-08-18:
@krupis 

I would recommend that you use tables to creat services, it looks cleaner and it's easy to extend.

you can refer this example [gatt_server_service_table](https://github.com/espressif/esp-idf/tree/master/examples/bluetooth/bluedroid/ble/gatt_server_service_table)

krupis, 2021-08-19:
@xiewenxiang  thanks for the response. I used gatt_server_service_table and managed to create multiple characteristics quite easy. Do you know why they show 2 different methods of creating characteristics and services? In GATT_Server example and gatt_server_serivce_table ?

xiewenxiang, 2021-09-02:
@krupis 

The GATT_SERVER example was used the legacy method to create gatt service. It is more cumbersome, and not easy to modify.

Then we improved to use the table to create the service.

Alvin1Zhang, 2021-11-23:
Thanks for reporting, feel free to reopen.

closed 2021-11-23 by Alvin1Zhang

mngr0, 2022-07-30:
hello,
I want to declare a service with more than 256 attributes.
I have been told to follow the gatt_server example