espressif/esp-idf #7434: Failing to Assign IP when joining AP - HomeKit (IDFGH-5687)

codspeed, 2021-08-18:
## Environment

- Module or chip used:  ESP32-WROOM-32D
- IDF version (run ``git describe --tags`` to find it): v4.3
- Build System:         idf.py
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it): xtensa-esp32-elf-gcc (crosstool-NG esp-2020r3) 8.4.0
- Operating System:    macOS
- Using an IDE?: VCode
- Power Supply:         USB/external 3.3V

## Problem Description

For our RM + HK setup and custom application based of HomeKit_Switch
On many occasions we are finding that when we join the device in WAC mode using the Home app that it does not assign an IP or Start the Pair Setup. Very often on the first attempt after a reboot.
Please see example where it took 4 attempts to successfully initiate.

### Expected Behavior
IP Assigned and device begins the Pair Setup immediately.

### Actual Behavior
No IP Assigned or Paring Started

### Steps to reproduce

1. Device should be in WAC mode
2. Home App -> Add Accessory -> Tap the Device

## Debug Logs

```
I (1823) EVENT_HANDLER: HAP_WAC_EVENT_REQ_SOFTAP_START
I (1823) ap_prcs: isWAC: 0
I (1823) ap_prcs: Attempt to set state(1): 0x01
I (1833) ap_prcs: Attempt to set state(2): 0x01
I (1843) ap_prcs: SetStateRegister 1
I (7853) wifi:new:<1,0>, old:<1,1>, ap:<1,1>, sta:<0,0>, prof:1
I (7863) wifi:station: 8e:43:3c:7a:7c:59 join, AID=1, bgn, 20
I (22233) wifi:station: 8e:43:3c:7a:7c:59 leave, AID = 1, bss_flags is 134259, bss:0x3ffe15d0
I (22233) wifi:new:<1,0>, old:<1,0>, ap:<1,1>, sta:<0,0>, prof:1
I (30593) wifi:new:<1,0>, old:<1,0>, ap:<1,1>, sta:<0,0>, prof:1
I (30593) wifi:station: 8e:43:3c:7a:7c:59 join, AID=1, bgn, 20
W (31243) wifi:Haven't to connect to a suitable AP now!
I (47763) wifi:station: 8e:43:3c:7a:7c:59 leave, AID = 1, bss_flags is 134259, bss:0x3ffe15d0
I (47763) wifi:new:<1,0>, old:<1,0>, ap:<1,1>, sta:<0,0>, prof:1
W (61243) wifi:Haven't to connect to a suitable AP now!
I (72353) wifi:new:<1,0>, old:<1,0>, ap:<1,1>, sta:<0,0>, prof:1
I (72353) wifi:station: 8e:43:3c:7a:7c:59 join, AID=1, bgn, 20
I (88523) wifi:station: 8e:43:3c:7a:7c:59 leave, AID = 1, bss_flags is 134259, bss:0x3ffe15d0
I (88523) wifi:new:<1,0>, old:<1,0>, ap:<1,1>, sta:<0,0>, prof:1
W (91243) wifi:Haven't to connect to a suitable AP now!
I (99573) wifi:new:<1,0>, old:<1,0>, ap:<1,1>, sta:<0,0>, prof:1
I (99583) wifi:station: 8e:43:3c:7a:7c:59 join, AID=1, bgn, 20
I (102363) esp_netif_lwip: DHCP server assigned IP to a station, IP is: 192.168.4.2
I (102363) EVENT_HANDLER: IP_EVENT_AP_STAIPASSIGNED
W (104253) wifi:<ba-add>idx:4 (ifx:1, 8e:43:3c:7a:7c:59), tid:0, ssn:0, winSize:64
######## Starting Pair Setup ########
Pair Setup M1 Received
Pair Setup M2 Successful
I (104763) EVENT_HANDLER: Pairing Started
```

Thanks.

myhomesense, 2021-08-29:
> The log in the middle shows how many times the AP failed to join
Correct, this is multiple attempts of trying, then disconnecting and trying again manually each time.

> From the log, there is no reason. It may be necessary to capture packets to analyze this problem.
How do you suggest we do this exactly? I will have it done and send the results.

> In addition, can I reproduce this problem locally?
I am unsure, we would need to send you the binaries of the project.

codspeed, 2021-08-29:
> The log in the middle shows how many times the AP failed to join

Correct, this is multiple attempts of trying, then disconnecting and trying again manually each time.

> From the log, there is no reason. It may be necessary to capture packets to analyze this problem.

How do you suggest we do this exactly? I will have it done and send the results.

> In addition, can I reproduce this problem locally?

I am unsure, we would need to send you the binaries of the project.

xueyunfei998, 2021-08-31:
hi @codspeed 
Can you provide a code to reproduce the problem? I want to reproduce the problem locally

xueyunfei998, 2021-11-12:
hi @codspeed 
Does this problem still exist? If this problem still exists, please provide the next recurring demo and idf commit iD and router model.

closed 2021-11-25 by Alvin1Zhang