espressif/esp-idf #7438: Time of execution NVS init functions with Power Management config

Batov, 2021-08-18:
## Environment

- Development Kit: Custom board with ESP32-D0WDQ6-V3
- IDF version: v4.1.1
- Build System: idf.py
- Compiler version : 8.4.0
- Operating System: Windows
- (Windows only) environment type: ESP Command Prompt
- Using an IDE?: Yes - VS Code
- Power Supply: Battery

## Problem Description
Strange behavior with time of execution NVS init functions with Power Management config.
71650 ms when CONFIG_PM_ENABLE  vs 2310 ms without CONFIG_PM_ENABLE?! Is it ok?
`log` is custom partition with 3M size.

### Code to reproduce this issue
```
static const char *TAG = "main";

void init_nvs(char *from)
{
    TickType_t init_start_ticks = xTaskGetTickCount();

    esp_err_t err = nvs_flash_init();
    ESP_LOGI(from, "nvs_flash_init: 0x%x, %d ms", err, pdTICKS_TO_MS(xTaskGetTickCount() - init_start_ticks));

    err = nvs_flash_init_partition("log");
    ESP_LOGI(from, "nvs_flash_init_partition: 0x%x, %d ms", err, pdTICKS_TO_MS(xTaskGetTickCount() - init_start_ticks));
}

void thread(void *arg)
{
    init_nvs("from_thread");
    vTaskDelete(NULL);
}

void app_main(void)
{
    #ifdef CONFIG_PM_ENABLE
    ESP_LOGI(TAG, "CONFIG_PM_ENABLE");
    #else
    ESP_LOGI(TAG, "NO CONFIG_PM_ENABLE");
    #endif

    xTaskCreate(thread, "NVS_INIT", 1024*32, NULL, 2, NULL);
}
```

## Debug Logs

With CONFIG_PM_ENABLE at sdkconfig
```
I (0) cpu_start: Starting scheduler on APP CPU.
I (342) main: CONFIG_PM_ENABLE
I (2872) from_thread: nvs_flash_init: 0x0, 2530 ms
I (71992) from_thread: nvs_flash_init_partition: 0x0, 71650 ms
```

Without CONFIG_PM_ENABLE at sdkconfig
```
I (0) cpu_start: Starting scheduler on APP CPU.
I (342) main: NO CONFIG_PM_ENABLE
I (422) from_thread: nvs_flash_init: 0x0, 80 ms
I (2652) from_thread: nvs_flash_init_partition: 0x0, 2310 ms
```

[sdkconfig.txt](https://github.com/espressif/esp-idf/files/7007590/sdkconfig.txt)
[partitions.csv](https://github.com/espressif/esp-idf/files/7007612/partitions.csv)

0xjakob, 2021-08-19:
@Batov Thanks for reporting this. Well have a look. We faced a similar issue before, maybe both can be solved jointly.

Batov, 2021-08-19:
FYI.

Time of execution NVS init functions at `app_main` task with Power Management config was similar of time from custom thread without Power Management config

```
#include "esp_log.h"
#include "esp_err.h"
#include "sdkconfig.h"
#include "nvs_flash.h"
#include "freertos/FreeRTOS.h"
#include "freertos/task.h"

static const char *TAG = "main";

void init_nvs(char *from)
{
    TickType_t init_start_ticks = xTaskGetTickCount();

    esp_err_t err = nvs_flash_init();
    ESP_LOGI(from, "nvs_flash_init: 0x%x, %d ms", err, pdTICKS_TO_MS(xTaskGetTickCount() - init_start_ticks));

    err = nvs_flash_init_partition("log");
    ESP_LOGI(from, "nvs_flash_init_partition: 0x%x, %d ms", err, pdTICKS_TO_MS(xTaskGetTickCount() - init_start_ticks));
}

void app_main(void)
{
    #ifdef CONFIG_PM_ENABLE
    ESP_LOGI(TAG, "CONFIG_PM_ENABLE");
    #else
    ESP_LOGI(TAG, "NO CONFIG_PM_ENABLE");
    #endif

    init_nvs("from_app_main");
}
```

Result:
```
I (0) cpu_start: Starting scheduler on APP CPU.
I (342) main: CONFIG_PM_ENABLE
I (422) from_app_main: nvs_flash_init: 0x0, 80 ms
I (2852) from_app_main: nvs_flash_init_partition: 0x0, 2510 ms
```

Batov, 2021-08-19:
Pin thread to core fix my issue.

```
#include "esp_log.h"
#include "esp_err.h"
#include "sdkconfig.h"
#include "nvs_flash.h"
#include "freertos/FreeRTOS.h"
#include "freertos/task.h"

static const char *TAG = "main";

void init_nvs(char *from)
{
    TickType_t init_start_ticks = xTaskGetTickCount();

    esp_err_t err = nvs_flash_init();
    ESP_LOGI(from, "nvs_flash_init: 0x%x, %d ms", err, pdTICKS_TO_MS(xTaskGetTickCount() - init_start_ticks));

    err = nvs_flash_init_partition("log");
    ESP_LOGI(from, "nvs_flash_init_partition: 0x%x, %d ms", err, pdTICKS_TO_MS(xTaskGetTickCount() - init_start_ticks));
}

void thread(void *arg)
{
    init_nvs("from_pinned_thread");
    vTaskDelete(NULL);
}

void app_main(void)
{
    #ifdef CONFIG_PM_ENABLE
    ESP_LOGI(TAG, "CONFIG_PM_ENABLE");
    #else
    ESP_LOGI(TAG, "NO CONFIG_PM_ENABLE");
    #endif

    xTaskCreatePinnedToCore(thread, "NVS_INIT", 1024*32, NULL, 2, NULL, 1);
}
```

```
I (0) cpu_start: Starting scheduler on APP CPU.
I (347) main: CONFIG_PM_ENABLE
I (427) from_pinned_thread: nvs_flash_init: 0x0, 80 ms
I (2877) from_pinned_thread: nvs_flash_init_partition: 0x0, 2530 ms

```

0xjakob, 2021-09-29:
@Batov could you try to apply [this patch](https://github.com/espressif/esp-idf/files/7230636/0001-flash-permance.zip) and see if it helps?

Alvin1Zhang, 2021-10-13:
@Batov Thanks for reporting, would you please help share any updates as suggested in https://github.com/espressif/esp-idf/issues/7438#issuecomment-930109689? Thanks.

Alvin1Zhang, 2021-11-23:
@Batov Thanks for reporting, would you please help share any updates for the issue? Thanks.

Batov, 2021-11-23:
Ohh... Sorry for delay. But I am unable to reproduce issue. I changed storage system to littlefs. It turned of that littlefs is more suitable for 3M data stotage =)

0xjakob, 2021-11-23:
@Batov No worries. We're happy to hear that the problem is solved for you. We're closing this issue. If you find out that the issue still persists, feel free to re-open it.

And yes, NVS is most suitable for key-value pairs of relatively small size.

closed 2021-11-23 by 0xjakob