espressif/esp-idf #7440: BluFi w/ Nimble Example in Panic

jolu1977, 2021-08-18:
**Problem Description**
Using IDF from master where the blufi nimble implementation and example are available.  Example runs fine with bluedroid though when switched to nimble, enabling CONFIG_BT_NIMBLE_BLUFI_ENABLE and flashing to devboard (ESP32-C3-DevKitM-1) things go into a panic.

sdkconfig and stack dump are attached.

[sdkconfig.txt](https://github.com/espressif/esp-idf/files/7008630/sdkconfig.txt)
[blufi_nimble_stack_dump.txt](https://github.com/espressif/esp-idf/files/7008619/blufi_nimble_stack_dump.txt)

rahult-github, 2021-08-19:
Hi @jolu1977 , 

Can you please confirm if the issue is resolved with attached changes ?
[fix_init_crash_c3.txt](https://github.com/espressif/esp-idf/files/7011630/fix_init_crash_c3.txt)

Thanks,
 Rahul

jolu1977, 2021-08-19:
Confirmed, example code now works.

Thanks!

closed 2021-08-19 by jolu1977