espressif/esp-idf #7442: switching i2s peripheral from PDM to ADC modes and back

oclyke, 2021-08-18:
## Environment

- Development Kit: ESP32-DevKitC
- Kit version: v2
- IDF version: v4.4-dev-2594-ga20df743f1
- Build System: idf.py
- Compiler version: 8.2.0
- Operating System: macOS
- Using an IDE?: No
- Power Supply: USB

## Description

I have an audio application in which the input can be changed from a PDM microphone to an analog input that is sampled via the ADC, but only one is needed at a time. To maximise performance I'd like to record the input with DMA, and this seems possible thank to the I2S peripheral. 

So far I have demonstrated both the PDM and ADC sources independently, but I have run into trouble when trying to combine the two in an application that switches back and forth.

I am using the I2S [ESP-IDF API](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/peripherals/i2s.html) to install i2s drivers, read from i2s in a loop, and then uninstall the driver before installing a different driver.

### Expected Behavior

When applying proper de-initialization and re-initialization it should be possible to switch the I2S peripheral back and forth between PDM and ADC modes.

### Actual Behavior

Despite using identical configurations and uninstalling drivers according to the ESP-IDF I find that I2S only works for the first driver installed, after which reconfiguration completes successfully but the data appears corrupted.

### Steps to reproduce

1. attach PDM microphone clk pin to GPIO5 and data pin to GPIO35. (use left channel select, and don't forget power supply pins)
2. attach mono analog signal (1kHz ac, 2vpp, 1vdc bias) to GPIO36 (SENSOR_VN) with reference to ESP32 gnd
3. run gist code with various tests selected (uncommented).

you will find that each driver alone works as intended, but when switching between drivers only the first works

### Code to Reproduce

build this [gist](https://gist.github.com/oclyke/bac5f2fd4b10fba1e337b9e2070ab9dc)

FHeilmann, 2021-11-16:
I'm experiencing a similar issue,
I'm trying to switch back and forth between an I²S MEMS microphone (INMP441) and ADC-via-I²S sampling, using a MAX9814.
I have similar observations as you, only in my case I either get no data back from the I2S-read calls at all, or my code crashes during some internal RTOS queue call. What I've found is the following:

- Switching i2s modes back-to-back causes problems, as the 2nd mode will return corrupted or no data at all.
- Soft resetting the esp32 via `esp_restart()` inbetween swapping modes **also doesn't work**
- Resetting via reset button works as expected.

Therefore, it is my belief that the api calls to uninstall the i2s driver leave the i2s peripheral in some weird state, that makes subsequent use impossible without a hw-reset. This is corroborated by the fact that a soft-reset does not fix this problem (as the docs state that esp_restart() doesn't reset all peripherals).

I've tried adding a manual peripheral reset (via `periph_module_reset`), but that didn't help, either.

I hope a solution can be found

Bests,

Florian

FHeilmann, 2021-11-16:
Some peculiar differences that I found with the following two sequences of i2s drivers:

A:
- Manual reset
- I2S_ADC (works)
- I2S (doesn't work)
- I2S_ADC (works)

B:
- Manual reset
- I2S (works)
- I2S_ADC (doesn't work)
- I2S (doesn't work)

L-KAYA, 2022-07-20:
The test shows that the normal reset can't reset the ADC module, i.e. can't disconnect the I2S signal from built-in ADC, only CPU reset can do that. It's now a known issue of hardware.

closed 2022-09-02 by Alvin1Zhang

FHeilmann, 2022-09-02:
@Alvin1Zhang I don't agree with the closure as no additional information was requested from your side. Please let us know which additional feedback should be provided and reopen this issue.

softhack007, 2022-10-22:
Any news on the topic?? Actually we have a number of user reporting that "their arms are not long enough" to reach to the ceiling and press RST 😉  every time they change ADC config...

@L-KAYA you said "It's now a known issue of hardware" does it mean that a new ESP32 "revision 4" HW could fix the problem? Or is it planned to solve this in a future release of IDF?

netmindz, 2022-10-23:
I see it's been given the label of "Won't Do", but without giving proper reason. Would be good to establish if this is really issue with the HW or Software

L-KAYA, 2022-10-25:
It's now a confirmed digital issue, if I2S0 rx_reset is set and cleared while I2S0 peripheral is enabled, then ADC will always read '0'.

Here is an example on for master and 5.0 to reproduce the issue:
[esp32_adc_read_failure.zip](https://github.com/espressif/esp-idf/files/9858525/esp32_adc_read_failure.zip)

L-KAYA, 2022-10-25:
> Any news on the topic?? Actually we have a number of user reporting that "their arms are not long enough" to reach to the ceiling and press RST 😉 every time they change ADC config...
>
> @L-KAYA you [...]

I've reported this issue for several months but still not get any reply, so I can't tell an exact time when we can get a solution, and neither not sure if the fix will be included in revision 4, I'll keep tracing and urging this issue. 

Feel free to reopen this thread~