espressif/esp-idf #7443: Embedding git version in a C program using the CMake build system (IEP-512)

icodk, 2021-08-19:
**This is a question  not a bug**
I need to embed the current git tag and hash in the C program
How do I do it 

**My current attempt:**
I added the following line to the CMakeLists.txt
`execute_process(COMMAND cmd /C git describe --abbrev=7 --dirty --always --tags  )`
But have no idea how I can save the output to a variable I can use in my  the C program
**Expected behavior**
A variable I can use in my program that will  contain a string with the current git hash, and tag as extracted by the git command above.

**Environment Information:** 
```
esp-idf-v4.2
ESP-IDF Eclipse Plugin
Version: 2.1.2.202106231127
Windows 10
```
**eclipse**
```
Version: 2021-03 (4.19.0)
Build id: 20210312-0638
```

**ESP-IDF Tools Information:**
```
Executing C:\Dev\esp32-idf\idf-tools\.espressif\python_env\idf4.2_py3.9_env\Scripts\python.exe C:\Dev\esp32-idf\esp-idf-v4.2\tools\idf_tools.py list
WARNING: directory for tool idf-exe version 1.0.1 is present, but tool was not found
* xtensa-esp32-elf: Toolchain for Xtensa (ESP32) based on GCC
  - esp-2020r3-8.4.0 (recommended, installed)
* xtensa-esp32s2-elf: Toolchain for Xtensa (ESP32-S2) based on GCC
  - esp-2020r3-8.4.0 (recommended, installed)
* esp32ulp-elf: Toolchain for ESP32 ULP coprocessor
  - 2.28.51-esp-20191205 (recommended, installed)
* esp32s2ulp-elf: Toolchain for ESP32-S2 ULP coprocessor
  - 2.28.51-esp-20191205 (recommended, installed)
* cmake: CMake build system
  - 3.16.4 (recommended, installed)
* openocd-esp32: OpenOCD for ESP32
  - v0.10.0-esp32-20200709 (recommended, installed)
* ninja: Ninja build system
  - 1.10.0 (recommended, installed)
* idf-exe: IDF wrapper tool for Windows
  - 1.0.1 (recommended)
* ccache: Ccache (compiler cache)
  - 3.7 (recommended, installed)
* dfu-util: dfu-util (Device Firmware Upgrade Utilities)
  - 0.9 (recommended, installed)
```

o-marshmallow, 2021-08-19:
Hi @icodk ,
A simple solution to realize this regardless of your IDE is to modify your example's `CMakeLists.txt` file to retrieve the git version and pass it to the compiler through a macro. Let's take `hello-world` for example, by modifying its `CMakeLists.txt` (`esp-idf/examples/get-started/CMakeLists.txt`) to look like this:
```
cmake_minimum_required(VERSION 3.5)

include($ENV{IDF_PATH}/tools/cmake/project.cmake)
execute_process(
  # The output of the git command will be stored in "version" cmake variable
  OUTPUT_VARIABLE version
  # Remove any trailing white space, mainly NEWLINE characters
  OUTPUT_STRIP_TRAILING_WHITESPACE
 # Adapt the command to your OS, this one is for Linux
  COMMAND git describe --abbrev=7 --dirty --always --tags)

# Now transfer the cmake variable containing the git version to the C files through STR_GIT_VERSION macro
add_compile_definitions(STR_GIT_VERSION="${version}")

project(hello-world)
```
And by using the macro `STR_GIT_VERSION` inside your C code:
```
void app_main(void)
{
    printf("Git version:%s\n", STR_GIT_VERSION);
}
```
You'll get the output of git command directly within your ESP program.

(This was tested on Ubuntu 20.20 with CMake 3.16.3)

igrr, 2021-08-19:
One more thing, ESP-IDF provides a `PROJECT_VER` CMake variable which stores the project version, see [the build system guide](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/build-system.html#build-project-variables). There are a few different ways to set it, but `git describe` output will be used by default if your project is in a git repository.
You can then use `add_compile_definitions` or `idf_build_set_property(COMPILE_DEFINITIONS "-DSTR_GIT_VERSION=${PROJECT_VER}" APPEND)`.

icodk, 2021-08-19:
@igrr This works perfect. However I added "" to make it as string:
`idf_build_set_property(COMPILE_DEFINITIONS "-DSTR_GIT_VERSION=\"${PROJECT_VER}\"" APPEND) `
And then I could use it in my code:
 `printf("Starting version: %s\n",STR_GIT_VERSION);`
Haven't try @o-marshmallow but thanks for the insight

closed 2021-08-19 by icodk