espressif/esp-idf #7444: Support SNTP get servers from DHCP

joncmaloney, 2021-08-19:
**Is your feature request related to a problem? Please describe.**

A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]

**Describe the solution you'd like**

A clear and concise description of what you want to happen.

**Describe alternatives you've considered**

A clear and concise description of any alternative solutions or features you've considered.

Please give as many details as you can. Include suggestions for useful APIs or interfaces if relevant.

**Additional context**

Add any other context or screenshots about the feature request here.

joncmaloney, 2021-08-19:
There is currently provision in esp-idf/components/lwip/lwip/src/include/lwip/apps/sntp.h for the option SNTP_GET_SERVERS_FROM_DHCP however this feature doesn't seem to link to any menuconfig #defines. 

Is it possible to add support so that lwip can get NTP server address from DHCP options?

david-cermak, 2021-08-19:
Hi @joncmaloney 

We have this PR https://github.com/espressif/esp-idf/pull/7336 , currently in progress of merging to master, which adds `CONFIG_LWIP_DHCP_GET_NTP_SRV` as the configuration option.

joncmaloney, 2021-08-19:
Thanks David. Great looking forward to it.

closed 2021-08-19 by joncmaloney