espressif/esp-idf #7445: Debug session gets cpu reset after app_main() in ESP IDF 4.3

nitinbhosaleemb, 2021-08-19:
Logs of eclipse debug session :

683,468 2-gdb-version
683,469 ~"GNU gdb (crosstool-NG esp-2020r3) 8.1.0.20180627-git\n"
683,469 ~"Copyright (C) 2018 Free Software Foundation, Inc.\n"
683,469 ~"License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>\nThis is fre\
e software: you are free to change and redistribute it.\nThere is NO WARRANTY, to the extent permitt\
ed by law.  Type \"show copying\"\nand \"show warranty\" for details.\n"
683,469 ~"This GDB was configured as \"--host=x86_64-host_w64-mingw32 --target=xtensa-esp32s3-elf\".\
\nType \"show configuration\" for configuration details."
683,469 ~"\nFor bug reporting instructions, please see:\n"
683,469 ~"<http://www.gnu.org/software/gdb/bugs/>.\n"
683,469 ~"Find the GDB manual and other documentation resources online at:\n<http://www.gnu.org/soft\
ware/gdb/documentation/>.\n"
683,469 ~"For help, type \"help\".\n"
683,469 ~"Type \"apropos word\" to search for commands related to \"word\".\n"
683,469 2^done
683,469 (gdb) 
683,470 3-environment-cd C:/Users/e0624432/eclipse-workspace/ble_sensor_server
683,480 3^done
683,480 (gdb) 
683,481 4-gdb-set breakpoint pending on
683,492 4^done
683,492 (gdb) 
683,492 5-enable-pretty-printing
683,503 5^done
683,503 (gdb) 
683,503 6-gdb-set python print-stack none
683,514 6^done
683,514 (gdb) 
683,514 7-gdb-set print object on
683,525 7^done
683,525 (gdb) 
683,525 8-gdb-set print sevenbit-strings on
683,536 8^done
683,536 (gdb) 
683,536 9-gdb-set charset ISO-8859-1
683,547 9^error,msg="Undefined item: \"ISO-8859-1\"."
683,547 (gdb) 
683,547 10-gdb-set dprintf-style gdb
683,558 10^done
683,558 (gdb) 
683,558 11source .gdbinit
683,569 &"source .gdbinit\n"
683,570 &".gdbinit: No such file or directory.\n"
683,570 11^error,msg=".gdbinit: No such file or directory."
683,570 (gdb) 
683,570 12-gdb-set target-async off
683,581 12^done
683,581 (gdb) 
683,581 13-gdb-set auto-solib-add on
683,581 13^done
683,581 (gdb) 
683,583 14symbol-file C:\\Users\\e0624432\\eclipse-workspace\\ble_sensor_server\\build\\sensor_serve\
r.elf

683,593 &"symbol-file C:\\\\Users\\\\e0624432\\\\eclipse-workspace\\\\ble_sensor_server\\\\build\\\\\
sensor_server.elf\n"
683,595 ~"Reading symbols from C:\\Users\\e0624432\\eclipse-workspace\\ble_sensor_server\\build\\sen\
sor_server.elf..."
683,667 ~"done.\n"
683,672 14^done
683,673 (gdb) 
683,673 &"\n"
683,673 ^done
683,673 (gdb) 
683,673 15-target-select remote localhost:3333

683,709 =thread-group-started,id="i1",pid="42000"
683,709 &"warning: No executable has been specified and target does not support\ndetermining executa\
[... 195 lines cut ...]
force hard breakpoints
Info : Listening on port 6666 for tcl connections
Info : Listening on port 4444 for telnet connections
Info : ftdi: if you experience problems at higher adapter clocks, try the command "ftdi_tdo_sample_edge falling"
Info : clock speed 20000 kHz
Info : JTAG tap: esp32s3.cpu0 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
Info : JTAG tap: esp32s3.cpu1 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
Info : esp32s3.cpu0: Target halted, PC=0x4200926F, debug_reason=00000004
Info : Listening on port 3333 for gdb connections
Info : accepting 'gdb' connection on tcp/3333
Warn : No symbols for FreeRTOS!
Info : New GDB Connection: 1, Target esp32s3.cpu0, state: halted
Info : JTAG tap: esp32s3.cpu0 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
Info : JTAG tap: esp32s3.cpu1 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
Info : esp32s3.cpu0: Debug controller was reset.
Info : esp32s3.cpu0: Core was reset.
Info : esp32s3.cpu0: Target halted, PC=0x500000C4, debug_reason=00000000
Info : Set GDB target to 'esp32s3.cpu0'

Info : esp32s3.cpu0: Core was reset.
Info : esp32s3.cpu0: Target halted, PC=0x40000400, debug_reason=00000000
Warn : address 0x3c0a0020 not writable
Warn : address 0x3c0a0120 not writable
Info : esp32s3.cpu0: Target halted, PC=0x4200926C, debug_reason=00000001
Info : Set GDB target to 'esp32s3.cpu0

igrr, 2021-08-23:
@nitinbhosaleemb It looks like you might have forgot to uncheck "load image" checkbox in the jtag debugging configuration? (See step 9 in https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/jtag-debugging/using-debugger.html#eclipse).

Alvin1Zhang, 2021-11-22:
Thanks for reporting, will close due to shore of feedback, feel free to reopen wit more updates.

closed 2021-11-22 by Alvin1Zhang