espressif/esp-idf #7446: Stop TX when 485 collision detects

Dandjinh, 2021-08-20:
**Is your feature request related to a problem? Please describe.**

I'm always frustrated when 485 collision detects, it only clears the rx fifo while tx is still working.
Now it's only possible to know the collision after TX is done because no events will be posted and uart_wait_tx_done() doesn't handle such issue.
Also there is no api to stop TX according to _uart.h_.
When the packet is large it's waste of time.

**Describe the solution you'd like**

When 485 collision detects, stops TX in the ISR handler and posts an event.

**Describe alternatives you've considered**

1. clear txfifo in uart_rx_intr_handler_default() when collision detects.
2. add an event UART_485_BUS_COLLISION and post it when collision detects.
3. add an api that stops UART TX.

alisitsyn, 2021-08-20:
@Dandjinh,

Thank you for your issue. The UART driver currently supports collision detection using hardware feature and corresponded flag `UART_INTR_RS485_CLASH`. When the transmission is complete and collision is detected UART prepares for receiving data by flushing the receive buffer. This approach is ok for J1708 networks while the `Message Identification character (MID)` is used. The user application can get status of collision `uart_get_collision_flag()` and start collision-retry procedure. I agree the API are very limited now and allows checking if the collision as described above.
Your proposals will be taken into account and it is planned to check and update collision support feature.
For now if you need customized handling of collision detection in your project I would propose you to realize it using customized UART interrupt handler and UART HAL layer.

Alvin1Zhang, 2022-09-06:
Thanks for reporting, seems this duplicates https://github.com/espressif/esp-idf/issues/3569, will close this issue. Let us track progress in https://github.com/espressif/esp-idf/issues/3569. Thanks.

closed 2022-09-06 by Alvin1Zhang

alisitsyn, 2022-09-07:
@Dandjinh, @AshUK,

The example which uses HAL layer and adopted RS485 software driver to detect collisions in RS485 bus based networks is located [here](https://github.com/alisitsyn/modbus_support/tree/feature/driver_collision_detect/rs485_collision_detector) .

[image]

[collision_detection_log.txt](https://github.com/espressif/esp-idf/files/9505293/collision_detection_log.txt)

It is not free of some minor issues and still in WIP state but can describe what configuration of internal UART hardware can be used to detect collisions in RS485 networks.
Once this feature adequately tested it will be implemented in standard UART drive.

Please check this and report if you have issues.
Thanks.

alisitsyn, 2022-09-29:
@Dandjinh,

Could you try the solution proposed to detect collisions in RS485 communication as above?

d4rkmen, 2024-01-18:
@alisitsyn for now tested your example in full duplex mode and collision detection works, thanks. When transmitted characted <> received in same moment, interrupt fired. This screen shows retry on collision after some delay
[image]

alisitsyn, 2024-01-18:
@d4rkmen,

Thank you for your update!
It is good to hear it works on your side. Please keep me posted if you have an update in your project. I can continue with the improvement of the feature once other higher priority tasks are completed.