espressif/esp-idf #7448: �[0;31mE (492351) MQTT_CLIENT: mqtt_message_receive: transport_read() error: errno=128�[0m �[0;31mE (492351) MQTT_CLIENT: mqtt_process_receive: mqtt_message_receive() returned -1�[0m �[0;33mW (492351) MQTT_CLIENT: Client asked to stop, but was not started�[0m offline

anudeepsharma254, 2021-08-20:
Hi I'm having same issue as @https://github.com/espressif/esp-idf/issues/6341#issue-776428094 with ESP32 but in my case device always disconnects on time after 10 seconds of connecting to the Mqtt server, then reconnects within seconds, and then disconnects after 10 seconds of reconnection.
I'm using nodeMCU (https://nodemcu.readthedocs.io/en/dev-esp32/modules/mqtt/). I can't identify issue myself(whether it is related to server or device)

### Environment
- Development Kit: ESP32 DEVKITV1
- Module or chip used: ESP32-WROOM-32
- Firmware: nodeMCU
- Operating System: Windows
- IDE: ESPlorer(v0.2.0)

### Here my code..

local client_id,user_id,password,host_id,port="House1","temp1","temp2","198.xx.xx.xxx",18xx
topics={"iot/ANUD45/HAll","iot/ADR1/KITCH","iot/ADR5/BED"}
m = mqtt.Client(client_id, 0,user_id,password)
m:lwt("/lwt", "offline", 0, 0)
m:on("connect", function(client) print ("connection estblished") end)
m:on("offline", function(client) print ("offline") do_mqtt_connect end)
m:on("message", function(client, topic, data)
if data ~= nil then
data=string.sub(data,2,#data-1)
print('mqtt',data)
end
end)
function handle_mqtt_error(client, reason)
tmr.create():alarm(10 * 1000, tmr.ALARM_SINGLE, do_mqtt_connect)
end
function do_mqtt_connect()
m:connect(host_id, port, 0)
m:subscribe(topics[3], 0, function(client_3) print("subscribe success_9") end)
m:subscribe(topics[1], 0, function(client_1) print("subscribe success_1") end)
m:subscribe(topics[2], 0, function(client_2) print("subscribe success_4") end)
end
do_mqtt_connect()

### Here is my error..
�[0;31mE (492351) MQTT_CLIENT: mqtt_message_receive: transport_read() error: errno=128�[0m
�[0;31mE (492351) MQTT_CLIENT: mqtt_process_receive: mqtt_message_receive() returned -1�[0m
�[0;33mW (492351) MQTT_CLIENT: Client asked to stop, but was not started�[0m
offline

Alvin1Zhang, 2021-08-20:
@anudeepsharma254 Thanks for reporting, would you please help update issue title? Thanks.

anudeepsharma254, 2021-08-20:
Hi @Alvin1Zhang thanks for the help I thought it's better to specify complete error as issue name

david-cermak, 2021-08-20:
Hi @anudeepsharma254 

Could you please check for the same points as suggested in the referenced issue?
I'd like to see
* debug logs from the board
* wireshark capture
* logs from the broker

It would be also great to share
* idf version
* mqtt client configuration structure
* which broker you're using

In general, the error=128 (`ENOTCONN`) occurred during reading means that the client was trying to read from the socket which was already closed by the server. So it was the server that closed the connection, but still doesn't mean there's no issue with the client. The server might have closed the connection due to some client's error, maybe. this should be seen in the server logs.

Alvin1Zhang, 2021-08-31:
@anudeepsharma254 Thanks for reporting, would you please help share if any updates for the issue? Thanks.

d0ct0rd-dalimo, 2021-09-11:
I have the same problem with connection to local MQTT broker, although other clients connects properly.

The logs are
```
[I][mqtt_messaging.cpp:249] mqtt_event_handler(): MQTT_EVENT_DISCONNECTED
[I][mqtt_messaging.cpp:462] mqtt_event_handler(): Other event id:7
E (164519) MQTT_CLIENT: mqtt_message_receive: transport_read() error: errno=128
[I][mqtt_messaging.cpp:459] mqtt_event_handler(): MQTT_EVENT_ERROR
E (164521) MQTT_CLIENT: esp_mqtt_connect: mqtt_message_receive() returned -1
E (164528) MQTT_CLIENT: MQTT connect failed
[I][mqtt_messaging.cpp:249] mqtt_event_handler(): MQTT_EVENT_DISCONNECTED
[I][mqtt_messaging.cpp:462] mqtt_event_handler(): Other event id:7
E (174563) MQTT_CLIENT: mqtt_message_receive: transport_read() error: errno=128
[I][mqtt_messaging.cpp:459] mqtt_event_handler(): MQTT_EVENT_ERROR
E (174566) MQTT_CLIENT: esp_mqtt_connect: mqtt_message_receive() returned -1
E (174573) MQTT_CLIENT: MQTT connect failed
```
I tried to tweak MQTT version, change host or server address format (it's IP-address), with config, delay between connect with WiFi and connect to MQTT but it didn't make any effect. But once per 100 it can connect to MQTT. ESP-IDF version is v3.3.5-1-g85c43024c, used from ESP32-Arduino

anudeepsharma254, 2021-09-11:
Hi @Alvin1Zhang, Instead of subscribing individual topics used '#' wildcard for subscribing all subtopics of a topic, with this usage disconnection is delayed (now disconnecting from server not in a regular intervals means disconnecting 20mins or 30mins(sometimes) after connection established) (previously  device always disconnects on time after 10 seconds of connecting to the Mqtt server) still waiting for effective solution ...

d0ct0rd-dalimo, 2021-09-11:
Checked with mosquitto, it gives the errors as follows:
```

2021-09-11T11:38:22: Socket error on client <unknown>, disconnecting.,
2021-09-11T11:38:22: New connection from 192.168.0.94 on port 1883.,
2021-09-11T11:38:12: Socket error on client <unknown>, disconnecting.,
2021-09-11T11:38:12: New connection from 192.168.0.94 on port 1883.,
2021-09-11T11:38:02: Socket error on client <unknown>, disconnecting.,
2021-09-11T11:38:02: New connection from 192.168.0.94 on port 1883.
```

david-cermak, 2021-09-13:
@d0ct0rd-dalimo Thanks for sharing the logs from moquitto, this is helpful. Could you also please share more detailed log with `mosquitto -v` ?

> ESP-IDF version is v3.3.5-1-g85c43024c, used from ESP32-Arduino

Are you sure it was this version? I've seen these kind of errors (`Socket error on client <unknown>` ) on very old version of esp-mqtt as a result of wrong packet fragmentation, but this has been fixed in `v3.3.5`. Also tried to stress test the library publishing as fast as possible to a local mosqutto, but no disconnection seen.
From the logs it seems you're using plain tcp with a local server. Any authentication, timeout setting? Is the client's connection stable?

> But once per 100 it can connect to MQTT.

Once it *can* connect and 99 times it can not? Would you please share your application? Could even be arduino sketch.

---

@anudeepsharma254 Could you please share more details as requested in https://github.com/espressif/esp-idf/issues/7448#issuecomment-902551516 ?

> still waiting for effective solution ...

Sorry, can only help if you provided more information

anudeepsharma254, 2021-09-17:
Sure @david-cermak but right now I don't have server access and I'm using my friend's server so that's why I can't but I'll share in few days and I don't have any debug logs in nodemcu and I'm using NodeMCU firmware not IDF tool, in NodeMCU  they use mqtt V3.1.1 I think, this is nodeMCU mqtt link https://nodemcu.readthedocs.io/en/dev-esp32/modules/mqtt/

david-cermak, 2021-09-17:
@anudeepsharma254 

> I don't have server access and I'm using my friend's server so that's why I can't but I'll share in few days 

Could you please tell us, which kind of broker is it that?

> I don't have any debug logs in nodemcu 

Please check the [NodeMCU docs](https://nodemcu.readthedocs.io/en/dev-esp32/build/#build-options) on build and configuration, navigate to components, log and enable debug logs

>in NodeMCU they use mqtt V3.1.1

This is the version of mqtt protocol, not the version of SDK. Please navigate to your NodeMCU root and type
```
cd sdk/esp32-esp-idf
git rev-parse HEAD
```
and share the result with us.

The best guess I can get is that you're on `dev-esp32` branch of the nodemcu, which references [idf 3.3.4](https://github.com/espressif/esp-idf/commits/b64b3752342a23469ada0188d4838a4fb96fe172) which in turn uses this [mqtt version](https://github.com/espressif/esp-mqtt/tree/6bc94add892437d0fd50f26bfabe78c646648c13). Although it's quite old, there haven't been any significant and major bugfixes since, so this version (if this is indeed the version you use) should cause no disconnections.

Please share your configuration options, too. You can just paste the init portion of your code using NodeMCU. I'm not familiar with NodeMCU (and AFAIK it uses unsupported IDF's) but still can try to look into the issue.

It would be also good to provide wireshark packet capture.

anudeepsharma254, 2021-09-17:
Thank you @david-cermak sure I'll share those details shortly and thanks once again for detailed instructions

Alvin1Zhang, 2021-12-27:
@anudeepsharma254 Thanks for reporting, will close due to short of feedback, feel free to reopen with more details.

closed 2021-12-27 by Alvin1Zhang