espressif/esp-idf #7449: MCPWM: Guru Meditation Error: Core  1 panic'ed (Cache disabled but cached memory region accessed)

p1279030826, 2021-08-20:
## Environment

- Development Kit:      `none`
- Kit version (for WroverKit/PicoKit/DevKitC): [v1|v2|v3|v4]
- Module or chip used:  `ESP32-WROOM-32`
- IDF version (run ``git describe --tags`` to find it):  `v4.2-47-g2532ddd9f`
- Build System: `Make`
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it): `esp-2020r3-8.4.0`
- Operating System:    `Linux`
- Using an IDE?: `No`
- Power Supply:  `external 3.3V`

## Problem Description
1、Open a task that writes data (NVS) every 5 seconds
2、Open a task to read the queue of mcpwm capture data

Error reporting function `mcpwm_capture_signal_get_value`

[Fatal Errors](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/fatal-errors.html)
It cannot be solved according to the processing method of the document

### Expected Behavior
Capture data normally

### Actual Behavior
Irregular restart

### Steps to reproduce

1. step1
2. ...

// If possible, attach a picture of your setup/wiring here.

### Code to reproduce this issue

```c
#include <stdio.h>
#include "string.h"
#include "freertos/FreeRTOS.h"
#include "freertos/task.h"
#include "freertos/queue.h"
#include "esp_attr.h"

#define CAP0_INT_EN     BIT(27)  //Capture 0 interrupt bit
#define CAP1_INT_EN     BIT(28)  //Capture 1 interrupt bit
#define CAP2_INT_EN     BIT(29)  //Capture 2 interrupt bit

#define GPIO_CAP0_IN   GPIO_NUM_39
#define GPIO_CAP1_IN   GPIO_NUM_34
#define GPIO_CAP2_IN   GPIO_NUM_35

typedef struct {
    uint8_t sel_cap_signal;
    uint32_t capture_value;
} capture;

xQueueHandle cap_queue;
static mcpwm_dev_t *MCPWM = {&MCPWM0};

/**
 * @brief ISR
 */
static void IRAM_ATTR isr_handler(void *arg)
{
    uint32_t mcpwm_intr_status;
    capture evt;

    mcpwm_intr_status = MCPWM->int_st.val;
    if (mcpwm_intr_status & CAP0_INT_EN) {
        evt.sel_cap_signal = MCPWM_SELECT_CAP0;
        evt.capture_value = mcpwm_capture_signal_get_value(MCPWM_UNIT_0, MCPWM_SELECT_CAP0);
        xQueueSendFromISR(cap_queue, &evt, NULL);
    }

    if (mcpwm_intr_status & CAP1_INT_EN) {
        evt.sel_cap_signal = MCPWM_SELECT_CAP1;
        evt.capture_value = mcpwm_capture_signal_get_value(MCPWM_UNIT_0, MCPWM_SELECT_CAP1);
        xQueueSendFromISR(cap_queue, &evt, NULL);
    }

    if (mcpwm_intr_status & CAP2_INT_EN) {
        evt.sel_cap_signal = MCPWM_SELECT_CAP2;
        evt.capture_value = mcpwm_capture_signal_get_value(MCPWM_UNIT_0, MCPWM_SELECT_CAP2);
        xQueueSendFromISR(cap_queue, &evt, NULL);
    }
    MCPWM->int_clr.val = mcpwm_intr_status;
}

/**
 * @brief Configure MCPWM capture
 */
void mcpwm_capture_init(void)
{
    cap_queue = xQueueCreate(20, sizeof(capture));

    // mcpwm gpio initialization
    mcpwm_pin_config_t pin_config = {
        .mcpwm_cap0_in_num = GPIO_CAP0_IN,
        .mcpwm_cap1_in_num = GPIO_CAP1_IN,
        .mcpwm_cap2_in_num = GPIO_CAP2_IN,
    };
    mcpwm_set_pin(MCPWM_UNIT_0, &pin_config);

    gpio_pullup_en(GPIO_CAP0_IN); 
    gpio_pullup_en(GPIO_CAP1_IN);
    gpio_pullup_en(GPIO_CAP2_IN);

    // Capture configuration
    mcpwm_capture_enable(MCPWM_UNIT_0, MCPWM_SELECT_CAP0, MCPWM_NEG_EDGE, 0);
    mcpwm_capture_enable(MCPWM_UNIT_0, MCPWM_SELECT_CAP1, MCPWM_NEG_EDGE, 0);
    mcpwm_capture_enable(MCPWM_UNIT_0, MCPWM_SELECT_CAP2, MCPWM_NEG_EDGE, 0);

    MCPWM->int_ena.val = CAP0_INT_EN | CAP1_INT_EN | CAP2_INT_EN; 
    mcpwm_isr_register(MCPWM_UNIT_0, isr_handler, NULL, ESP_INTR_FLAG_IRAM, NULL);
}
```

## Debug Logs
```
Guru Meditation Error: Core  1 panic'ed (Cache disabled but cached memory region accessed). 

Core  1 register dump:
PC      : 0x40105134  PS      : 0x00060034  A0      : 0x80102b36  A1      : 0x3ffb0c30  
0x40105134: mcpwm_ll_get_intr at /home/pjw/ESP32/esp-idf/components/soc/src/esp32/include/hal/mcpwm_ll.h:79
 (inlined by) mcpwm_hal_capture_get_result at /home/pjw/ESP32/esp-idf/components/soc/src/hal/mcpwm_hal.c:222

A2      : 0x3ff5e000  A3      : 0x3ff5e000  A4      : 0x3ffb0c60  A5      : 0x08000000  
A6      : 0x00000001  A7      : 0x00060523  A8      : 0x08000000  A9      : 0x3ff5e000  
A10     : 0x00000001  A11     : 0x0000003c  A12     : 0x00000000  A13     : 0x3ffb253c  
A14     : 0x00000003  A15     : 0x00060023  SAR     : 0x00000005  EXCCAUSE: 0x00000007  
EXCVADDR: 0x00000000  LBEG    : 0x00000000  LEND    : 0x00000000  LCOUNT  : 0x00000000  

Backtrace:0x40105131:0x3ffb0c30 0x40102b33:0x3ffb0c50 0x40084d83:0x3ffb0c90 0x40084dce:0x3ffb0cb0 0x40082401:0x3ffb0cd0 0x40082d67:0x3ffb4c80 0x40081b69:0x3ffb4ca0 0x4008827a:0x3ffb4cc0
0x40105131: mcpwm_ll_clear_intr at /home/pjw/ESP32/esp-idf/components/soc/src/esp32/include/hal/mcpwm_ll.h:90
 (inlined by) mcpwm_hal_capture_get_result at /home/pjw/ESP32/esp-idf/components/soc/src/hal/mcpwm_hal.c:220

0x40102b33: mcpwm_capture_signal_get_value at /home/pjw/ESP32/esp-idf/components/driver/mcpwm.c:577 (discriminator 2)

0x40084d86: isr_handler at /home/pjw/ESP32/My_project/V4.2/mes/components/bsp/src/bsp_mcpwm.c:35
```

suda-morris, 2021-08-21:
Hi @p1279030826 This is because you enabled the `ESP_INTR_FLAG_IRAM` flag in function `mcpwm_isr_register`, which will make the MCPWM interrupt can still get serviced even when the cache is disabled (e.g. any NVS write operation).

You can either:
1. Remove that flag if you don't expect the MCPWM capture to work when the cache is disabled (MCPWM capture interrupt can still be triggered after NVS write operation finished)
2. Add `IRAM_ATTR` to `mcpwm_capture_signal_get_value`

p1279030826, 2021-08-21:
Thank you for your reply，I tried to modify `components/driver/mcpwm.c` and add `IRAM_ATTR` to `mcpwm_capture_signal_get_value`，but I don't see any improvement and will still restart.
Amend as follows:

```c
uint32_t IRAM_ATTR mcpwm_capture_signal_get_value(mcpwm_unit_t mcpwm_num, mcpwm_capture_signal_t cap_sig)
```

I expect the mcpwm capture to still work when caching is disabled,but if this problem cannot be solved, the first method can only be used

suda-morris, 2021-08-22:
I see, as you're on 4.2 branch, where `mcpwm_capture_signal_get_value` invokes `mcpwm_hal_capture_get_result`, which function is also not in IRAM by default. Please try to add `IRAM_ATTR` to [that function](https://github.com/espressif/esp-idf/blob/release/v4.2/components/soc/src/hal/mcpwm_hal.c#L210)

p1279030826, 2021-08-24:
Thanks. This method is feasible. After two days of testing, there is no error report and restart.

To sum up, the solutions to this problem are as follows: 

1. `mcpwm.c`, add `IRAM_ATTR` to `mcpwm_capture_signal_get_value`

	```
	uint32_t IRAM_ATTR mcpwm_capture_signal_get_value(mcpwm_unit_t mcpwm_num, mcpwm_capture_signal_t cap_sig)
	{
	    MCPWM_CHECK(mcpwm_num < SOC_MCPWM_PERIPH_NUM, MCPWM_UNIT_NUM_ERROR, ESP_ERR_INVALID_ARG);
	
	    uint32_t captured_value;
	    mcpwm_hal_capture_get_result(&context[mcpwm_num].hal, cap_sig, &captured_value, NULL);
	    return captured_value;
	}
	```

2. `mcpwm_hal.c`, add `IRAM_ATTR` to `mcpwm_hal_capture_get_result`

	```
	esp_err_t IRAM_ATTR mcpwm_hal_capture_get_result(mcpwm_hal_context_t *hal, int cap_sig, uint32_t *out_count, mcpwm_capture_on_edge_t *out_edge)
	{
	    const mcpwm_intr_t sig_intr = mcpwm_ll_get_cap_intr_def(cap_sig);
	    //unconditionally update the output value
	    if (out_edge) {
	        *out_edge = mcpwm_ll_get_captured_edge(hal->dev, cap_sig);
	    }
	    *out_count = mcpwm_ll_get_capture_val(hal->dev, cap_sig);
	
	    if (mcpwm_ll_get_intr(hal->dev) & sig_intr) {
	        mcpwm_ll_clear_intr(hal->dev, sig_intr);
	    }
	    return (mcpwm_ll_get_intr(hal->dev) & sig_intr ? ESP_OK : ESP_ERR_NOT_FOUND);
	}
	```

3. `mcpwm_hal.h`

	```
	#include "esp_attr.h"
	
	esp_err_t IRAM_ATTR mcpwm_hal_capture_get_result(mcpwm_hal_context_t *hal, int cap_sig, uint32_t *out_count,
	                                       mcpwm_capture_on_edge_t *out_edge);
	```

closed 2021-08-29 by espressif-bot