espressif/esp-idf #7450: ESP MESH - Cannot connect to router after self organizing disabled and then reenabled

mvarchdev, 2021-08-21:
----------------------------- Delete below -----------------------------

**Reminder: If your issue is a general question, starts similar to "How do I..", or is related to 3rd party development kits/libs, please discuss this on our community forum at https://esp32.com instead.**

INSTRUCTIONS
============

Before submitting a new issue, please follow the checklist and try to find the answer.

- [ ] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [ ] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [ ] I have searched the issue tracker for a similar issue and not found a similar issue.

If the issue cannot be solved after the steps before, please follow these instructions so we can get the needed information to help you in a quick and effective fashion.

1. Fill in all the fields under **Environment** marked with [ ] by picking the correct option for you in each case and deleting the others.
2. Describe your problem.
3. Include [debug logs from the "monitor" tool](https://docs.espressif.com/projects/esp-idf/en/latest/get-started/idf-monitor.html#automatically-decoding-addresses), or [coredumps](https://docs.espressif.com/projects/esp-idf/en/latest/api-guides/core_dump.html).
4. Providing as much information as possible under **Other items if possible** will help us locate and fix the problem.
5. Use [Markdown](https://guides.github.com/features/mastering-markdown/) (see formatting buttons above) and the Preview tab to check what the issue will look like.
6. Delete these instructions from the above to the below marker lines before submitting this issue.

**IMPORTANT: If you do not follow these instructions and provide the necessary details, your issue may not be resolved.**

----------------------------- Delete above -----------------------------

## Environment

- Development Kit:      ESP32-DevKitC
- Kit version: v4
- Module or chip used:  ESP32-WROOM-32U
- IDF version: v4.4-dev-2594-ga20df743f1
- Build System:         idf.py
- Compiler version: xtensa-esp32-elf-gcc (crosstool-NG esp-2021r1) 8.4.0
- Operating System:     Linux
- Using an IDE?: Yes - Visual Studio code with installed esp-idf extension, but building and flashing from the server (Ubuntu with esp-idf, same version)
- Power Supply:         USB

## Problem Description

The problem is simple but really annoying and cannot work with it. When the router is disconnected, the root will automatically switch to self organised disabled. When the router is found by the external task, the message of router found is broadcasted over the mesh (via esp_mesh_send() func.) and thereafter all nodes recognize this message and this code is executed on all nodes including root:

```cpp
if (esp_mesh_is_root())
                    {
                        ESP_ERROR_CHECK_WITHOUT_ABORT(esp_mesh_disconnect());
                        ESP_ERROR_CHECK_WITHOUT_ABORT(esp_mesh_connect());
                    }
                    ESP_ERROR_CHECK_WITHOUT_ABORT(esp_mesh_set_self_organized(1, 1));

```

### Expected Behavior
It should just connect to router.
### Actual Behavior
It just scanning and trying to connect infinitely.
### Steps to reproduce

1. Create example of mesh internal communication
2. Wait for the router is connected. then disable self organizing from mesh event callback - handler.
3. wati appr. 10 sec. (vtaskdelay)
4. then reenable the self organiziing:
```cpp
ESP_ERROR_CHECK_WITHOUT_ABORT(esp_mesh_set_self_organized(1, 1));
```

## Debug Logs

```
[logmesh.log](https://github.com/espressif/esp-idf/files/7025127/logmesh.log)
```

vinicius-smartme, 2021-12-17:
I am having the same problem. 
I am following the example described here [esp-wifi-mesh](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/network/esp-wifi-mesh.html).

What I am doing is to disable the auto wifi scan when I cannot find a router and I have I timer where I try to re-enable it after 10 seconds.

To stop/disable the auto scan I am doing:

`//Disable self organized networking
esp_mesh_set_self_organized(0, 0);

//Stop any scans already in progress
esp_wifi_scan_stop(); `

To re-enable the auto scan I am doing:
`esp_wifi_scan_start();

//At the SCAN_DONE event
//Re-enable self organized networking if root and disconnected
esp_mesh_set_self_organized(1, 0);  //Don't select new parent`

vinicius-smartme, 2021-12-17:
It seems another issue that when there is no router (manual mesh networking), the root keeps scanning for the router and it causes the mesh to have a poor performance, once the scanning is "occupying" the wifi. This is causing the mesh to loose lots of packets.

shenjun7, 2022-07-27:
If the manual networking node cannot find the parent, it will always scan. Setting up a network without a router can be achieved by setting fix-root  `esp_mesh_fix_root(true)` `esp_mesh_set_type()`.

vinicius-smartme, 2022-08-02:
The `esp_mesh_fix_root(true)` doesn't apply to my case, as I don't want to manually define the root, but use the self organized feature. The `esp_mesh_set_type()` defined as `IDLE` at the initialization.

vinicius-smartme, 2022-08-02:
> If the manual networking node cannot find the parent, it will always scan. Setting up a network without a router can be achieved by setting fix-root `esp_mesh_fix_root(true)` `esp_mesh_set_type()`.

@shenjun7, why did I receive the same response for a different issue [here](https://github.com/espressif/esp-idf/issues/8177#issuecomment-1196224105) ?

shenjun7, 2022-08-09:
The self-organized network will elect a new root after the root leaves within a specific time. You can increase the disconnection detection time so that root does not re-elect by `esp_mesh_set_ap_assoc_expire()`.

vinicius-smartme, 2022-08-09:
> The self-organized network will elect a new root after the root leaves within a specific time. You can increase the disconnection detection time so that root does not re-elect by `esp_mesh_set_ap_assoc_expire()`.

The `esp_mesh_set_ap_assoc_expire()` does not affect the the problem, as I can wait many minutes and no root is setup. The default value for `esp_mesh_set_ap_assoc_expire()` is 10 seconds and works in normal router disconnection.

Alvin1Zhang, 2024-02-27:
Thanks for reporting, feel free to reopen. Thanks.

closed 2024-02-27 by Alvin1Zhang

kieennt13, 2024-11-07:
> If the manual networking node cannot find the parent, it will always scan. Setting up a network without a router can be achieved by setting fix-root `esp_mesh_fix_root(true)` `esp_mesh_set_type()`.

Hi @shenjun7 ,
How should I setting mesh_config.router in the no router manual networking scenario?
If I use a dummy SSID for mesh_config.router, set `esp_mesh_set_type(MESH_ROOT)` and comment out the Wifi scan functions, root is just showing this and nothing else:
`
I (23768) wifi:Set ps type: 0

I (23778) wifi:mode : sta (ac:67:b2:c8:b3:70) + softAP (ac:67:b2:c8:b3:71)
I (23778) wifi:Total power save buffer number: 16
I (23778) wifi:Init max length of beacon: 752/752
I (23778) wifi:Init max length of beacon: 752/752
I (23788) mesh: <nvs>read layer:0, err:0x1102
I (23788) mesh: <nvs>read assoc:0, err:0x1102
I (23798) mesh: [IO]disable self-organizing<reconnect>
I (23808) wifi:Total power save buffer number: 16
I (23958) wifi:state: run -> init (0)
I (23958) wifi:pm stop, total sleep time: 15558382 us / 22060505 us

I (23958) wifi:<ba-del>idx
I (23958) wifi:<ba-del>idx
I (23958) wifi:new:<1,0>, old:<1,0>, ap:<1,1>, sta:<1,0>, prof:1
wifi_reconnect_task: 0
I (23968) wifi station: connect to the AP fail
I (23978) mesh_main: <MESH_EVENT_STARTED>ID:77:77:77:77:77:76
I (23978) mesh: [CONFIG]connect to router:XXLL, 00:00:00:00:00:00
-------------------SUCCESS
I (23978) mesh_main: mesh starts successfully, heap:83240

I (24018) mesh: [IO]disable self-organizing<stop reconnect>
I (24038) mesh: <MESH_NWK_PARENT_DISCONNECTED>already disconnected, ignore it
I (26858) wifi station: connect to the AP fail
I (26858) mesh_main: <MESH_EVENT_PARENT_DISCONNECTED>reason:201
I (26858) mesh: [wifi]disconnected reason:201(), continuous:1/max:12, root, vote(,)<><>
I (26868) mesh: <mesh_nwk_task_main,4587>parent is set<stop reconnect>g_is_wifi_connecting:0, g_is_wifi_disconnecting:0, g_mesh_stop_reconnection:1
`
Does this mean I've succeed?
Or if not, what kind of log it should printed out?