espressif/esp-idf #7451: WIFI in apsta is not working in LONG RNAGE mode when Bluetooth enabled

sowmimalu, 2021-08-21:
Hai ,
Module or chip used: ESP32-WROVER
IDF version: v4.3
Compiler version: xtensa-esp32-elf-gcc (crosstool-NG esp-2020r3) 8.4.0

my application has wriiten with Both Bluetooth dual mode and wifi in APSTA mode
other ESP32 device is connecting with ESP32 access point in normal range properly ,
but When i turned into LONG RANGE mode other esp32 device is also in LR STA mode but not connecting ,when i disabled Bluetooth dual its working in LONG RANGE mode.
enable BLE & wifi coexistence option in menuconfig also.
same scenario when checked in IDF 4.1 both modes was worked fine .

I'm not sure what caused this problem ,I would like to hear your comments on this.
Thanks,
Sowmiya

sowmimalu, 2021-08-22:
any suggestions on this ? awaiting for reply

TianaESP, 2021-11-16:
@sowmimalu can you please try the latest master? thanks.

Espressif-liuuuu, 2022-07-19:
Hi @sowmimalu , sorry for late reply. Its not stable to connect an ESP32 AP when BT is also enabled now, especially when LR mode is enabled, more time would be needed by AP to receive connection package, which means BT would affect WiFi transmission more seriously. The performance of AP is up to how your BT works.
We are developing the performance of AP in coexistence mode recently and would be appreicated if you could give us more information here:
1. What jobs did your BT do, e.g advertising/scan? and what parameters did you configure? e.g. scan interval/window
2. Did you connect your ESP32(APSTA mode) to router?
Many thanks for your report

closed 2024-02-27 by Alvin1Zhang