espressif/esp-idf #7452: mbedtls_x509_crt_info calls mbedtls_x509_key_size_helper with buffer that's too small

dannybackx, 2021-08-21:
## Environment

- Development Kit:      any esp32
- Kit version (for WroverKit/PicoKit/DevKitC): N/A
- Module or chip used:  [ESP32]
- IDF version (run ``git describe --tags`` to find it): 4.2.1
- Build System:         idf.py
- Compiler version : any (crosstool-ng-1.22.0-80-g6c4433a 5.2.0)
- Operating System:     any (Linux)
- (Windows only) environment type: N/A
- Using an IDE?: No
- Power Supply:         USB

## Problem Description

When you pass an invalid PK to mbedtls_x509_crt_info() it will call mbedtls_x509_key_size_helper on line 1645 with parameter "invalid PK".
Together with the hardcoded string in mbedtls_x509_key_size_helper, this doesn't fit in the 18 char buffer as specified.

It does work with 20, so suggested fix :
1559,1560c1580,1583
< #define BEFORE_COLON    18
< #define BC              "18"
---
> // #define BEFORE_COLON    18
> // #define BC              "18"
> #define BEFORE_COLON    20
> #define BC              "20"

The lines in the diff may be off a little, I've done other small edits in my local copy, irrelevant for this case.

### Actual Behavior

E (12:43:37.191) JSONServer: mbedtls_x509_crt_info -> -10624
(this is MBEDTLS_ERR_X509_BUFFER_TOO_SMALL)

AdityaHPatwardhan, 2021-08-23:
Hi @dannybackx I may be mistaken is this a change for the file named `x509_crt.c` which is part of the `mbedtls` submodule ?

dannybackx, 2021-08-23:
Apologies, my original message lacks info. You're right, that is the source file I mean.

Please note that my use case (with "invalid PK" as parameter) requires a larger allocation as I indicate, but I've not scanned the sources for other cases.

The problem is also present in esp-idf-v4.3 : 

hp: {8} diff ./mbedtls/library/x509_crt.c.orig ./mbedtls/library/x509_crt.c 
1559,1560c1559,1560
< #define BEFORE_COLON    18
< #define BC              "18"
---
> #define BEFORE_COLON    20
> #define BC              "20"

AdityaHPatwardhan, 2021-08-23:
Okay,  thanks for clarifying. As the `x509_crt.c` is a part of the `mbedtls` submodule. I don't think we can make any changes here. If you think it is a bug in `mbedtls` implementation, can you please raise the issue in https://github.com/ARMmbed/mbedtls

closed 2021-09-02 by Alvin1Zhang