espressif/esp-idf #7453: Appending questions to MDNS responses breaks Avahi-Daemon

d8ahazard, 2021-08-21:
This is to loop in this library on the discussion being had here:

https://github.com/Aircoookie/WLED/issues/2009#issuecomment-902983255

and here:

https://github.com/lathiat/avahi/issues/348

Basically, the esp-mdns implementation is echoing th question message in responses, which causes avahi-daemon to discard the response as invalid. 

I can see this happening here at line 1085:

https://github.com/espressif/esp-idf/blob/master/components/mdns/mdns.c

Apparently, the proper approach as per the RFC for MDNS is to ignore these responses, so it would stand to reason that the behavior on this side is fine, but I just wanted to link it to the other issues for completeness.

negativekelvin, 2021-08-21:
https://github.com/espressif/esp-idf/issues/7124

david-cermak, 2021-08-23:
@negativekelvin Thanks for the reference

@d8ahazard This bug was fixed as
* f601cb0 on `v4.4`
* 3a588d7 on `v4.3`
* 93921e0 on `v4.2`

Please update the IDF version to include the above commits. It's also possible to work around this bug setting 
* [CONFIG_MDNS_STRICT_MODE](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/kconfig.html#config-mdns-strict-mode) to `y`

david-cermak, 2021-09-08:
@d8ahazard Could you please close the issue?

closed 2021-09-10 by Alvin1Zhang