espressif/esp-idf #7454: Smartconfig causing "pll_cap_ext" error && eventually abort()

RoscoeTheDog, 2021-08-21:
Hello esp friends. I have a project  setup to toggle a switch on my esp device and interrupt into a "pairing mode" for the smartconfig wifi feature. When the switch is toggled, an ISR notifies a task configures and invokes the initializer esp_smartconfig_start() method (but only if the radio is initialized first). The service runs successfully but then immediately after the console is flooded with "pll_cap_ext" errors and aborts, restarting the system in the process. This was also an issue on the older version of IDF 4.2.1. When I updated to the newest version from the repo it did not resolve the issue. Note that this project is built in the platformIO environment the error may be related to the espidf framework itself and I was referred to post this bug here. 

```
Build:Oct 25 2019
rst:0x1 (POWERON),boot:0x8 (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3ffe6100,len:0x1900
load:0x4004c000,len:0x9e8
load:0x40050000,len:0x2b70
entry 0x4004c1ec
␛[0;32mI (21) boot: ESP-IDF 4.3.0 2nd stage bootloader␛[0m
␛[0;32mI (21) boot: compile time 15:36:43␛[0m
␛[0;32mI (21) boot: chip revision: 0␛[0m
␛[0;32mI (24) boot.esp32s2: SPI Speed      : 80MHz␛[0m
␛[0;32mI (29) boot.esp32s2: SPI Mode       : DIO␛[0m
␛[0;32mI (33) boot.esp32s2: SPI Flash Size : 4MB␛[0m
␛[0;32mI (38) boot: Enabling RNG early entropy source...␛[0m
␛[0;32mI (43) boot: Partition Table:␛[0m
␛[0;32mI (47) boot: ## Label            Usage          Type ST Offset   Length␛[0m
␛[0;32mI (54) boot:  0 nvs              WiFi data        01 02 00009000 00006000␛[0m
␛[0;32mI (62) boot:  1 phy_init         RF data          01 01 0000f000 00001000␛[0m
␛[0;32mI (69) boot:  2 factory          factory app      00 00 00010000 00100000␛[0m
␛[0;32mI (77) boot: End of partition table␛[0m
␛[0;32mI (81) esp_image: segment 0: paddr=00010020 vaddr=3f000020 size=21e4ch (138828) map␛[0m
␛[0;32mI (119) esp_image: segment 1: paddr=00031e74 vaddr=3ffc83e0 size=04098h ( 16536) load␛[0m
␛[0;32mI (123) esp_image: segment 2: paddr=00035f14 vaddr=40022000 size=0a104h ( 41220) load␛[0m
␛[0;32mI (136) esp_image: segment 3: paddr=00040020 vaddr=40080020 size=a1e04h (663044) map␛[0m
␛[0;32mI (279) esp_image: segment 4: paddr=000e1e2c vaddr=4002c104 size=0c2dch ( 49884) load␛[0m
␛[0;32mI (292) esp_image: segment 5: paddr=000ee110 vaddr=50000000 size=00010h (    16) load␛[0m
␛[0;32mI (304) boot: Loaded app from partition at offset 0x10000␛[0m
␛[0;32mI (304) boot: Disabling RNG early entropy source...␛[0m
␛[0;32mI (316) ca␇␆�f�~␀�␘��.�Սѥ���cache        : size 8KB, 4Ways, cache line size 32Byte␛[0m
␛[0;32mI (316) cpu_start: Pro cpu up.␛[0m
␛[0;32mI (330) cpu_start: Pro cpu start user code␛[0m
␛[0;32mI (330) cpu_start: cpu freq: 160000000␛[0m
␛[0;32mI (330) cpu_start: Application information:␛[0m
␛[0;32mI (335) cpu_start: Project name:     ESP32_S2_Saola1␛[0m
␛[0;32mI (340) cpu_start: App version:      09d31e8-dirty␛[0m
␛[0;32mI (346) cpu_start: Compile time:     Aug 19 2021 15:32:39␛[0m
␛[0;32mI (352) cpu_start: ELF file SHA256:  27a2c3ee744a9e4d...␛[0m
␛[0;32mI (358) cpu_start: ESP-IDF:          4.3.0␛[0m
␛[0;32mI (363) heap_init: Initializing. RAM available for dynamic allocation:␛[0m
␛[0;32mI (370) heap_init: At 3FF9E000 len 00002000 (8 KiB): RTCRAM␛[0m
␛[0;32mI (376) heap_init: At 3FFD18F0 len 0002A710 (169 KiB): DRAM␛[0m
␛[0;32mI (382) heap_init: At 3FFFC000 len 00003A10 (14 KiB): DRAM␛[0m
␛[0;32mI (389) spi_flash: detected chip: generic␛[0m
␛[0;32mI (393) spi_flash: flash io: dio␛[0m
␛[0;32mI (401) cpu_start: Starting scheduler on PRO CPU.␛[0m
␛[0;32mI (403) gpio: GPIO[33]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 1| Intr:0 ␛[0m
␛[0;32mI (403) gpio: GPIO[16]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 0| Pulldown: 1| Intr:0 ␛[0m
␛[0;32mI (413) gpio: GPIO[15]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 0| Pulldown: 1| Intr:0 ␛[0m
␛[0;32mI (423) gpio: GPIO[2]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 1| Intr:0 ␛[0m
␛[0;32mI (433) gpio: GPIO[0]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 1| Intr:0 ␛[0m
␛[0;32mI (443) gpio: GPIO[21]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 0| Pulldown: 1| Intr:0 ␛[0m
␛[0;32mI (453) gpio: GPIO[19]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 1| Intr:0 ␛[0m
␛[0;32mI (463) gpio: GPIO[26]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 0| Pulldown: 1| Intr:0 ␛[0m
␛[0;32mI (473) gpio: GPIO[34]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 ␛[0m
␛[0;32mI (483) gpio: GPIO[35]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 ␛[0m
␛[0;32mI (493) gpio: GPIO[36]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 ␛[0m
␛[0;32mI (503) initialize_ledc_config_0: LEDC FREQUENCY:  100␛[0m
␛[0;32mI (503) initialize_ledc_config_0: LEDC BIT RESOLUTION:  11␛[0m
␛[0;32mI (513) initialize_ledc_config_0: LEDC BIT INTEGER MAX:  2047␛[0m
␛[0;32mI (2523) nvsRestoreSystemState: RESTORING PREVIOUS STATE␛[0m
␛[0;32mI (2533) nvsRestoreSystemState: STATE RESTORED: {"MOTOR_SPEED_RPM":100,"MOTOR_POSITION_STEPS":76965,"CURTAIN_PERCENTAGE":33.404949188232422}␛[0m
␛[0;32mI (3933) initializeRTOSTasks: initializing vTaskRTOSDebug␛[0m
␛[0;32mI (3933) initializeRTOSTasks: initializing vTaskSleep␛[0m
␛[0;32mI (3933) initializeRTOSTasks: initializing vTaskStatusLEDWatchdog␛[0m
␛[0;32mI (3933) initializeRTOSTasks: initializing vTaskLEDFade␛[0m
␛[0;32mI (3943) initializeRTOSTasks: initializing vTaskMoveStepperForward␛[0m
[... 85 lines cut ...]
			if (!RADIO_INITIALIZED) {
				initializeWifi();
			}
			ESP_LOGI(TAG, "configuring smartconfig service");
			ESP_ERROR_CHECK(esp_smartconfig_set_type(SC_TYPE_ESPTOUCH));
			smartconfig_start_config_t cfg = {.enable_log = false};

			ESP_LOGI(TAG, "starting smartconfig service");
			esp_err_t err = esp_smartconfig_start(&cfg);
			ESP_ERROR_CHECK(err);
			if (err == ESP_OK) {
				ESP_SMARTCONFIG_STATUS = true;
				ESP_LOGI(TAG, "smartconfig service started successfully");
			} else {
				ESP_LOGI(TAG, "smartconfig service failed to initialize");
			}
			
			vTaskDelay(1);			
		}

	}
}
```

You can find the full source code of my dev branch [here](https://github.com/RoscoeTheDog/ESP32-S2-Saola-Project/tree/dev). I believe you should be able to do a pull repo and flash it on a esp32s2 using platformIO fairly easily if you have one lying around. I am using the ESP32-S2 devkitm-1 [from here](https://www.amazon.com/gp/product/B08N6BRN4Q/ref=ppx_yo_dt_b_search_asin_title?ie=UTF8&psc=1).

jack0c, 2022-05-20:
#7398

zhangyanjiaoesp, 2022-05-23:
@RoscoeTheDog Please update your IDF version to the latest v4.3, the phy version has update to` I (459) phy_init: phy_version 2300,d67cf06,Feb 10 2022,10:03:07`

closed 2022-07-13 by Alvin1Zhang