espressif/esp-idf #7456: How to add external header directories in Cmake

dimpalsharm, 2021-08-23:
How we can add external header directories "esp-idf\components\esp_system\port\CMakeLists.txt" in Cmake.

******************Code****************************

target_include_directories(${COMPONENT_LIB} PRIVATE include)
set(srcs "panic_handler.c" "panic_handler_asm.S")
add_prefix(srcs "${CMAKE_CURRENT_LIST_DIR}/" ${srcs})

target_sources(${COMPONENT_LIB} PRIVATE ${srcs})

idf_build_get_property(target IDF_TARGET)
add_subdirectory(${target})

Tried adding like below Code but are getting error.
*********************Code*************************
target_include_directories(${COMPONENT_LIB} PRIVATE include)

set(srcs "panic_handler.c" "panic_handler_asm.S")
**set (include_dirs "header.h")**
add_prefix(srcs "${CMAKE_CURRENT_LIST_DIR}/" ${srcs})

target_sources(${COMPONENT_LIB} PRIVATE ${srcs})

idf_build_get_property(target IDF_TARGET)
add_subdirectory(${target})

igrr, 2021-08-23:
@dimpalsharm could you please explain what problem you are trying to solve by adding header files to `esp_system` component?

Alvin1Zhang, 2021-11-22:
@dimpalsharm Thanks for reporting, any updates per https://github.com/espressif/esp-idf/issues/7456#issuecomment-903583702? Thanks.

Alvin1Zhang, 2022-01-25:
Thanks for reporting, will close due to short of feedback, feel free to reopen with more details, thanks.

closed 2022-01-25 by Alvin1Zhang