espressif/esp-idf #7457: Websocket server does not always properly receive frames longer than 2 MTU (1440*2 btytes)

bbrother, 2021-08-23:
----------------------------- Delete below -----------------------------

**Reminder: If your issue is a general question, starts similar to "How do I..", or is related to 3rd party development kits/libs, please discuss this on our community forum at https://esp32.com instead.**

## Environment

- Development Kit:      none
- Module or chip used:  ESP32S2
- IDF version (run ``git describe --tags`` to find it): v4.3-356-g48ae2309fd
- Build System:         idf.py
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it):
    // 1.22.0-80-g6c4433a
- Operating System:     [Windows|Linux|macOS]
- (Windows only) environment type: [MSYS2 mingw32|ESP Command Prompt|Plain Command Prompt|PowerShell].
- Using an IDE?: [No|Yes (please give details)]
- Power Supply:         [USB|external 5V|external 3.3V|Battery]

## Problem Description
the attempt to fix #7202 ([here](https://github.com/espressif/esp-idf/commit/f2a0b48cf9856a411970faa9943a0b9c7464c3ab?branch=f2a0b48cf9856a411970faa9943a0b9c7464c3ab&diff=unified)) will only work if the rest of the websocket is recevied on the second attempt.  Otherwise the offset is wrong and the received data is incorrect.  Simple code inspection shows that this patch causes a new bug. 

The following patch should fix this bug:
```
diff --git a/components/esp_http_server/src/httpd_ws.c b/components/esp_http_server/src/httpd_ws.c
index 629497fd1b..e6957c8ff9 100644
--- a/components/esp_http_server/src/httpd_ws.c
+++ b/components/esp_http_server/src/httpd_ws.c
@@ -334,7 +334,7 @@ esp_err_t httpd_ws_recv_frame(httpd_req_t *req, httpd_ws_frame_t *frame, size_t
 
     int read_len = 0, left_len = frame->len;
     while (left_len > 0) {
-        if ((read_len = httpd_recv_with_opt(req, (char *)frame->payload + read_len, left_len, false)) <= 0) {
+        if ((read_len = httpd_recv_with_opt(req, (char *)frame->payload + (frame->len - left_len), left_len, false)) <= 0) {
             ESP_LOGW(TAG, LOG_FMT("Failed to receive payload"));
             return ESP_FAIL;
         }
```

ESP-YJM, 2021-08-25:
@bbrother Yes, thanks for reporting this bug. We will fix this.

ESP-YJM, 2021-08-25:
```
     int read_len = 0, left_len = frame->len;
     while (left_len > 0) {
-        if ((read_len = httpd_recv_with_opt(req, (char *)frame->payload + read_len, left_len, false)) <= 0) {
+        if ((read_len += httpd_recv_with_opt(req, (char *)frame->payload + read_len, left_len, false)) <= 0) {
             ESP_LOGW(TAG, LOG_FMT("Failed to receive payload"));
             return ESP_FAIL;
         }
```
@bbrother Please help check if it is ok for you. Thanks.

closed 2021-09-01 by espressif-bot