espressif/esp-idf #7458: How to execute a "add_custom_command" every time ?

franz-ms-muc, 2021-08-23:
Hello, 

i want to copy some files to Spiffs. 

`# After build, copy the iop file to spiffs
add_custom_command(TARGET ${COMPONENT_LIB}
                  POST_BUILD
                  COMMAND ${CMAKE_COMMAND} -E copy ${CMAKE_SOURCE_DIR}/components/ISODesigner/MyWorkspace1/MyProject1/Output/MyProject1.iop ${CMAKE_SOURCE_DIR}/spiffs_image/pools
                  COMMENT "Copying MyProject1.iop to spiffs_image...")

# Create a SPIFFS image from the contents of the 'spiffs_image' directory
# that fits the partition named 'storage'. FLASH_IN_PROJECT indicates that
# the generated image should be flashed when the entire project is flashed to
# the target with 'idf.py -p PORT flash'. 
spiffs_create_partition_image(storage ${CMAKE_SOURCE_DIR}/spiffs_image FLASH_IN_PROJECT)`

so far this works, but it is only executed on a fresh build. 

how to modify to make it on every build ? 
OR to modify it to make it when the source File "MyProject1.iop" has changed ? 

Thanks, 
Franz

franz-ms-muc, 2021-08-23:
```
# After build, copy the iop file to spiffs
add_custom_command(TARGET ${COMPONENT_LIB}
                  POST_BUILD
                  COMMAND ${CMAKE_COMMAND} -E copy ${CMAKE_SOURCE_DIR}/components/ISODesigner/MyWorkspace1/MyProject1/Output/MyProject1.iop ${CMAKE_SOURCE_DIR}/spiffs_image/pools
                  COMMENT "Copying MyProject1.iop to spiffs_image...")

# Create a SPIFFS image from the contents of the 'spiffs_image' directory
# that fits the partition named 'storage'. FLASH_IN_PROJECT indicates that
# the generated image should be flashed when the entire project is flashed to
# the target with 'idf.py -p PORT flash'. 
spiffs_create_partition_image(storage ${CMAKE_SOURCE_DIR}/spiffs_image FLASH_IN_PROJECT)
```

igrr, 2021-08-23:
Hi @franz-ms-muc, the following should work with spiffsgen example in ESP-IDF. You can modify this to suit your project layout:
```cmake
idf_component_register(SRCS "spiffsgen_example_main.c"
                    INCLUDE_DIRS ".")

# as an example, add the source file into SPIFFS partition:
set(src_file ${CMAKE_CURRENT_LIST_DIR}/spiffsgen_example_main.c)
set(dst_file ${CMAKE_CURRENT_LIST_DIR}/../spiffs_image/spiffsgen_example_main.c)

add_custom_command(OUTPUT ${dst_file}
                   COMMAND ${CMAKE_COMMAND} -E echo "Copying spiffsgen_example_main.c to spiffs_image..."
                   COMMAND ${CMAKE_COMMAND} -E copy ${src_file} ${dst_file}
                   DEPENDS ${src_file})
add_custom_target(update_spiffs DEPENDS "${dst_file}")

spiffs_create_partition_image(storage ../spiffs_image FLASH_IN_PROJECT DEPENDS update_spiffs)
```

The key points here are:
* use `add_custom_command` with `OUTPUT` specifying the destination, and `DEPENDS` specifying the source,
* use `add_custom_target` to define a CMake target around the custom command,
* use `DEPENDS` argument of `spiffs_create_partition_image` to add a dependency of the copying step.

franz-ms-muc, 2021-08-23:
should work also for non-C Files ? as one cannot register a binary File as source ? but i will try, yes ! Tanks so far ..

igrr, 2021-08-24:
The choice of `src_file` in my example is completely arbitrary. You can change `src_file` and `dst_file` to any other file, the rest of the steps should work as is.

Alvin1Zhang, 2021-08-31:
@franz-ms-muc Thanks for reporting. Would you please help share if any updates? Thanks.

Alvin1Zhang, 2021-09-07:
@franz-ms-muc Thanks for reporting, would you please help share if any updates for the issue? Thanks.

franz-ms-muc, 2021-09-07:
CONFIRM this works good !

franz-ms-muc, 2021-09-07:
close !

closed 2021-09-07 by franz-ms-muc