espressif/esp-idf #7459: Problem with using external arduino library!

Zaki23madrid, 2021-08-23:
I am always facing arduino library problem!
Actually, I have a smart servo motor that needs to work with it's specific arduino library! the problem is that i can't add the external arduino library with esp idf firmware! the compiler always tells me that there is no such file for Arduino.h and WProgram.h headers that the arduino library uses...
it's been more than 20 days since i started working on that problem but in vain...

there is another alternative, they told me to add arduino as a component, but believe me, it's frustrating, i can't get this work! it's very complicated and every time i try to build my project i get thousands of errors..., when i try to google them, there is hardly any documentation...

i am using Vs code and I also have eclipse but i couldn't it make it work with both...

Thank you!

VojtechBartoska, 2021-09-01:
Hello @Zaki23madrid,

for using Arduino as component you can take a look on [Arduino ESP32 Documentation](https://docs.espressif.com/projects/arduino-esp32/en/latest/esp-idf_component.html).

Zaki23madrid, 2021-09-14:
did not work, alot of errors occured

VojtechBartoska, 2021-11-02:
@Zaki23madrid Can you share your code and displayed error messages, please?