espressif/esp-idf #7460: PCNT Example in the IDF is not working correctly

aircable, 2021-08-23:
## Environment

- Development Kit:     custom

- Module or chip used:  ESP32-PICO-V3
- IDF version: v4.4-dev-2594-ga20df743f1 (but also V4.2 release)
- Build System:   IDF
- Compiler version: xtensa-esp32-elf-gcc (crosstool-NG esp-2021r1) 8.4.0
- Operating System:   Linux

## Problem Description

The upper and lower limits when configuring pcnt_config_t are not handled correctly
``
   pcnt_config_t dev_config = {
        .pulse_gpio_num = config->phase_a_gpio_num,
        .ctrl_gpio_num = config->phase_b_gpio_num,
        .channel = PCNT_CHANNEL_0,
        .unit = ec11->pcnt_unit,
        .pos_mode = PCNT_COUNT_DEC,
        .neg_mode = PCNT_COUNT_INC,
        .lctrl_mode = PCNT_MODE_REVERSE,
        .hctrl_mode = PCNT_MODE_KEEP,
        .counter_h_lim = EC11_PCNT_DEFAULT_HIGH_LIMIT,
        .counter_l_lim = EC11_PCNT_DEFAULT_LOW_LIMIT,
    };
``

### Expected Behavior

### Actual Behavior

Since PCNT_MODE_KEEP is being used when the counter reaches the upper or lower limits (counter_x_lim) the counter should not change and stay at the high limit.

### Steps to reproduce

1. configure example 
2. ...

// If possible, attach a picture of your setup/wiring here.

### Code to reproduce this issue

```cpp
// the code should be wrapped in the ```cpp tag so that it will be displayed better.
#include "esp_log.h"

void app_main()
{

}

```
// If your code is longer than 30 lines, [GIST](https://gist.github.com) is preferred.

## Debug Logs

```
Debug log goes here, should contain the backtrace, as well as the reset source if it is a crash.
Please copy the plain text here for us to search the error log. Or attach the complete logs but leave the main part here if the log is *too* long.
```

## Other items if possible

- [ ] sdkconfig file (attach the sdkconfig file from your project folder)
- [ ] elf file in the ``build`` folder (**note this may contain all the code details and symbols of your project.**)
- [ ] coredump (This provides stacks of tasks.)

aircable, 2021-08-24:
WORKAROUND:

define the limits like this:
```
#define EC11_PCNT_DEFAULT_HIGH_LIMIT (1)
#define EC11_PCNT_DEFAULT_LOW_LIMIT  (-1)
```
Then you get interrupts whenever increment or decrement of one count is being recognized.
Implement the interrupt routine like this, with your own global variable `int rotary_value`.
```
static void ec11_pcnt_overflow_handler(void *arg)
{
    if (status & PCNT_EVT_H_LIM) {
        rotary_value++;
    } else if (status & PCNT_EVT_L_LIM) {
        rotary_value--;
    } 
}
```
Manage upper and lower limit in your application code and you can use the PCNT for your rotary encoder.
```
    // limit the values 
    if( rotary_value > ROTARY_MAX_VALUE ) rotary_value = ROTARY_MAX_VALUE;
    if( rotary_value <= ROTARY_MIN_VALUE ) rotary_value = ROTARY_MIN_VALUE;
```

SUGGESTION FOR ESPRESSIF DEVELOPERS:

Since the HAL implementation of the PCNT is not available for normal mortals, just remove the values the PCNT manages, including the managing of the upper and lower limits, and just provide events for the users, so they can manage whatever their application needs. I bet it's simpler and more flexible.

CHEERS
Juergen

suda-morris, 2021-08-24:
> Since PCNT_MODE_KEEP is being used when the counter reaches the upper or lower limits (counter_x_lim) the counter should not change and stay at the high limit.

I don't think the hardware will act in this way. `PCNT_MODE_KEEP` is saying `keep the current count mode` instead of `hold the count value`. 
If the limit value is reached, it will returns back to zero, it is a hardware behavior.

aircable, 2021-08-24:
So, you mean, for a rotary encoder implementation, the original example is useless? Jumping back to zero at limit is not what a rotary encoder example would want, correct? Think of e.g. a volume control rotary button.

How would you propose to implement the example then?

suda-morris, 2021-08-25:
you can make use of the threshold and/or limit interrupt, where you can handle the overflow or saturation by software. The original rotary example is useful for measuring DC motor speed as far as I can tell. See https://github.com/espressif/esp-idf/tree/master/examples/peripherals/mcpwm/mcpwm_brushed_dc_control

Alvin1Zhang, 2021-11-22:
Thanks for reporting, feel free to reopen.

closed 2021-11-22 by Alvin1Zhang

aaaaahaaaaa, 2024-04-01:
Facing the same challenge. I have to say that the API doesn't make a lot of sense to me either in terms of value proposition. If the range forbids 0 as a lower bound, always reset to 0 in both direction AND/OR doesn't allow to hold the counter value, then what is the point of configuring the range at all?