espressif/esp-idf #7461: Problem with sending audio stream with HTTPS/2 connection

michalber, 2021-08-24:
## Environment

- Development Kit:      custom based on ESP32-LyraT
- Module or chip used:  ESP32-WROVER-B
- IDF version v4.3
- ADF version v2.3
- Build System:         CMake|idf.py
- Compiler version (crosstool-NG esp-2020r3) 8.4.0
- Operating System:     Linux
- Using an IDE?: Yes, VSCode (without ESP extension)
- Power Supply:         external 3.3V

## Problem Description

In my application I need to use HTTP2 connection and recently I started updating libraries from IDF v3.3.2 (ADF 2.1) to IDF 4.3 (ADF 2.3). I had problem with sending audio data stream to cloud (in 20ms chunks - 640 bytes). I am using nghttp2, sh2lib and esp-tls (with mbedtls). When sending stream data after few chunks connection is closed with error -902 from sh2lib. Before sending audio data I am making connection and doing one request which is not stream.

### Expected Behaviour

Sending stream data should work without any problems.

### Actual Behaviour

When sending stream data after few chunks connection is closed with error -902 from sh2lib. Before sending audio data I am making connection and doing one request which is not stream. Also I managed to suppress this errors by adding extra delays between each `sh2lib_execute` but this slow down connection significantly, so it is not suitable for me.

## Debug Logs

```
I (09:34:47.497) SH2Lib Client: Already connected to ----------------------------------------------- server
I (09:34:47.512) SH2Lib Client: Do POST request to -----------------------------------------------
I (09:34:47.525) SH2Lib Client: Process send/receive HTTP2 Request
I (09:34:47.534) sh2lib: [frame-send] frame type HEADERS
I (09:34:47.539) sh2lib: [frame-send] C ----------------------------> S (HEADERS)
I (09:34:47.547) sh2lib: [frame-send] headers nv-len = 5
I (09:34:47.553) sh2lib: [frame-send] :method : POST
I (09:34:47.558) sh2lib: [frame-send] :scheme : https
I (09:34:47.564) sh2lib: [frame-send] :authority : -----------------------------------------------
I (09:34:47.571) sh2lib: [frame-send] :path : -----------------------------------------------
I (09:34:47.580) sh2lib: [frame-send] content-type : application/octet-stream
I (09:34:47.592) sh2lib: [frame-send] frame type DATA
I (09:34:47.597) sh2lib: [frame-send] frame type DATA
I (09:34:47.602) sh2lib: [frame-send] frame type DATA
I (09:34:47.619) sh2lib: [frame-send] frame type DATA
I (09:34:47.623) sh2lib: [frame-send] frame type DATA
I (09:34:47.627) sh2lib: [frame-send] frame type DATA
I (09:34:47.637) sh2lib: [frame-send] frame type DATA
I (09:34:47.642) sh2lib: [frame-send] frame type DATA
E (09:34:47.646) sh2lib: [sh2-execute] HTTP2 session send failed -902
E (09:34:47.650) SH2Lib Client: Error in send/receive
E (09:34:47.661) Web Service: Error in doing Send Activation Request
```

## Solution

I found out that -902 error is coming from [esp_mbedtls_write function](https://github.com/espressif/esp-idf/blob/8e3e65a47b7d9b5dc4f52eb56660a748fda1884e/components/esp-tls/esp_tls_mbedtls.c#L187). I changed `return written` to `return ret` and the problem is gone. Is this a typo in library or all is good and my problem is somewhere else?

AdityaHPatwardhan, 2021-09-12:
Hi @michalber 

```

 * @return
 *             - >=0  if write operation was successful, the return value is the number
 *                   of bytes actually written to the TLS/SSL connection.
 *             - <0  if write operation was not successful, because either an
 *                   error occured or an action must be taken by the calling process.
 */
```

The `written` value corresponds to the number of bytes written to the TLS/SSL connection.
I don't think the problem is in `esp_mbedtls_write`, but maybe in the  use of the esp_mbedtls_write function in the http2 support.

AdityaHPatwardhan, 2021-09-12:
Hi @michalber, . 
Thank you very much for the issue.
I think there might be this problem in esp_mbedtls_write. Since when the `ret <= 0` there is no need to return the `written` field and should return only what the `mbedtls_ssl_write` function has returned.

michalber, 2021-09-13:
Hi @AdityaHPatwardhan , thank you so much for the response.
I don't think there was problem with `mbedtls_ssl_write` function because when I spot this problem I had error -902 (`NGHTTP2_ERR_CALLBACK_FAILURE`) in callback_send_inner from sh2lib (code below), which means that `esp_tls_conn_write` returned something different than ` ESP_TLS_ERR_SSL_WANT_READ || rv == ESP_TLS_ERR_SSL_WANT_WRITE`. I had backtraced this error and got to `esp_mbedtls_write` where `mbedtls_ssl_write` returned `MBEDTLS_ERR_SSL_WANT_WRITE` so `esp_mbedtls_write` returned `written` field which was 0. Changing `return written` to `return ret` solve the problem because `callback_send_inner` got correct error and it was able to continue sending data.

```
static ssize_t callback_send_inner(struct sh2lib_handle *hd,
                                    const uint8_t *data, size_t length) {
    int rv = esp_tls_conn_write(hd->http2_tls, data, length);
    if (rv <= 0) {
        if (rv == ESP_TLS_ERR_SSL_WANT_READ || rv == ESP_TLS_ERR_SSL_WANT_WRITE) {
            rv = NGHTTP2_ERR_WOULDBLOCK;
        } else {
            rv = NGHTTP2_ERR_CALLBACK_FAILURE;
        }
    }
    return rv;
}
```

This problem was hard to spot because in my case it was occuring only when I was sending audio data in small chunks with small delay between each of `sh2lib_execute` call.

AdityaHPatwardhan, 2021-09-13:
Hi @michalber 
> I don't think there was problem with mbedtls_ssl_write function
I was referring to the `esp_mbedtls_write` and not `mbedtls_ssl_write`.

so changing `return written` to `return ret` fixes your issue, right ?

michalber, 2021-09-13:
> I was referring to the esp_mbedtls_write and not mbedtls_ssl_write.

Yes, you are right, sorry.

> so changing return written to return ret fixes your issue, right ?

Exactly

AdityaHPatwardhan, 2021-09-14:
@michalber 
Can you please check if 
`return (written > 0) ? written: ret;` 
Fixes your issue ?
I am afraid keeping it to `return ret` would cause problems in case of partial write !.

michalber, 2021-09-15:
@AdityaHPatwardhan 
With `return (written > 0) ? written: ret;` everything is working the same.

AdityaHPatwardhan, 2021-09-16:
@michalber I have updated the internal codebase, it should be updated on GitHub in a couple of days,
Thanks for your contribution in fixing this.

closed 2021-09-24 by espressif-bot