espressif/esp-idf #7464: esp_http_client: impossible to change the port

caipiblack, 2021-08-24:
## Environment

- Development Kit: none
- Module or chip used:  ESP32
- IDF version: Release 4.3
- Build System: idf.py
- Compiler version: tensa-esp32-elf-gcc (crosstool-NG esp-2020r3) 8.4.0
- Operating System: Windows
- (Windows only) environment type: ESP Command Prompt
- Using an IDE?: No
- Power Supply:         USB

## Problem Description

When we use the esp_http_client the port configured with the function `esp_http_client_init()` is not used by the client.

The client always use "80" for "http" and "443" for "https" transport

The documentation say's:

```
esp_http_client_init(): To use the HTTP client, the first thing we must do is create an esp_http_client by pass into this function with the esp_http_client_config_t configurations. Which configuration values we do not define, the library will use default.
```

So, the values that we define should be used.

### Expected Behavior

The port given to the function `esp_http_client_init()` with the configuration structure `esp_http_client_config_t` must be used by the http client.

### Actual Behavior

The port is always replaced by 80 or 443

### Code to reproduce this issue

```cpp
void http_test()
{
    esp_http_client_config_t config = {
        .user_agent = "My Agent",
        .timeout_ms = 5000,
        .url = "http://127.0.0.1/my_path",
        .port = 9595,
        .host = "127.0.0.1"
    };
    esp_http_client_handle_t client = esp_http_client_init(&config);
    esp_http_client_open(client, 0);
    // ...
}
```

## Other items if possible

The port is replaced here: 

[image]

atanisoft, 2021-08-24:
Your url should also have the port declared in it to be valid. If you put that URL into a browser it will also use port 80/443 since it does not declare a specific port to use.

Try:
```
esp_http_client_config_t config = {
        .user_agent = "My Agent",
        .timeout_ms = 5000,
        .url = "http://127.0.0.1:9995/my_path",
        .port = 9595,
        .host = "127.0.0.1"
    };
```

caipiblack, 2021-08-24:
> Your url should also have the port declared in it to be valid. If you put that URL into a browser it will also use port 80/443 since it does not declare a specific port to use.
>
> Try:
>
> ``` [...]

Problem solved.

closed 2021-08-24 by caipiblack

chegewara, 2021-08-24:
You can use:
```
        .url = "http://127.0.0.1:9995/my_path",
```
or
```
        .path = "my_path",
        .port = 9595,
        .host = "127.0.0.1"
```

Alvin1Zhang, 2021-08-25:
@caipiblack Thanks for reporting, thanks @atanisoft @chegewara for the notes.