espressif/esp-idf #7467: DAC output is off

aaly00, 2021-08-24:
## Environment

- Development Kit: ESP32S2-DevKitC
- IDF version (run ``git describe --tags`` to find it):
    // v3.2-dev-1148-g96cd3b75c
- Build System:         [|CMake|idf.py]
- Operating System:     [Windows||]
- Using an IDE?: [|Yes (please give details)] VSCODE
- Power Supply:         [|external 5V|]

## Problem Description

I used [Digital To Analog Converter](https://docs.espressif.com/projects/esp-idf/en/latest/esp32s2/api-reference/peripherals/dac.html) to setup my code. The DAC value is wrong by two decimal places. When I output 3.0, I'd get 3.2. How can I fix this?

krzychb, 2021-09-01:
Hi @aaly00, 
How do you set DAC to "output 3.0"? 
What is the voltage measured on pin 3V3 of your board?

aaly00, 2021-09-02:
I am using the development board. I connected 5VDC to VCC. The input on the 3V3 pin is ~3.3V. 
Iused this from the docs `Setting DAC channel 1 (GPIO17) voltage to approx 0.78 of VDD_A voltage (VDD * 200 / 255). For VDD_A 3.3V, this is 2.59V:`. 
I assumed since 3.3*200/255 = 2.59V. 
Then Vout = VDD * PWM/255. PWM = 255*Vout/VDD. 
Since I know VDD and Vout. 
PWM = 255*3.0V/3.3V = 232 
The value that I got was off by around 0.2

krzychb, 2021-09-03:
From what I see DAC has some offset at both [minimum](https://esp32.com/viewtopic.php?t=1339) and [maximum](https://esp32.com/viewtopic.php?t=5098) output. 
I would first measure the minimum and the maximum voltage I can get:
```
dac_output_voltage(DAC_CHANNEL_1, 0);
dac_output_voltage(DAC_CHANNEL_1, 255);
```
and then use the values to calibrate the output voltage.

closed 2024-07-08 by Alvin1Zhang