espressif/esp-idf #7468: Unknow WiFi event (21) - outside enum list

hassandraga, 2021-08-24:
## Environment

- Development Kit: -
- Kit version -
- Module or chip used:  [ESP32-WROOM-32U]
- IDF version v4.4-dev-1254-g639e7ad49
- Build System: [idf.py]
- Compiler version: xtensa-esp32-elf-gcc (crosstool-NG esp-2020r3) 8.4.0
- Operating System: [Windows/Linux]
- (Windows only) environment type: [Plain Command Prompt].
- Using an IDE?: [No]
- Power Supply: [24V DC]

## Problem Description

WiFi event (21) is outside enum list.

### Expected Behavior

WiFi event between 1 - 16.

### Actual Behavior

WiFi event 21.

### Steps to reproduce

1. Run WiFi example, connect to an AP.
2. https://github.com/espressif/esp-idf/blob/master/examples/wifi/getting_started/station/main/station_example_main.c

### Code to reproduce this issue

```cpp
static void event_handler(void* arg, esp_event_base_t event_base, int32_t event_id, void* event_data){

    if (event_base == WIFI_EVENT){
    
        printf("WiFi event: %d", event_id);
    }
}
```

## Debug Logs

```
WiFi event: 21
```

hassandraga, 2021-08-24:
One minute after creating this issue I found that Master branch has the new enum: 
https://github.com/espressif/esp-idf/blob/master/components/esp_wifi/include/esp_wifi_types.h#L576

Nevermind.

closed 2021-08-24 by hassandraga

Alvin1Zhang, 2021-08-25:
Good to know your issue resolved.