espressif/esp-idf #7469: pppos restart failed

achao1104, 2021-08-25:
Hello,

I referred the pppos sample, it works fine. but if I deinit and then start again, it reports the following exception, I have used the deinit codes in the sample.

I (59650) ppposutil: PPP state changed event 259
I (59650) ppposutil: PPP state changed event 262
I (59650) ppposutil: PPP state changed event 263
I (59650) ppposutil: PPP state changed event 265
I (59670) esp-netif_lwip-ppp: Connected
I (59670) esp-netif_lwip-ppp: Name Server1: 218.4.4.4
I (59670) esp-netif_lwip-ppp: Name Server2: 218.2.2.2
Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.

Core  0 register dump:
PC      : 0x400a3ede  PS      : 0x00060230  A0      : 0x800a402c  A1      : 0x3ffdd620
0x400a3ede: netif_do_set_ipaddr at D:/project/esp32/es-idf-v4.3-release/components/lwip/lwip/src/core/netif.c:473 (discriminator 2)

A2      : 0x00000000  A3      : 0x3f01383c  A4      : 0x3ffdd660  A5      : 0x3ffdd704
A6      : 0x3ffde210  A7      : 0x00000000  A8      : 0x00000000  A9      : 0x00000000
A10     : 0x3ffe760c  A11     : 0x00000000  A12     : 0x3ff9ffa8  A13     : 0x00000000
A14     : 0x00000000  A15     : 0x00000003  SAR     : 0x0000001c  EXCCAUSE: 0x0000001c
EXCVADDR: 0x00000004  LBEG    : 0x3ff9ffa8  LEND    : 0x00000000  LCOUNT  : 0x4002771d
0x4002771d: _xt_user_exc at D:/project/esp32/es-idf-v4.3-release/components/freertos/port/xtensa/xtensa_vectors.S:627

Backtrace:0x400a3edb:0x3ffdd620 0x400a4029:0x3ffdd660 0x4009e42b:0x3ffdd6a0 0x4009e06f:0x3ffdd6c0 0x40038049:0x3ffdd6e0 0x40038028:0x3ffdd700 0x400322d9:0x3ffdd730
0x400a3edb: netif_do_set_ipaddr at D:/project/esp32/es-idf-v4.3-release/components/lwip/lwip/src/core/netif.c:470 (discriminator 1)

0x400a4029: netif_set_addr at D:/project/esp32/es-idf-v4.3-release/components/lwip/lwip/src/core/netif.c:702

0x4009e42b: esp_netif_up_api at D:/project/esp32/es-idf-v4.3-release/components/esp_netif/lwip/esp_netif_lwip.c:1215

0x4009e06f: esp_netif_api_cb at D:/project/esp32/es-idf-v4.3-release/components/esp_netif/lwip/esp_netif_lwip.c:107

0x40038049: tcpip_thread_handle_msg at D:/project/esp32/es-idf-v4.3-release/components/lwip/lwip/src/api/tcpip.c:168

0x40038028: tcpip_thread at D:/project/esp32/es-idf-v4.3-release/components/lwip/lwip/src/api/tcpip.c:154

0x400322d9: vPortTaskWrapper at D:/project/esp32/es-idf-v4.3-release/components/freertos/port/xtensa/port.c:168

Thanks,
Peter

david-cermak, 2021-08-25:
Hi Peter,

How do you deinitialize the client? Assuming these [3 lines](https://github.com/espressif/esp-idf/blob/release/v4.3/examples/protocols/pppos_client/main/pppos_client_main.c#L323-L326) should work provided you correctly reinitialize the ppp glue and netif itself. Could you please share your init and deinit code?

Please, share also the exact commit in ESP-IDF, you're using. We had an issue with deinitialization, but that was fixed in 6f1b8fe062ec098bb48279a0eb92bdf8fc4b8e90 (should be included in `v4.3` though)

david-cermak, 2021-08-31:
> BTW, where is the USB-4G sample, it looks the previous link is not accessable.

You mean a component manager's link to the `esp_modem` component, or which link? No (official) USB example for the PPP/modem component has been made public yet.

closed 2021-08-31 by achao1104

leeebo, 2021-09-06:
@achao1104 @david-cermak We implemented a preliminary `usb_4g_module` example based on esp-modem (replace UART with the USB I/O interface), You can find it here [usb_cdc_4g_module](https://github.com/espressif/esp-iot-solution/tree/usb/add_usb_solutions/examples/usb/host/usb_cdc_4g_module). P.S. please checkout IDF to b63ec47238fd6a as the master branch has changed some USB header files.