espressif/esp-idf #7470: esp_wifi_start() does not keep the set channel

jonasfehr, 2021-08-25:
## Environment
- Module or chip used:  [ESP32-WROOM-32E]
- IDF: I was mainly working with arduinoESP32 (IDF 3.3.5) but a brief check with the newest branch 2.0.0.0-rc2 (IDF 4.4dev) which did not work either.
- Build System: PlatformIO
- 
## Problem Description

I work on a project where I use ESPNow and light sleep.
Due to some channel issues we had to change the channel of the wifi (station mode) to 11.
```
WiFi.mode(WIFI_STA); // Arduino object for wifi setup
esp_wifi_set_promiscuous(true);
esp_wifi_set_channel(channel, WIFI_SECOND_CHAN_NONE);
esp_wifi_set_promiscuous(false);
esp_now_init();
...
```
I did following to enter lightsleep:
```
esp_now_stop();
esp_light_sleep_start();
esp_now_start();
```
esp would send, but not successfully and I could not figure out why.
It took me quite some time to figure out, but setting the channel again would solve the issue.
```
esp_now_stop();
esp_light_sleep_start();
esp_now_start();
esp_wifi_set_promiscuous(true);
esp_wifi_set_channel(11, WIFI_SECOND_CHAN_NONE);
esp_wifi_set_promiscuous(false);
```
**I suggest that esp_now_start() would also start with the right channel setting?**

### Expected Behavior
Not to have to set the channel again.

### Actual Behavior
Channel was not part of the reloaded configuration and hat to be set again.

Villaquiranm, 2021-08-28:
Hello, 
I was trying to reproduce this issue modifying the light_sleep project present on the examples folder but I was unable to have your behavior. Do you see something I'm forgetting to set or do ?
This is mi main code:
```c
void app_main(void)
{
    while (true) {
        /* Wake up in 5 seconds, or when button is pressed */
        esp_sleep_enable_timer_wakeup(5000000);
        esp_sleep_enable_gpio_wakeup();
        esp_err_t ret = nvs_flash_init();
        if (ret == ESP_ERR_NVS_NO_FREE_PAGES || ret == ESP_ERR_NVS_NEW_VERSION_FOUND) {
            ESP_ERROR_CHECK(nvs_flash_erase());
            ret = nvs_flash_init();
        }
        ESP_ERROR_CHECK(ret);
        wifi_init_config_t cfg = WIFI_INIT_CONFIG_DEFAULT();
        uint8_t primary = -1;
        wifi_second_chan_t secondary;
        ESP_ERROR_CHECK(esp_wifi_init(&cfg));
        printf("[DEBUG] Wifi initialized without problems...\n");
        ESP_ERROR_CHECK(esp_wifi_start());
        printf("[DEBUG] Wifi started without problems...\n");
        ESP_ERROR_CHECK(esp_wifi_set_mode(WIFI_MODE_STA));
        printf("[DEBUG] Wifi set mode STA\n");
        ESP_ERROR_CHECK(esp_wifi_set_promiscuous(true));
        printf("[DEBUG] Wifi setting promiscuous = true\n");
        ESP_ERROR_CHECK(esp_wifi_get_channel(&primary, &secondary));
        printf("[DEBUG] Retrieved channel before setting channel: %d\n", primary);
        ESP_ERROR_CHECK(esp_wifi_set_channel(11, WIFI_SECOND_CHAN_NONE));
        printf("[DEBUG] Wifi setting channel  = 11\n");
        ESP_ERROR_CHECK(esp_wifi_set_promiscuous(false));
        printf("[DEBUG] Wifi setting promiscuous = false\n");
        ESP_ERROR_CHECK(esp_now_init());
        printf("[DEBUG] Wifi esp_now initialized\n");
        ESP_ERROR_CHECK(esp_wifi_get_channel(&primary, &secondary));
        printf("[DEBUG] Retrieved channel after setting it : %d\n", primary);
        /* To make sure the complete line is printed before entering sleep mode,
         * need to wait until UART TX FIFO is empty:
         */
        uart_wait_tx_idle_polling(CONFIG_ESP_CONSOLE_UART_NUM);
        /* Enter sleep mode */
        ESP_ERROR_CHECK(esp_wifi_stop());
        printf("[DEBUG] Wifi stopped without problems...\n");
        printf("[DEBUG] Entering light sleep...\n");
        ESP_ERROR_CHECK(esp_light_sleep_start());
        printf("[DEBUG] ESP module woke up...\n");
        ESP_ERROR_CHECK(esp_wifi_start());
        printf("[DEBUG] Wifi started without problems...\n");
        printf("[DEBUG] Retrieved channel after sleeping: %d\n", primary);
    }

}
```
Down the logs I got:
```
entry 0x40080680
I (27) boot: ESP-IDF v4.3-356-g48ae2309fd-dirty 2nd stage bootloader
I (27) boot: compile time 14:09:20
I (27) boot: chip revision: 1
I (32) boot_comm: chip revision: 1, min. bootloader chip revision: 0
I (39) boot.esp32: SPI Speed      : 40MHz
I (43) boot.esp32: SPI Mode       : DIO
I (48) boot.esp32: SPI Flash Size : 2MB
I (52) boot: Enabling RNG early entropy source...
I (58) boot: Partition Table:
I (61) boot: ## Label            Usage          Type ST Offset   Length
I (69) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (76) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (84) boot:  2 factory          factory app      00 00 00010000 00100000
[... 59 lines cut ...]
I (649) phy_init: phy_version 4670,719f9f6,Feb 18 2021,17:07:07
I (799) wifi:mode : sta (24:0a:c4:58:c5:e0)
I (799) wifi:enable tsf
[DEBUG] Wifi started without problems...
[DEBUG] Wifi set mode STA
I (799) wifi:ic_enable_sniffer
[DEBUG] Wifi setting promiscuous = true
[DEBUG] Retrieved channel before setting channel: 1
[DEBUG] Wifi setting channel  = 11
I (819) wifi:ic_disable_sniffer
[DEBUG] Wifi setting promiscuous = false
I (819) ESPNOW: espnow [version: 1.0] init
[DEBUG] Wifi esp_now initialized
[DEBUG] Retrieved channel after setting it : 11
I (849) wifi:flush txq
I (849) wifi:stop sw txq
I (849) wifi:lmac stop hw txq
[DEBUG] Wifi stopped without problems...
[DEBUG] Entering light sleep...
[DEBUG] ESP module woke up...
I (859) wifi:mode : sta (24:0a:c4:58:c5:e0)
I (859) wifi:enable tsf
[DEBUG] Wifi started without problems...
[DEBUG] Retrieved channel after sleeping: 11
```

Villaquiranm, 2021-08-28:
Hello I was able to reproduce your behavior, using the function `esp_wifi_get_channel` I was able to fetch the configured channel. After waking up it seems the module getting the channel set to **1** as you described.

I've modified the example of light_sleep present on the examples folder.
My current ESP-IDF version is 4.3.

This is my final main code:
```c
void app_main(void)
{
    while (true) {
        /* Wake up in 5 seconds, or when button is pressed */
        esp_sleep_enable_timer_wakeup(5000000);
        esp_sleep_enable_gpio_wakeup();
        esp_err_t ret = nvs_flash_init();
        if (ret == ESP_ERR_NVS_NO_FREE_PAGES || ret == ESP_ERR_NVS_NEW_VERSION_FOUND) {
            ESP_ERROR_CHECK(nvs_flash_erase());
            ret = nvs_flash_init();
        }
        ESP_ERROR_CHECK(ret);
        wifi_init_config_t cfg = WIFI_INIT_CONFIG_DEFAULT();
        uint8_t primary = -1;
        wifi_second_chan_t secondary;
        ESP_ERROR_CHECK(esp_wifi_init(&cfg));
        printf("[DEBUG] Wifi initialized without problems...\n");
        ESP_ERROR_CHECK(esp_wifi_start());
        printf("[DEBUG] Wifi started without problems...\n");
        ESP_ERROR_CHECK(esp_wifi_set_mode(WIFI_MODE_STA));
        printf("[DEBUG] Wifi set mode STA\n");
        ESP_ERROR_CHECK(esp_wifi_set_promiscuous(true));
        printf("[DEBUG] Wifi setting promiscuous = true\n");
        ESP_ERROR_CHECK(esp_wifi_get_channel(&primary, &secondary));
        printf("[DEBUG] Retrieved channel before setting channel: %d\n", primary);
        ESP_ERROR_CHECK(esp_wifi_set_channel(11, WIFI_SECOND_CHAN_NONE));
        printf("[DEBUG] Wifi setting channel  = 11\n");
        ESP_ERROR_CHECK(esp_wifi_set_promiscuous(false));
        printf("[DEBUG] Wifi setting promiscuous = false\n");
        ESP_ERROR_CHECK(esp_now_init());
        printf("[DEBUG] Wifi esp_now initialized\n");
        ESP_ERROR_CHECK(esp_wifi_get_channel(&primary, &secondary));
        printf("[DEBUG] Retrieved channel after setting it : %d\n", primary);
        /* To make sure the complete line is printed before entering sleep mode,
         * need to wait until UART TX FIFO is empty:
         */
        uart_wait_tx_idle_polling(CONFIG_ESP_CONSOLE_UART_NUM);
        /* Enter sleep mode */
        ESP_ERROR_CHECK(esp_wifi_stop());
        printf("[DEBUG] Wifi stopped without problems...\n");
        printf("[DEBUG] Entering light sleep...\n");
        ESP_ERROR_CHECK(esp_light_sleep_start());
        printf("[DEBUG] ESP module woke up...\n");
        ESP_ERROR_CHECK(esp_wifi_start());
        printf("[DEBUG] Wifi started without problems...\n");
        ESP_ERROR_CHECK(esp_wifi_get_channel(&primary, &secondary));
        printf("[DEBUG] Retrieved channel after sleeping: %d\n", primary);
    }

}
```
These are the logs I obtained:
```
entry 0x40080680
I (27) boot: ESP-IDF v4.3-356-g48ae2309fd-dirty 2nd stage bootloader
I (27) boot: compile time 14:09:20
I (27) boot: chip revision: 1
I (32) boot_comm: chip revision: 1, min. bootloader chip revision: 0
I (39) boot.esp32: SPI Speed      : 40MHz
I (43) boot.esp32: SPI Mode       : DIO
I (48) boot.esp32: SPI Flash Size : 2MB
I (52) boot: Enabling RNG early entropy source...
I (58) boot: Partition Table:
[... 62 lines cut ...]
I (691) phy_init: phy_version 4670,719f9f6,Feb 18 2021,17:07:07
I (821) wifi:mode : sta (24:0a:c4:58:c5:e0)
I (821) wifi:enable tsf
[DEBUG] Wifi started without problems...
[DEBUG] Wifi set mode STA
I (831) wifi:ic_enable_sniffer
[DEBUG] Wifi setting promiscuous = true
[DEBUG] Retrieved channel before setting channel: 1
[DEBUG] Wifi setting channel  = 11
I (841) wifi:ic_disable_sniffer
[DEBUG] Wifi setting promiscuous = false
I (851) ESPNOW: espnow [version: 1.0] init
[DEBUG] Wifi esp_now initialized
[DEBUG] Retrieved channel after setting it : 11
I (871) wifi:flush txq
I (871) wifi:stop sw txq
I (871) wifi:lmac stop hw txq
[DEBUG] Wifi stopped without problems...
[DEBUG] Entering light sleep...
[DEBUG] ESP module woke up...
I (881) wifi:mode : sta (24:0a:c4:58:c5:e0)
I (881) wifi:enable tsf
[DEBUG] Wifi started without problems...
[DEBUG] Retrieved channel after sleeping: 1
```

jonasfehr, 2021-08-30:
Thanks for testing and confirming. I hope espressif will fix it in upcoming versions. For now, the workaround works for me.

jack0c, 2022-05-20:
@jonasfehr It looks like esp_wifi_start() will not keep the channel used before esp_wifi_stop, right?
If so, we can fix it.

zhangyanjiaoesp, 2022-05-25:
esp_wifi_start() indeed doesn't keep the channel used before esp_wifi_stop, will fix it soon.

zhangyanjiaoesp, 2022-06-07:
@jonasfehr If you want to keep the channel info, you can call `esp_wifi_set_config()` to set the channel.

jonasfehr, 2022-06-07:
@zhangyanjiaoesp Thanks for looking into it. 
What is the best practice for my case. I have a device where i can activate a web-interface/configuration page to change the channel. The device goes to deep sleep, and when it wakes up, it should go back to the latest channel selected.
Right now, I save the channel to the NVS, and on each wakeup, I set the channel to the one I read from the NVS.
Is the esp_wifi_set_config() function also writing to the NVS. Which would mean I can update the config when I change the channel with my interface, but otherwise I can just use esp_wifi_start() recalling the settings automatically?

zhangyanjiaoesp, 2022-06-27:
@jonasfehr Yes, the `esp_wifi_set_config()` will write to the NVS. Since you said 

> However, in my project I solved it by setting the channel again after waking up, 

I think setting the channel again after waking up may be better for your case.

Alvin1Zhang, 2023-01-12:
Thanks for reporting and sharing the updates, sorry for slow turnaround, fix on master branch is available at https://github.com/espressif/esp-idf/commit/e6c57c0d14ab134be77fc5f7f477395b81cdeacb and on release/4.4 is https://github.com/espressif/esp-idf/commit/1c0d0e505d9fc50045bce4625a8c95aaf55af362. Feel free to reopen.

closed 2023-01-12 by Alvin1Zhang