espressif/esp-idf #7480: wprintf problem

lion1986, 2021-08-28:
demo code:
wchar_t wStr[] = L"demo wchat_t";
wprintf(L"wStr(%d):%s\n", wcslen(wStr), wStr);

print result: 
**wStr(12):d**
just only print the first char,  the string length is right.

negativekelvin, 2021-08-28:
>%ls shall be used for wide strings (and not %s).

Alvin1Zhang, 2021-08-30:
Thanks for reporting. Would you please help provide more details as suggested in the [issue template](https://github.com/espressif/esp-idf/issues/new?assignees=&labels=&template=bug_report.md&title=)? Information like elf, sdk configuration, backtrace, log outputs, commit ID, hardware and etc. would help us debug further. Thanks.

lion1986, 2021-08-30:
> 
> 
> > %ls shall be used for wide strings (and not %s).

When use %ls in wprintf function, the result is right. But when use %ls in printf function, the result is not right(just like %s in wprintf, only print the first char).

ESP-Marius, 2022-08-31:
> When use %ls in wprintf function, the result is right. But when use %ls in printf function, the result is not right(just like %s in wprintf, only print the first char).

Support for wide character support in streams was  disabled in [0dbbbbf68](https://github.com/espressif/crosstool-NG/commit/0dbbbbf68879ea2af2052b0c76f181cdb29baa39)

Currently there is no specific plans to re-enable this. Closing issue as the original issue regarding `wprintf` was already resolved.

closed 2022-08-31 by ESP-Marius