espressif/esp-idf #7483: custom CMake projects do not produce a project_description.json

jdoubleu, 2021-08-28:
## Environment
- IDF version (run ``git describe --tags`` to find it): `v4.3`
- Build System: CMake

## Problem Description
When using the [custom CMake project](https://docs.espressif.com/projects/esp-idf/en/stable/esp32/api-guides/build-system.html?highlight=build%20system#using-esp-idf-in-custom-cmake-projects) setup, the `project_description.json` will not be generated. That file is crucial for other tools like `idf.py monitor`.

The `project_description.json` will only be created by the default setup from [`project.cmake`](https://github.com/espressif/esp-idf/blob/master/tools/cmake/project.cmake#L115-L120).

It would be nice, if [`__project_info`](https://github.com/espressif/esp-idf/blob/master/tools/cmake/project.cmake#L70) could be exposed somewhere else, for example in `tools/cmake/idf.cmake` so that custom CMake projects could include it to setup all necessary meta information.

KonstantinKondrashov, 2024-06-04:
Hi @EpicKiwi!
Could give more information about the use case? The original is quite old and for v4.3. Are you also faced with this issue on v4.3?
Have you tried it on the master?

Alvin1Zhang, 2025-03-13:
@EpicKiwi Would you please help share if any updates? Thanks.

EpicKiwi, 2025-03-13:
Hey ! Thank you all and sorry for bot replying. It's been a long time since I worked on ESP 32 with C code and didn't tested recently if issue is still here.

My use case is the following:
I do have a software written in C with a CMake project helping to build it out. A goal of this software is to be cross-platform with microcontrollers in mind (ESP32 included) but it's currently only running on POSIX platform. We do have an abstraction layer that makes syscall and that will be adapted depending of the platform.
But I do not want to make a special cmake project for esp32 with Espressif toolkit becaus it would means that we should keep in sync two different cmake projects with differents API.

This is my project cmake file with a try to compile for esp32 https://github.com/archipel-network/archipel-core/blob/esp32/CMakeLists.txt

But it's been a long time since I worked on that so it's maybe broken with current version.

KonstantinKondrashov, 2025-03-13:
Hello @EpicKiwi @jdoubleu!
We have the example of such a use case when ESP-IDF components are used as libraries in custom CMake projects - [idf_as_lib ](https://github.com/espressif/esp-idf/blob/master/examples/build_system/cmake/idf_as_lib/CMakeLists.txt). I ran this example (for master) and do observe that the `project_description.json` file is not created.

I had a look at the code where this file is created. The `__project_info` function from the `tools/cmake/project.cmake` generates this JSON file. So, I would suggest calling this function directly from your CMake file. We definitely can not fix the v4.3 which is out of the maintenance period. 
```cmake
include($ENV{IDF_PATH}/tools/cmake/project.cmake)
__project_info("")
```
This approach works, however, it has some potential corner cases you should be aware of. Including the `project.cmake` file seems to automatically process the build using `idf_build_process` which is not what you expect.
Probably we could move `__project_info()` into the existing `idf.cmake` file ([that has to be already used in the custom CMake](https://docs.espressif.com/projects/esp-idf/en/stable/esp32/api-guides/build-system.html#using-esp-idf-in-custom-cmake-projects)) but it also requires to take a few more functions too (__all_component_info and __build_component_info).

We will discuss internally how to perform it better, at least now you have a solution to try. Thanks.