espressif/esp-idf #7485: Significant changes in 4.2.1 and 4.3 with freeRTOS?

RoscoeTheDog, 2021-08-30:
I started a project using platformio with esp-idf 4.2.1 as the framework. Much later after it is almost done, I ran into an issue with any and all GPIO pins flooding the console with "phy -- pll_cap_ext" errors whenever the HTTP client is initialized for some reason. It seemed this was an issue related to the logger and the wifi module in some other posts. I updated esp-idf to version 4.3 via platformIO and now I am running to even more unrelated issues. Many tasks which worked flawless before, now oddly enough cause the watchdog task wdt to trip and I'm getting also sorts of weird blocking behaviors between tasks. Again, this was completely fine before the update.

vTaskList() method was removed in the update (at least in the platformIO branch). However, it still exists in the documentation and from what I can see also the framework headers. I'm not sure if the "vTaskList()" build error I am encountering is related to the build system of platformIO or if there is a legacy compatibility issue with esp-idf and freeRTOS.

What has changed between 4.2.1 and 4.3 in esp-idf and freeRTOS The documentation for esp-idf seems to be up to date with the latest 4.3 version and I'm not sure how it would disclose any changes between versions (**if** there is even to begin with). I know the older documentation is also available as well.

Is there something I am missing between these versions? Do you think my issues are related to the platformIO build system. freeRTOS or something else?

I realize there is no code example offered here because my project is fairly big and I'm just looking for general information about the legacy support between these two framework versions.

Dazza0, 2021-08-30:
@RoscoeTheDog Yes, v4.3 upgraded FreeRTOS from v8.2.0 to v10.2.1. But the API interface of `vTaskList()` should not have changed. FreeRTOS upgrades should always API backwards compatible unless expressly stated otherwise.

Could you check if you've enabled the `CONFIG_FREERTOS_USE_STATS_FORMATTING_FUNCTIONS` option in your menuconfig?

igrr, 2021-08-30:
Regarding the changes between v4.2.1 and v4.3, unfortunately we don't have a direct comparison between these two versions. There is a changelog in the release notes of v4.3, which is written compared to release v4.2: https://github.com/espressif/esp-idf/releases/tag/v4.3. Some of these changes (mainly bug fixes) are also present in v4.2.1, when compared to v4.2: https://github.com/espressif/esp-idf/releases/tag/v4.2.1. So the list of differences between v4.2.1 and v4.3 is a bit smaller than what is listed in v4.3 release notes.

RoscoeTheDog, 2021-08-30:
I am cleaning up my project a bit and I'll have a repository published in a moment that you can pull, upload to a esp32s2 and test this bug.

RoscoeTheDog, 2021-08-30:
Here is the link to a branch special made for troubleshooting this issue. You should be able to download the zip, open vscode and platformIO and build it without problems. When it's uploaded to an s2 board you should see exactly what I described.

o-marshmallow, 2021-09-07:
Hello @RoscoeTheDog !
Thank you for sharing an example. I "converted" the project you linked (I just noticed it was your master branch, but should also work for the other branch) to compile and ran on our latest version, as well as on 4.3. I did run into the issues you described.
Here is what I've found: 
* You initialize the wifi after calling [initializeTasks](https://github.com/RoscoeTheDog/ESP32-S2-Saola-Project/blob/master/ESP32_S2_Saola1/src/main.c#L110), which is too late, because this one will be in charge of creating and executing `vTaskWifiReconnect` task.
* SSID and PASSWORD were copied after calling `initializeTasks` (on `master`, but doesn't seem to be the case on `troubleshooting` branch)
* In `vTaskWifiReconnect` function, you call `vTaskDelay(pdMS_TO_TICKS(1));`, but 1ms is too low. Indeed, by default, FreeRTOS' tick is set to 10ms in the menuconfig. Thus, any value below will cause the task to not sleep. the easiest solution is your case, which is the one I test, it so get rid of `esp_task_wdt_reset();`, `esp_task_wdt_delete(xHandleWifiReconnect);`, `esp_task_wdt_add(xHandleWifiReconnect);` and simply replace `vTaskDelay(pdMS_TO_TICKS(1));` by `vTaskDelay(pdMS_TO_TICKS(10));` (or `vTaskDelay(1);`, which is equivalent)

After this, you will be able to compile, flash and run. Here is the output I got:
```
I (402) gpio: GPIO[33]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 1| Intr:0 
I (402) gpio: GPIO[2]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 1| Intr:0 
I (412) gpio: GPIO[0]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 1| Intr:0 
I (422) gpio: GPIO[21]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 0| Pulldown: 1| Intr:0 
I (432) gpio: GPIO[19]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 1| Intr:0 
I (442) gpio: GPIO[26]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 0| Pulldown: 1| Intr:0 
I (452) gpio: GPIO[34]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
I (462) gpio: GPIO[35]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
I (472) gpio: GPIO[36]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
I (482) initialize_ledc_config_0: LEDC FREQUENCY:  100
I (482) initialize_ledc_config_0: LEDC BIT RESOLUTION:  11
I (492) initialize_ledc_config_0: LEDC BIT INTEGER MAX:  2047
I (2502) nvsRestoreSystemState: RESTORING PREVIOUS STATE
I (3512) updateWifiConfig: active wifi config updated
I (3522) wifi:wifi driver task: 3ffd8c28, prio:23, stack:6656, core=0
I (3522) system_api: Base MAC address is not set
I (3522) system_api: read default base MAC address from EFUSE
I (3532) wifi:wifi firmware version: 42b8729
I (3532) wifi:wifi certification version: v7.0
I (3532) wifi:config NVS flash: enabled
I (3532) wifi:config nano formating: disabled
I (3542) wifi:Init data frame dynamic rx buffer num: 32
I (3542) wifi:Init management frame dynamic rx buffer num: 32
I (3552) wifi:Init management short buffer num: 32
I (3552) wifi:Init dynamic tx buffer num: 32
I (3562) wifi:Init static rx buffer size: 1600
I (3562) wifi:Init static rx buffer num: 10
I (3562) wifi:Init dynamic rx buffer num: 32
I (3572) wifi_init: rx ba win: 6
I (3572) wifi_init: tcpip mbox: 32
I (3582) wifi_init: udp mbox: 6
I (3582) wifi_init: tcp mbox: 6
I (3582) wifi_init: tcp tx win: 5744
I (3592) wifi_init: tcp rx win: 5744
I (3592) wifi_init: tcp mss: 1440
I (3602) wifi_init: WiFi IRAM OP enabled
I (3602) wifi_init: WiFi RX IRAM OP enabled
I (3612) phy_init: phy_version 1800,e7ef680,Apr 13 2021,11:45:08
I (3742) wifi:mode : sta (7c:df:a1:05:d0:0c)
I (3742) wifi:enable tsf
I (3742) initializeTasks: initializing vTaskRTOSDebug
I (3742) initializeTasks: initializing vTaskSleep
I (3752) initializeTasks: initializing vTaskStatusLEDWatchdog
I (3752) initializeTasks: initializing vTaskLEDFade
I (3762) initializeTasks: initializing vTaskMoveStepperForward
I (3772) initializeTasks: initializing vTaskMoveStepperReverse
I (3772) initializeTasks: initializing vTaskWifiReconnect
I (3782) updateWifiConfig: active wifi config updated
I (3792) vTaskWifiReconnect: Attempting connect to AP...
I (3792) vTaskWifiReconnect: SSID: MY_SSID
I (3802) vTaskWifiReconnect: PASSWORD:  MY_PASSWORD
I (3802) vTaskWifiReconnect: Connected.
I (3812) initializeTasks: initializing vTaskSubmitLocalData
I (3812) initializeTasks: initializing vTaskUpdateMotor
I (3822) initializeTasks: initializing vTaskPollServer
I (3832) initializeTasks: initializing vTaskWifiPersistingTasks
I (3852) wifi:new:<13,0>, old:<1,0>, ap:<255,255>, sta:<13,0>, prof:1
I (3862) wifi:state: init -> auth (b0)
I (3862) wifi:state: auth -> assoc (0)
I (3882) wifi:state: assoc -> run (10)
I (3912) wifi:connected with RedmiN5, aid = 9, channel 13, BW20, bssid = d8:32:e3:05:3f:b8
I (3912) wifi:security: WPA2-PSK, phy: bgn, rssi: -20
I (3922) wifi:pm start, type: 1

Connected.
I (3922) initializeSntpUpdate: initializing sntpUpdate service
W (3932) wifi:<ba-add>idx:0 (ifx:0, d8:32:e3:05:3f:b8), tid:0, ssn:0, winSize:64
I (3952) wifi:AP's beacon interval = 102400 us, DTIM period = 2
I (4512) esp_netif_handlers: sta ip: 192.168.43.170, mask: 255.255.255.0, gw: 192.168.43.1
I (4752) vTaskStatusLEDWatchdog: SYSTEM WAITING ON DATETIME SYNC
I (5952) vTaskStatusLEDWatchdog: SYSTEM WAITING ON DATETIME SYNC
I (7152) vTaskStatusLEDWatchdog: SYSTEM WAITING ON DATETIME SYNC
I (7242) sntpUpdateNotificiation: server updated date/time to Mon Sep  6 20:59:36 2021 PST
I (7832) httpPostData: HTTP REQUEST -- {"READ_KEY":"abc","DeviceID":"1"}
I (7832) initializeHttpClient: setting POST url to: http://192.168.1.84:8000/devices/smartblinds/
I (7832) initializeHttpClient: initializing new http client
```
Of course I couldn't go further because I don't have any server set up at `192.168.1.84:8000`

o-marshmallow, 2021-09-07:
@RoscoeTheDog You'll find the modifications for making your project compile and working with ESP-IDF here: https://github.com/RoscoeTheDog/ESP32-S2-Saola-Project/pull/6

Dazza0, 2021-09-23:
@RoscoeTheDog  Closing this issue as it doesn't seem to be a bug within FreeRTOS. Feel free to reopen

closed 2021-09-23 by Dazza0

RoscoeTheDog, 2022-07-12:
> @RoscoeTheDog You'll find the modifications for making your project compile and working with ESP-IDF here: [RoscoeTheDog/ESP32-S2-Saola-Project#6](https://github.com/RoscoeTheDog/ESP32-S2-Saola-Project/pull/6)

Thank you for the invaluable insights. Life stuff got chaotic and I had to put my project aside for awhile. I'll be sure to check out your notes and report back my results after I'm done cleaning up my repositories!