espressif/esp-idf #7486: Socket leak on "esp_https_server" component

baldhead69, 2021-08-30:
Hi,

I am writing a websocket server and i did some tests opening and closing connections with the server.

After about 9 requests the server rejects new connections.
The registered closed callback function are called in all 9 closed sockets.

Error:
```c
    httpd: httpd_accept_conn: error in accept (23)
    httpd: httpd_server: error accepting new connection
```
Config:
```c
    httpd_ssl_config_t config = HTTPD_SSL_CONFIG_DEFAULT();  
    config.httpd.core_id = tskNO_AFFINITY; 
    config.httpd.max_open_sockets = 5;
    config.httpd.task_priority =  tskIDLE_PRIORITY + 5;      
    config.httpd.close_fn = closed_socket_handler;
    config.transport_mode = HTTPD_SSL_TRANSPORT_INSECURE;   
    config.httpd.lru_purge_enable = false;  // The lib limits to 5 openned sockets, but closes oldest socket and opens another if "lru_purge_enable = true".  
```
Why the the server are rejecting new connections if callback "closed_socket_handler" are called in all times ?

Some help are welcome.

Thank's.

ESP-YJM, 2021-08-31:
@baldhead69 When we close socket, if socket not be closed by `Four-way handshake`, the socket will enter `Time-wait` state, maybe 120s (not sure). In this state,  the socket will not be completely released, and the socket fd will still be occupied. For solve this issue, you can use `SO_LINGER` option and force close the socket.

ESP-YJM, 2021-08-31:
@baldhead69 Could you please set `lru_purge_enable ` true in `httpd_config_t ` and try.

baldhead69, 2021-09-01:
> @baldhead69 When we close socket, if socket not be closed by `Four-way handshake`, the socket will enter `Time-wait` state, maybe 120s (not sure). In this state, the socket will not be completely [...]

How to do this and where ?

Thank's.

baldhead69, 2021-09-01:
I enabled "Enable SO_LINGER processing" in menuconfig "Component config->LWIP", but not solved.

I think i need set SO_LINGER = true and/or TIME_WAIT = 0, but i don't know where or how.

baldhead69, 2021-09-01:
> @baldhead69 Could you please set `lru_purge_enable ` true in `httpd_config_t ` and try.

Not solved, same problem.

After nine connection:
```c
W (00:00:39.136) httpd: httpd_accept_conn: error in accept (23)
W (00:00:39.136) httpd: httpd_server: error accepting new connection
W (00:00:44.358) httpd: httpd_accept_conn: error in accept (23)
W (00:00:44.358) httpd: httpd_server: error accepting new connection
W (00:03:18.262) httpd: httpd_accept_conn: error in accept (23)
W (00:03:18.262) httpd: httpd_server: error accepting new connection
```
[image]
[image]

ESP-YJM, 2021-09-01:
> 
> 
> I enabled "Enable SO_LINGER processing" in menuconfig "Component config->LWIP", but not solved.
> 
> I think i need set SO_LINGER = true and/or TIME_WAIT = 0, but i don't know where or how.

@baldhead69 You can refer to https://esp32.com/viewtopic.php?f=13&t=20705

baldhead69, 2021-09-01:
> @baldhead69 When we close socket, if socket not be closed by `Four-way handshake`, the socket will enter `Time-wait` state, maybe 120s (not sure). In this state, the socket will not be completely [...]

Looks like it's solved.

baldhead69, 2021-09-01:
@ESP-YJM 

I don't know why closing the connection by Jmeter is returning an error, maybe the websocket server is not responding correctly with a closing frame.

I must open a new issue for this ?

ESP-YJM, 2021-09-01:
> Looks like it's solved.

Glad to hear the good news.  From the wireshark packet, i think the websocket layer close is right. But need check the TCP layer close. I think you can follow this issue and no need open a new.

baldhead69, 2021-09-01:
@ESP-YJM 

I did two tests.

1. First test: open and close connection without user(i) send a websocket close frame.

[image]

The server are not sending a websocket close frame(ie, not a clean websocket close protocol).
[image]

2. Second test: open and close connection with user(i) send a websocket close frame.

[image]

The server send a websocket close frame(ie, a clean websocket close protocol).
[image]

baldhead69, 2021-09-09:
@ESP-YJM 

I use visual studio code with esp-idf.
The code compile ok, but visual studio code show that not encounter some data structs.

I include "#include <lwip/sockets.h>", but inside this file appear in line 44 that "LWIP_SOCKET" is not enabled.

Do you know how/where i can enable "LWIP_SOCKET"  ???

Thank's.

[image]

[image]

ESP-YJM, 2021-09-10:
@baldhead69 I think LWIP_SOCKET is enabled in opt.h https://github.com/espressif/esp-lwip/blob/2c9c531f0a7e0ee536db9de4f9dc54e453712087/src/include/lwip/opt.h#L1985

baldhead69, 2021-09-13:
@ESP-YJM 

I think visual studio code are buggy, but i dont have assurance.

[image]

[image]

ESP-YJM, 2021-09-14:
@baldhead69 I think you no need care about the display in vscode. `C:\esp-idf\components\coap\libcoap\examples\lwip\lwipopts.h` we don't use it. If you want to know whether `LWIP_SOCKET` defined, you can do some build test.

baldhead69, 2021-09-15:
> @baldhead69 I think you no need care about the display in vscode. `C:\esp-idf\components\coap\libcoap\examples\lwip\lwipopts.h` we don't use it. If you want to know whether `LWIP_SOCKET` defined, you can do some build test.

Yes, but it's annoying to see the failure in the code if you use this feature to find errors in the program.

ESP-YJM, 2021-09-16:
> Yes, but it's annoying to see the failure in the code if you use this feature to find errors in the program.

I agree that but it is difficult to make vscode identify which macro is defined, due to some macro generated when make build. I think this topic is not related to the title issue. Do you have any doubt or comments about title issue? If no, i think we can close it and open other topic issue to disscuss vscode display.

baldhead69, 2021-09-16:
@ESP-YJM 

The close frame issue remains to be resolved.

May be you can close here and you can continue here with the close frame:
https://github.com/espressif/esp-idf/issues/7493#issuecomment-912203846

The uri handler are not called when client close frame and the "background websocket layer" do not send a close message to do a clean close. 

Thank's.

ESP-YJM, 2021-09-16:
@baldhead69 Yes, current version not support do a clean close. I will keep #7493 open and fixed in this issue tickets. And #7486 maybe can be closed. You can help close #7486.

closed 2021-09-16 by baldhead69

sxsx2yzyz, 2021-11-24:
> > I enabled "Enable SO_LINGER processing" in menuconfig "Component config->LWIP", but not solved.
> > I think i need set SO_LINGER = true and/or TIME_WAIT = 0, but i don't know where or how.
> 
> @baldhead69 You can refer to https://esp32.com/viewtopic.php?f=13&t=20705

After I upgraded my idf version from 4.2 to 5.0, I encountered the same problem.This post helped me.Thanks