espressif/esp-idf #7488: How to get bluetooth device name?

takumi-kin, 2021-08-30:
Hello everyone,

I try to get bluetooth device name when pairing.
But I can't get device name of "MIDLAND BTX1 PRO S".
I could get other devices name.

I use sample code, "bluetooth\bluedroid\classic_bt\bt_discovery".
In update_device_info(), MIDLAND device has only two types, "ESP_BT_GAP_DEV_PROP_COD" and "ESP_BT_GAP_DEV_PROP_RSSI".
Eir is Null so I can't get name in get_name_from_eir().

Do you know how to get device name anyway?

```
static void update_device_info(esp_bt_gap_cb_param_t *param)
{
    char bda_str[18];
    uint32_t cod = 0;
    int32_t rssi = -129; /* invalid value */
    esp_bt_gap_dev_prop_t *p;

    ・・・

    memcpy(p_dev->bda, param->disc_res.bda, ESP_BD_ADDR_LEN);
    p_dev->dev_found = true;
    for (int i = 0; i < param->disc_res.num_prop; i++) {
        p = param->disc_res.prop + i;
        switch (p->type) {
        case ESP_BT_GAP_DEV_PROP_COD:
            p_dev->cod = *(uint32_t *)(p->val);
            break;
        case ESP_BT_GAP_DEV_PROP_RSSI:
            p_dev->rssi = *(int8_t *)(p->val);
            break;
        case ESP_BT_GAP_DEV_PROP_BDNAME: {
            uint8_t len = (p->len > ESP_BT_GAP_MAX_BDNAME_LEN) ? ESP_BT_GAP_MAX_BDNAME_LEN :
                          (uint8_t)p->len;
            memcpy(p_dev->bdname, (uint8_t *)(p->val), len);
            p_dev->bdname[len] = '\0';
            p_dev->bdname_len = len;
            break;
        }
        case ESP_BT_GAP_DEV_PROP_EIR: {
            memcpy(p_dev->eir, (uint8_t *)(p->val), p->len);
            p_dev->eir_len = p->len;
            break;
        }
        default:
            break;
        }
    }
    ・・・
}
```

```
static bool get_name_from_eir(uint8_t *eir, uint8_t *bdname, uint8_t *bdname_len)
{
    uint8_t *rmt_bdname = NULL;
    uint8_t rmt_bdname_len = 0;

    if (!eir) {
        return false;
    }

    rmt_bdname = esp_bt_gap_resolve_eir_data(eir, ESP_BT_EIR_TYPE_CMPL_LOCAL_NAME, &rmt_bdname_len);
    if (!rmt_bdname) {
        rmt_bdname = esp_bt_gap_resolve_eir_data(eir, ESP_BT_EIR_TYPE_SHORT_LOCAL_NAME, &rmt_bdname_len);
    }

    if (rmt_bdname) {
        if (rmt_bdname_len > ESP_BT_GAP_MAX_BDNAME_LEN) {
            rmt_bdname_len = ESP_BT_GAP_MAX_BDNAME_LEN;
        }

        if (bdname) {
            memcpy(bdname, rmt_bdname, rmt_bdname_len);
            bdname[rmt_bdname_len] = '\0';
        }
        if (bdname_len) {
            *bdname_len = rmt_bdname_len;
        }
        return true;
    }

    return false;
}
```

Wth-Esp, 2021-08-30:
Hi

The device name can be placed in EIR and also can be obtained by calling `[this API](https://gitlab.espressif.cn:6688/espressif/esp-idf/-/blob/master/components/bt/host/bluedroid/api/esp_gap_bt_api.c#L396)`.

Please have a try.

Thanks

Alvin1Zhang, 2021-08-31:
Thanks for reporting. For How questions, please raise a post on forum https://esp32.com, will close this ticket now.

closed 2021-08-31 by Alvin1Zhang