espressif/esp-idf #7489: udp_multicast_test can not read response by multicast

achao1104, 2021-08-30:
Hello,

I am running sample of "udp_multicast_test' in ipv6 only. esp32s2 can send and receive to the local link ip of ipv6 of slave device correctly.
but if esp32s2 multicast data to a multicast ipv6 address such as "FF02::1", the slave device can receive multicast data and respose to the ip of esp32s2, but esp32s2 can not read the data. The following line always return 0. Please kindly help.
int s = select(sock + 1, &rfds, NULL, NULL, &tv);

Peter

achao1104, 2021-08-30:
I installed wireshark to cap the pack. it looks like below:
if esp32 sending udp to ip of "FF02::1" of slave device, there are no neighbor advertisment or solicitation are issued from esp32. but there are UDP pack issued.
if esp32 sending udp to ip of local link of slave device, there are neighbor advertisment or solicitation are issued from esp32. the UDP works fine for both sending and recving, then how to solve it?

achao1104, 2021-08-30:
sorry for that the term multicast used here should be unicast for ipv6. thanks.

achao1104, 2021-08-31:
we installed wireshark, the issue looks like beow:
esp32 (fe80::ff:fe12:3456) sends multicast UDP to slave device(ff02::1)
slave device receives the UDP and make a neighbor solicitation request with destination address like below
ff02::1:ff12:3456 which is different from esp32 ip address(fe80::ff:fe12:3456)
the destination address's prefix looks come from the multicast address and the suffix comes from the esp32 address.
where is wrong? is there any parameter to set before sending UPD multicast pack?
thanks.

negativekelvin, 2021-08-31:
Maybe esp_netif_join_ip6_multicast_group first?

ESP-YJM, 2021-08-31:
@achao1104 When slave device send unicast IPv6 address to master, it will first send NS(neighbor solicitation) to multicast address of destination, which you saw in wireshark. It is right way to get destination device MAC address. The master device will receive this NS packet if the master device listened this multicast address when it achieve the link-local address. And master device will response NA(neighbor advertisement). If not, please check the master device whether listened itself multicast address.

achao1104, 2021-08-31:
thanks a lot, you are right.

closed 2021-08-31 by achao1104